probe-rs/probe-rs #2274: Panic handler not capturing all stack frames

tallamjr, 2024-03-10:
**Describe the bug**
When attempting to run an example in [ferrous-systems/rust-exercises](https://github.com/ferrous-systems/rust-exercises/blob/main/exercise-book/src/nrf52-radio-panicking.md) I get the following output:
```bash
$ cargo run --bin panic
    Finished dev [optimized + debuginfo] target(s) in 0.83s
     Running `probe-rs run --chip nRF52840_xxAA target/thumbv7em-none-eabihf/debug/panic`
      Erasing ✔ [00:00:00] [#####################################################################################################################################] 8.00 KiB/8.00 KiB @ 12.09 KiB/s (eta 0s )
  Programming ✔ [00:00:00] [######################################################################################################################################] 8.00 KiB/8.00 KiB @ 8.93 KiB/s (eta 0s )    Finished in 1.605s
ERROR panicked at src/bin/panic.rs:30:13:
index out of bounds: the len is 3 but the index is 3
└─ panic_probe::print_defmt::print @ /Users/tallamjr/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:104
<lvl> `dk::fail()` called; exiting ...
└─ dk::fail @ /Users/tallamjr/github/tallamjr/forks/rust-exercises/nrf52-code/boards/dk/src/lib.rs:322
 WARN probe_rs::debug::unit_info: Unsupported DW_AT_abstract_origin value: DebugInfoRef(DebugInfoOffset(7903))
 WARN probe_rs::debug::unit_info: Unsupported DW_AT_abstract_origin value: DebugInfoRef(DebugInfoOffset(7866))
 WARN probe_rs::debug::unit_info: Unsupported DW_AT_abstract_origin value: DebugInfoRef(DebugInfoOffset(7929))
Frame 0: fail @ 0x000012e4
       /Users/tallamjr/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-semihosting-0.5.0/src/lib.rs:201:13
Frame 1: <unknown function @ 0x00001654> @ 0x00001654
Error: Semihosting indicates exit with failure code: 0x020023 (131107)

Stack backtrace:
   0: std::backtrace::Backtrace::create
   1: std::backtrace::Backtrace::capture
   2: anyhow::error::<impl anyhow::Error>::msg
   3: anyhow::__private::format_err
   4: probe_rs::cmd::run::Cmd::run
   5: probe_rs::main
   6: std::sys_common::backtrace::__rust_begin_short_backtrace
   7: std::rt::lang_start::{{closure}}
   8: std::rt::lang_start_internal
   9: _main
  10: <unknown>
```
Which is different than the expected output shown in the example.

**To Reproduce**
Steps to reproduce the behavior:

1. Connect nRF board
2. `git clone https://github.com/ferrous-systems/rust-exercises.git rust-exercises`
3. `cd rust-exercises/nrf52-code/radio-app && cargo run --bin panic`

**Expected behavior**
I expected to see same output, or similar to that shown in the example with all stack frames shown:
```bash
$ cargo run --bin panic
   Compiling defmt-macros v0.3.6
   Compiling defmt v0.3.5
   Compiling defmt-rtt v0.4.0
   Compiling panic-probe v0.3.1
   Compiling dk v0.0.0 (/Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/boards/dk)
   Compiling radio_app v0.0.0 (/Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app)
    Finished dev [optimized + debuginfo] target(s) in 1.27s
     Running `probe-rs run --chip nRF52840_xxAA target/thumbv7em-none-eabihf/debug/panic`
      Erasing ✔ [00:00:00] [#######################################################################################################################################] 16.00 KiB/16.00 KiB @ 32.26 KiB/s (eta 0s )
  Programming ✔ [00:00:00] [#######################################################################################################################################] 16.00 KiB/16.00 KiB @ 41.48 KiB/s (eta 0s )    Finished in 0.904s
ERROR panicked at src/bin/panic.rs:30:13:
index out of bounds: the len is 3 but the index is 3
└─ panic_probe::print_defmt::print @ /Users/jonathan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:104
`dk::fail()` called; exiting ...
Frame 0: fail @ 0x00001308
       /Users/jonathan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-semihosting-0.5.0/src/lib.rs:201:13
Frame 1: __cortex_m_rt_HardFault @ 0x000016a6 inline
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/lib.rs:12:5
Frame 2: __cortex_m_rt_HardFault_trampoline @ 0x00000000000016a2
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/lib.rs:10:1
Frame 3: "HardFault handler. Cause: Escalated UsageFault (Undefined instruction)." @ 0x000016a6
Frame 4: __udf @ 0x00001530 inline
       ./asm/lib.rs:48:1
Frame 5: __udf @ 0x0000000000001530
       ./asm/lib.rs:51:17
Frame 6: udf @ 0x0000151c
       /Users/jonathan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-0.7.7/src/asm.rs:43:5
Frame 7: hard_fault @ 0x0000150e
       /Users/jonathan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:86:5
Frame 8: panic @ 0x000014dc
       /Users/jonathan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:54:9
Frame 9: panic_fmt @ 0x0000034a
       /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/panicking.rs:72:14
Frame 10: panic_bounds_check @ 0x000003fe
       /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/panicking.rs:190:5
Frame 11: bar @ 0x00000180
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/bin/panic.rs:30:13
Frame 12: foo @ 0x00000176
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/bin/panic.rs:24:2
Frame 13: __cortex_m_rt_main @ 0x000002de
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/bin/panic.rs:13:5
Frame 14: __cortex_m_rt_main_trampoline @ 0x0000018a
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/bin/panic.rs:9:1
Frame 15: memmove @ 0x0000013c
Frame 16: memmove @ 0x0000013c
Error: Semihosting indicates exit with failure code: 0x020023 (131107)
```

**Stacktrace**
```bash
Stack backtrace:
   0: std::backtrace::Backtrace::create
   1: std::backtrace::Backtrace::capture
   2: anyhow::error::<impl anyhow::Error>::msg
   3: anyhow::__private::format_err
   4: probe_rs::cmd::run::Cmd::run
   5: probe_rs::main
   6: std::sys_common::backtrace::__rust_begin_short_backtrace
   7: std::rt::lang_start::{{closure}}
   8: std::rt::lang_start_internal
   9: _main
  10: <unknown>
  ```

**Desktop (please complete the following information):**

- macOS Ventura 13.6.4: `RELEASE_X86_64 x86_64 i386 Darwin`

**Additional context**
- https://github.com/ferrous-systems/rust-exercises/blob/main/exercise-book/src/nrf52-radio-panicking.md
cc @thejpster who may be able to provide more context

thejpster, 2024-03-10:
This works fine for me on Apple Silicon with the same probe-rs version.

tallamjr, 2024-03-10:
> This works fine for me on Apple Silicon with the same probe-rs version.
> 
> @tallamjr what rustc version are you using? I was using stable.

My apologies, I've now added the `probe-rs` and `rustc` versions I was using in the additional context section. I have also just tested this using `rustc 1.76.0 (07dca489a 2024-02-04)` which produces similar error.

noppej, 2024-03-11:
@thejpster Can you please confirm for me what version of `cortex-m-rt` you are using? 

Release 7.2 introduced the `global_asm` stuff, which actually overwrites the EXC_RETURN magic values that are set by the hardware on a HardFault. This breaks the `probe-rs` ability to unwind beyond the exception handler.

I have it on my todo list to try to compensate/workaround the `cortex-m-rt` change, but have not gotten around to it.

noppej, 2024-03-12:
@thejpster I just looked at this again, and am confident that it is related to [this new issue I logged](https://github.com/rust-embedded/cortex-m/issues/514).

The unreleased code for `cortex-m-rt` has a workaround (I describe it in the above issue), but it requires a code change to the your hard fault handler code.  I'd love to hear if you have suggestions for how the `probe-rs` debug code can handle this differently.

thejpster, 2024-03-12:
The last cortex-m-rt release was a year ago. I'm pretty sure this is a recent issue, and we have two machines with the same embedded hardware running the same application where one crashes probe-rs and one does not.

jonathanpallant, 2024-03-13:
I have cortex-m-rt 0.7.3 [in the Cargo.lock file](https://github.com/ferrous-systems/rust-exercises/blob/8088f671fe5c4e20b3c0372370c7e70961f4b91b/nrf52-code/radio-app/Cargo.lock#L70) I ship with the embedded project.

The OP's bug was specially about a program that causes a panic where the panic handler does a semihosting error. This works fine on my machine but does not work on the OP's machine - it causes *probe-rs itself to quit with an anyhow error*.

Here is what I see when I run the panic program.

```console
$ cargo run --bin panic
   Compiling defmt v0.3.5
   Compiling defmt-rtt v0.4.0
   Compiling panic-probe v0.3.1
   Compiling dk v0.0.0 (/Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/boards/dk)
   Compiling radio_app v0.0.0 (/Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app)
    Finished dev [optimized + debuginfo] target(s) in 1.43s
     Running `probe-rs run --chip nRF52840_xxAA target/thumbv7em-none-eabihf/debug/panic`
 WARN probe_rs::config::sequences::nrf52: Core is locked. Erase procedure will be started to unlock it.
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: An operation could not be performed because it lacked the permission to do so: erase_all
$ cargo run --bin panic  -- --allow-erase-all
    Finished dev [optimized + debuginfo] target(s) in 0.03s
     Running `probe-rs run --chip nRF52840_xxAA target/thumbv7em-none-eabihf/debug/panic --allow-erase-all`
 WARN probe_rs::config::sequences::nrf52: Core is locked. Erase procedure will be started to unlock it.
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
      Erasing ✔ [00:00:00] [############################################################] 8.00 KiB/8.00 KiB @ 27.68 KiB/s (eta 0s )
  Programming ✔ [00:00:00] [############################################################] 8.00 KiB/8.00 KiB @ 23.44 KiB/s (eta 0s )    Finished in 0.661s
ERROR panicked at src/bin/panic.rs:30:13:
index out of bounds: the len is 3 but the index is 3
└─ panic_probe::print_defmt::print @ /Users/jonathan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:104
<lvl> `dk::fail()` called; exiting ...
└─ dk::fail @ /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/boards/dk/src/lib.rs:345
 WARN probe_rs::debug::unit_info: Unsupported DW_AT_abstract_origin value: DebugInfoRef(DebugInfoOffset(7798))
 WARN probe_rs::debug::unit_info: Unsupported DW_AT_abstract_origin value: DebugInfoRef(DebugInfoOffset(7761))
 WARN probe_rs::debug::unit_info: Unsupported DW_AT_abstract_origin value: DebugInfoRef(DebugInfoOffset(7824))
 WARN probe_rs::debug::debug_info: UNWIND: Error while checking for exception context. The stack trace will not include the calling frames. : Register error: No value for Return Address register. Please report this as a bug.
 WARN probe_rs::debug::debug_info: UNWIND: Error while checking for exception context. The stack trace will not include the calling frames. : Register error: No value for Return Address register. Please report this as a bug.
Frame 0: fail @ 0x00001300
       /Users/jonathan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-semihosting-0.5.0/src/lib.rs:201:13
Frame 1: __cortex_m_rt_HardFault @ 0x0000169e inline
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/lib.rs:12:5
Frame 2: __cortex_m_rt_HardFault_trampoline @ 0x000000000000169a
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/lib.rs:10:1
Frame 3: "HardFault handler. Cause: Escalated UsageFault (Undefined instruction)." @ 0x0000169e
Frame 4: __udf @ 0x00001528 inline
       ./asm/lib.rs:48:1
Frame 5: __udf @ 0x0000000000001528
       ./asm/lib.rs:51:17
Frame 6: udf @ 0x00001514
       /Users/jonathan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-0.7.7/src/asm.rs:43:5
Frame 7: hard_fault @ 0x00001506
       /Users/jonathan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:86:5
Frame 8: panic @ 0x000014d4
       /Users/jonathan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:54:9
Frame 9: panic_fmt @ 0x00000342
       /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/panicking.rs:72:14
Frame 10: panic_bounds_check @ 0x000003f6
       /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/panicking.rs:190:5
Frame 11: bar @ 0x00000180
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/bin/panic.rs:30:13
Frame 12: foo @ 0x00000176
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/bin/panic.rs:24:2
Frame 13: __cortex_m_rt_main @ 0x000002d6
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/bin/panic.rs:13:5
Frame 14: __cortex_m_rt_main_trampoline @ 0x0000018a
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/bin/panic.rs:9:1
Frame 15: memmove @ 0x0000013c
Frame 16: memmove @ 0x0000013c
Error: Semihosting indicates exit with failure code: 0x020023 (131107)
$
```

Ignoring the very annoying log output (I don't think my Armv7m core being locked up is an ERROR, as ERRORs should only indicate probe-rs bugs not valid real-world state), we can see the tools runs to completion, notes that a semihosting exit occurred and prints a target backtrace.

noppej, 2024-03-13:
@thejpster Thank you for taking the time to clarify. I think we have a couple of things going on here, so we may have to pull it apart.
1. "an ERROR should only indicate probe-rs bugs" ... this is probably a UX choice which I'm not an expert on. In general, wherever I wrote code that indicates something that prevents the user action from completing successfully, I've used the term ERROR. If we can get consensus that this is wrong, then it will require a lot of message cleanup in probe-rs. I don't have a strong opinion either way, and just stating my current understanding.
2. In your log output, the HardFault handler is recognized, and unwinds the way (mostly) I would expect to see. The only surprise are the lines with `WARN probe_rs::debug::unit_info: Unsupported DW_AT_abstract_origin value: DebugInfoRef(DebugInfoOffset(7798))`. This suggests a previously un-encountered DWARF attribute in the binary, and we should probably implement support for it.
3. The OP has a different unwind / stacktrace. If you both have the same `probe-rs` version, and the same probe+target hardware, then that is almost certainly caused by differences in the binary and debug info data. In my experience, this can be caused by either different dependency versions or different rust compiler versions. I would suggest a diff on the `cargo tree output` and possibly a diff on the objdump of the binaries. I am happy to help with the latter if needed.

jonathanpallant, 2024-03-15:
[panic.zip](https://github.com/probe-rs/probe-rs/files/14618866/panic.zip)

Here's a binary, and here's what I see:

```console
$ probe-rs run --chip nRF52840_xxAA ./panic --allow-erase-all
 WARN probe_rs::config::sequences::nrf52: Core is locked. Erase procedure will be started to unlock it.
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
      Erasing ✔ [00:00:00] [######################] 8.00 KiB/8.00 KiB @ 28.36 KiB/s (eta 0s )
  Programming ✔ [00:00:00] [######################] 8.00 KiB/8.00 KiB @ 23.66 KiB/s (eta 0s )    Finished in 0.651s
ERROR panicked at src/bin/panic.rs:30:13:
index out of bounds: the len is 3 but the index is 3
└─ panic_probe::print_defmt::print @ /Users/jonathan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:104
<lvl> `dk::fail()` called; exiting ...
└─ dk::fail @ /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/boards/dk/src/lib.rs:345
 WARN probe_rs::debug::unit_info: Unsupported DW_AT_abstract_origin value: DebugInfoRef(DebugInfoOffset(7798))
 WARN probe_rs::debug::unit_info: Unsupported DW_AT_abstract_origin value: DebugInfoRef(DebugInfoOffset(7761))
 WARN probe_rs::debug::unit_info: Unsupported DW_AT_abstract_origin value: DebugInfoRef(DebugInfoOffset(7824))
 WARN probe_rs::debug::debug_info: UNWIND: Error while checking for exception context. The stack trace will not include the calling frames. : Register error: No value for Return Address register. Please report this as a bug.
 WARN probe_rs::debug::debug_info: UNWIND: Error while checking for exception context. The stack trace will not include the calling frames. : Register error: No value for Return Address register. Please report this as a bug.
Frame 0: fail @ 0x00001300
       /Users/jonathan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-semihosting-0.5.0/src/lib.rs:201:13
Frame 1: __cortex_m_rt_HardFault @ 0x0000169e inline
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/lib.rs:12:5
Frame 2: __cortex_m_rt_HardFault_trampoline @ 0x000000000000169a
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/lib.rs:10:1
Frame 3: "HardFault handler. Cause: Escalated UsageFault (Undefined instruction)." @ 0x0000169e
Frame 4: __udf @ 0x00001528 inline
       ./asm/lib.rs:48:1
Frame 5: __udf @ 0x0000000000001528
       ./asm/lib.rs:51:17
Frame 6: udf @ 0x00001514
       /Users/jonathan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-0.7.7/src/asm.rs:43:5
Frame 7: hard_fault @ 0x00001506
       /Users/jonathan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:86:5
Frame 8: panic @ 0x000014d4
       /Users/jonathan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:54:9
Frame 9: panic_fmt @ 0x00000342
       /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/panicking.rs:72:14
Frame 10: panic_bounds_check @ 0x000003f6
       /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/panicking.rs:190:5
Frame 11: bar @ 0x00000180
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/bin/panic.rs:30:13
Frame 12: foo @ 0x00000176
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/bin/panic.rs:24:2
Frame 13: __cortex_m_rt_main @ 0x000002d6
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/bin/panic.rs:13:5
Frame 14: __cortex_m_rt_main_trampoline @ 0x0000018a
       /Users/jonathan/Documents/ferrous-systems/rust-exercises/nrf52-code/radio-app/src/bin/panic.rs:9:1
Frame 15: memmove @ 0x0000013c
Frame 16: memmove @ 0x0000013c
Error: Semihosting indicates exit with failure code: 0x020023 (131107)
$ probe-rs --version
probe-rs 0.23.0
git commit: crates.io
```