probe-rs/probe-rs #2278: ESP32 Pros3 using EspJtag

bugadani, 2024-03-11:
### Discussed in https://github.com/probe-rs/probe-rs/discussions/2272

<div type='discussions-op-text'>

<sup>Originally posted by **Hecatron** March  9, 2024</sup>
I've recently tried hooking up one of the boards which is an ESP32 S3
https://esp32s3.com/pros3.html
I figured I'd put this here for anyone else trying to get this to work with probe-rs

I've placed it into boot mode to avoid circuit python booting up
(by holding down the boot button while pressing reset)

Now `probe-rs list` seems to work under windows
```
probe-rs list
The following debug probes were found:
[0]: ESP JTAG (VID: 303a, PID: 1001, Serial: 84:F7:03:C0:5B:7C, EspJtag)
```

But when trying `probe-rs info` or `probe-rs reset` under windows this results in
```
ERROR nusb::platform::windows_winusb::enumeration: Failed to get DeviceInterfaceGUIDs from registry: The system cannot find the file specified. (os error 2)
Error: Failed to open the debug probe.
```

This looks very similar to this issue
https://github.com/probe-rs/probe-rs/issues/2237
Tried switching winusb driver across on zadig for interface 0 but that didn't help
Tried switching the second interface, interface 2 across I think that's now working `probe-rs info` gives
```
❯ probe-rs info
Probing target via JTAG

 WARN probe_rs::probe::espusbjtag: More than one TAP detected, defaulting to tap0
No DAP interface was found on the connected probe. ARM-specific information cannot be printed.
Error while reading RISC-V info: Connected target is not a RISC-V device.
Xtensa Chip:
  IDCODE: 00120034e5
    Version:      1
    Part:         8195
    Manufacturer: 626 (Tensilica)

Probing target via SWD

Error identifying target using protocol SWD: Probe does not support SWD
```
but I think that's normal, will try building and flashing some code next
</div>

bugadani, 2024-03-11:
Hmm I was hoping github would transfer the comments, the relevan one is https://github.com/probe-rs/probe-rs/discussions/2272#discussioncomment-8737255

bugadani, 2024-03-11:
There are a few separate issues here:
 - `probe-rs erase` does not work
 - We don't document which USB interface is the JTAG one. I'm not sure we have to, as it's not probe-rs specific, but we could.
 - The board in question has 16MB QSPI Flash which we should be able to work with, but apparently something isn't quite right with the flashing algo?

Hecatron, 2024-03-11:
The code example I'm working off of is located here
https://github.com/Hecatron/Embedded.Rust.Examples/tree/master/src/ESP32-ProS3/general/blinky1-nostd-probe-rs

Although it won't actually flash an led, just turn the pin on and off due to the board using a ws2812 led

azais-corentin, 2026-01-14:
Trying it again today, flashing somehow worked (on the same version 0.30.0 commit eab9c5e) but the logs didn't work properly.
Installing probe-rs from git fixed all the issues, except for the programming speed:
```
   Compiling rcpc v0.1.0 (/workspaces/rcpc)
    Finished `release` profile [optimized + debuginfo] target(s) in 2.51s
     Running `probe-rs run --chip=esp32s3 --preverify --always-print-stacktrace --no-location --catch-hardfault target/xtensa-esp32s3-none-elf/release/rcpc`
    Verifying ✔ 100% [####################] 324.67 KiB @ 472.66 KiB/s (took 1s)                                                                                                                                                                                                       
      Erasing ✔ 100% [####################] 576.00 KiB @ 220.77 KiB/s (took 3s)
  Programming ✔ 100% [####################] 324.67 KiB @  28.82 KiB/s (took 11s)                                                                                                                                                                                                           Finished in 15.34s
[INFO ] Embassy initialized!
[INFO ] Hello world!
[INFO ] Hello world!
[INFO ] Hello world!
[INFO ] Hello world!
```