probe-rs/probe-rs #2280: ESP-WROVER-KIT - Stepping into / Stepping Over code errors

Hecatron, 2024-03-11:
Hi,
I've recently tried debugging with an ESP-WROVER-KIT device
this is a plain ESP32 with a UART/JTAG IC

I don't expect this to work currently based on what's mentioned in the other issues also since ESP support is only a month old.
But I figured I'd put any results here

  * Basic blinky app built
  * Flashed with `cargo espflash flash`. This is done via the COM / Serial port
  * Then debugging attempted via probe-rs by connecting to the JTAG interface which is set to winusb driver "WinUSB (v6.1.7600.16385)" by zadig

I've put an example here

  * https://github.com/Hecatron/Embedded.Rust.Examples/tree/master/src/ESP32-Wrover-Kit/general/blinky1-nostd-probe-rs

For Info I'm using Windows 10 as the host

## Probe List

Output from probe-rs list
```
The following debug probes were found:
[0]: Dual RS232-HS (VID: 0403, PID: 6010, FTDI)
```

## Probe Info

Output from probe-rs info
```
 WARN probe_rs::probe::ftdi::ftdaye: interface 1 does not match requested interface A
Probing target via JTAG

 WARN probe_rs::probe::ftdi: More than one TAP detected, defaulting to tap0
ERROR probe_rs::probe::arm_debug_interface: Unexpected DAP response: 0
ERROR probe_rs::probe::arm_debug_interface: Unexpected DAP response: 4
ERROR probe_rs::probe::arm_debug_interface: Unexpected DAP response: 4
ERROR probe_rs::probe::arm_debug_interface: Unexpected DAP response: 4
ERROR probe_rs::probe::arm_debug_interface: Unexpected DAP response: 5
ERROR probe_rs::probe::arm_debug_interface: Unexpected DAP response: 5
ERROR probe_rs::probe::arm_debug_interface: Unexpected DAP response: 5
ERROR probe_rs::probe::arm_debug_interface: Unexpected DAP response: 5
ERROR probe_rs::probe::arm_debug_interface: Unexpected DAP response: 5
ERROR probe_rs::probe::arm_debug_interface: Unexpected DAP response: 5
ERROR probe_rs::probe::arm_debug_interface: Unexpected DAP response: 5
ERROR probe_rs::probe::arm_debug_interface: Unexpected DAP response: 5
Error showing ARM chip information: An error occurred in the communication with an access port or debug port.

Caused by:
    Target device did not respond to request.
Error while reading RISC-V info: Connected target is not a RISC-V device.
Xtensa Chip:
  IDCODE: 00120034e5
    Version:      1
    Part:         8195
    Manufacturer: 626 (Tensilica)
```

## VSCode Attach

If I avoid using rtt logging, then I can set a breakpoint and have it stop at it
Such as where the led is toggled.
Or I can pause the execution after attaching.

Attempting to Step over gives
```
ERROR probe_rs::debug::debug_step: Error during step (OverStatement): Function Some("__xtensa_lx_rt_main") is marked as `noreturn`. Cannot step out of this function.
```

Attempting to Step into gives
```
ERROR probe_rs::cmd::dap_server::server::startup: probe-rs-debugger session ended: Error executing request. probe-rs-debug: CONSOLE: DAP Protocol server exiting
Debug Adapter terminated unexpectedly with an error: Other(Error executing request. Caused by: Unexpected error during stepping :Error using the probe)
```

bugadani, 2024-03-11:
Thanks! Some of this may be related to https://github.com/probe-rs/probe-rs/issues/2180 or at least some of the symptoms may change depending on that issue's resolution.

noppej, 2024-03-11:
@bugadani , you are correct. The stepping issue is related and I am busy with final testing on a fix, so hopefully we can alleviate this soon. 

The `probe-rs info` is a different problem, which I may not be able to help with.

bugadani, 2024-03-11:
`ERROR probe_rs::probe::arm_debug_interface: Unexpected DAP response: 5` is just some BS from the ARM interface detection, not relevant to ESP32

noppej, 2024-03-11:
Probably worth splitting this into two different issues???

closed 2026-02-08 by bugadani