probe-rs/probe-rs #2285: Temporarily attaching to cores in low-level interface

maxd-nordic, 2024-03-12:
AFAICT, users can currently choose between opening a session and a low-level interface.
In our application, we would need to access both in some way. This has to do with how our chips work under the hood.

Ideally, I would be able to attach to a target from the low-level interface and also be able to unattach if needed. Could you guide me how to do this?

Alternatively, I could also imagine switching between a session and the low-level interface.

Yatekii, 2024-03-24:
The idea is that the session encapsulates all features in a higher level interface and the lower level interface is not really required. Especially for new chip support. What exactly do you need to do?

maxd-nordic, 2024-03-25:
We have a device family that needs both low-level and high-level interfaces at the same time for unlocking. There is a special access port that the high-level interface cannot support and the AHB access port needs to be used before a reset is issued to retain access. Part of the implementation is also secret, so we cannot immediately add target-specific support to probe-rs.

maxd-nordic, 2024-03-26:
Maybe something like this? https://github.com/probe-rs/probe-rs/pull/2321