probe-rs/probe-rs #2299: probe-rs is much slower than OpenOCD in download speed

zh522130, 2024-03-17:
**Is your feature request related to a problem? Please describe.**

I used the STM32H743VGT6 and tested the download speeds of probe-rs, Keil, and OpenOCD with the same firmware. The test results show that the speeds of probe-rs and Keil are similar. For probe-rs, I used the command

probe-rs download .\STM32H743VIT6.hex --format hex --chip STM32H743VGTx --protocol swd --speed 10000

The results showed a total of 512KB erased and 507KB programmed, with an erasure speed of 71.35KB/s and a programming speed of 65.46KB/s, totaling 16.922 seconds.
OpenOCD was tested with the command

openocd -f interface/cmsis-dap.cfg -f target/stm32h7x.cfg -c "adapter_khz 10000" -c "program D:\STM32H743VIT6.hex verify reset exit" -d

the final results were calculated from the log messages, showing an erasure time of 3.413s at a speed of 150KB/s and a programming time of 4.406s at a speed of 115KB/s. I noticed through the timing prints that OpenOCD constantly calls the target_run_flash_async_algorithm function, caching the download data in the MCU. But probe-rs also enabled double buffering, I located in the probe-rs source code and removed std::thread::sleep(Duration::from_millis(1)); from wait_for_completion, which could increase the download speed to 75KB/s. However, it is still significantly slower compared to OpenOCD.

**Describe the solution you'd like**

Have you considered compatibility with OpenOCD's flash download algorithm or other methods to increase speed?

**Describe alternatives you've considered**

**Additional context**

Yatekii, 2024-03-23:
This is rather odd. probe-rs should easily reach the same speeds. Can you try and run the benchmark command? :)

zh522130, 2024-03-25:
> This is rather odd. probe-rs should easily reach the same speeds. Can you try and run the benchmark command? :)这很奇怪。 probe-rs 应该很容易达到相同的速度。您可以尝试运行基准测试命令吗？ :)

the benchmark result：
```
probe-rs benchmark --address 0x20000000 --chip STM32H743VGTx --speed 10000

Probe: Probe type CMSIS-DAP, debug interface not specified, target chip STM32H743VGTx

Test: Speed 10000, Word size 32bit, Data length 4 bytes, Number of iterations 5
Results: Read: 30944.29 bytes/s Std Dev 511.57, Write: 4068965.52 bytes/s Std Dev 1862068.97

Test: Speed 10000, Word size 32bit, Data length 32 bytes, Number of iterations 5
Results: Read: 104930.39 bytes/s Std Dev 28851.69, Write: 110136.56 bytes/s Std Dev 17480.59

Test: Speed 10000, Word size 32bit, Data length 128 bytes, Number of iterations 5
Results: Read: 175022.53 bytes/s Std Dev 29742.62, Write: 190013.51 bytes/s Std Dev 9636.99

Test: Speed 10000, Word size 32bit, Data length 2048 bytes, Number of iterations 5
Results: Read: 241678.32 bytes/s Std Dev 5935.73, Write: 258893.71 bytes/s Std Dev 16412.64

Test: Speed 10000, Word size 32bit, Data length 32768 bytes, Number of iterations 5
Results: Read: 254849.03 bytes/s Std Dev 2569.25, Write: 265996.05 bytes/s Std Dev 3259.28
```

Yatekii, 2024-03-27:
Would you mind also attaching a log on the DEBUG level when you flash with probe-rs? Maybe double buffering is not used. Because the raw probe speed is fine. So it should only be limited by flash speed. Ofc it can happen that the flash algo is imperformant in which case it's not strictly our fault. In that case we would have to write our own.

zh522130, 2024-03-28:
I have executed the code in debug mode and made sure that double buffering is enabled. To check whether double buffering is turned on in the code, it not only checks the disable flag but also assesses the size of RAM. The STM32H7 satisfies the requirements for enabling double buffer tuning, and through debugging, I have indeed witnessed the code running through the double buffer branch. I suspect that the variation in erase speeds may stem from differences in the download algorithms, but it appears that the programming speed might be linked to the logic within probe-rs. In contrast, enabling logging in OpenOCD does not noticeably impact the download speed. When testing the download speed on both a dual-core A7 1GHz ARM and an AMD 7840H processor platform, the difference is not significant—with the same 507KB firmware, the time difference is around 1 second. However, with probe-rs, turning on logging even on the AMD 7840H platform significantly slows down the download speed, which is less than half when logging is disabled. Similarly, with logging turned off, the benchmark results for probe-rs on the dual-core A7 platform are as follows: 

```
 probe-rs benchmark --address 0x20000000 --chip STM32H743VGTx --speed 10000

probe-rs benchmark --address 0x20000000 --chip STM32H743VGTx --speed 10000
Probe: Probe type CMSIS-DAP, debug interface not specified, target chip STM32H743VGTx

Test: Speed 10000, Word size 32bit, Data length 4 bytes, Number of iterations 5
Results: Read: 10461.04 bytes/s Std Dev 1084.36, Write: 553454.21 bytes/s Std Dev 70311.43

Test: Speed 10000, Word size 32bit, Data length 32 bytes, Number of iterations 5
Results: Read: 50189.17 bytes/s Std Dev 10040.47, Write: 38842.05 bytes/s Std Dev 5974.69

Test: Speed 10000, Word size 32bit, Data length 128 bytes, Number of iterations 5
Results: Read: 94418.22 bytes/s Std Dev 4258.57, Write: 88512.52 bytes/s Std Dev 8958.54

Test: Speed 10000, Word size 32bit, Data length 2048 bytes, Number of iterations 5
Results: Read: 129597.29 bytes/s Std Dev 13375.66, Write: 123760.49 bytes/s Std Dev 14233.32

Test: Speed 10000, Word size 32bit, Data length 32768 bytes, Number of iterations 5
Results: Read: 138249.18 bytes/s Std Dev 2835.41, Write: 131775.39 bytes/s Std Dev 1910.84
```

Yatekii, 2024-03-28:
Yeah, formatting values in Rust is not optimized for speed.
There was some optimizations WRT logging recently. What version of probe-rs do you run?

Also the second benchmark you posted is half the speed of the first one, yet you say you disabled logging in the second one?

If you believe it is the logging, could you disable the logging for a flash speed benchmark? :)

zh522130, 2024-03-28:
I'm very sorry if my answer caused any confusion. I conducted the tests using version 0.23.0. 

The first comparison was to test the impact of debug logging on download performance; if logging is enabled, the speed drops to about half of what it is with logging turned off. 

The second comparison looked at the effect of different processor performances on download speed, using the benchmark command included with probe-rs, which does not enable debug logging. The benchmark for the 7840H is as follows,
probe-rs benchmark --address 0x20000000 --chip STM32H743VGTx --speed 10000
```
Probe: Probe type CMSIS-DAP, debug interface not specified, target chip STM32H743VGTx

Test: Speed 10000, Word size 32bit, Data length 4 bytes, Number of iterations 5
Results: Read: 30944.29 bytes/s Std Dev 511.57, Write: 4068965.52 bytes/s Std Dev 1862068.97

Test: Speed 10000, Word size 32bit, Data length 32 bytes, Number of iterations 5
Results: Read: 104930.39 bytes/s Std Dev 28851.69, Write: 110136.56 bytes/s Std Dev 17480.59

Test: Speed 10000, Word size 32bit, Data length 128 bytes, Number of iterations 5
Results: Read: 175022.53 bytes/s Std Dev 29742.62, Write: 190013.51 bytes/s Std Dev 9636.99

Test: Speed 10000, Word size 32bit, Data length 2048 bytes, Number of iterations 5
Results: Read: 241678.32 bytes/s Std Dev 5935.73, Write: 258893.71 bytes/s Std Dev 16412.64

Test: Speed 10000, Word size 32bit, Data length 32768 bytes, Number of iterations 5
Results: Read: 254849.03 bytes/s Std Dev 2569.25, Write: 265996.05 bytes/s Std Dev 3259.28
```
and the benchmark for the dual-core 1GHz A7 is as follows: 
```
 probe-rs benchmark --address 0x20000000 --chip STM32H743VGTx --speed 10000

probe-rs benchmark --address 0x20000000 --chip STM32H743VGTx --speed 10000
Probe: Probe type CMSIS-DAP, debug interface not specified, target chip STM32H743VGTx

Test: Speed 10000, Word size 32bit, Data length 4 bytes, Number of iterations 5
Results: Read: 10461.04 bytes/s Std Dev 1084.36, Write: 553454.21 bytes/s Std Dev 70311.43

Test: Speed 10000, Word size 32bit, Data length 32 bytes, Number of iterations 5
Results: Read: 50189.17 bytes/s Std Dev 10040.47, Write: 38842.05 bytes/s Std Dev 5974.69

Test: Speed 10000, Word size 32bit, Data length 128 bytes, Number of iterations 5
Results: Read: 94418.22 bytes/s Std Dev 4258.57, Write: 88512.52 bytes/s Std Dev 8958.54

Test: Speed 10000, Word size 32bit, Data length 2048 bytes, Number of iterations 5
Results: Read: 129597.29 bytes/s Std Dev 13375.66, Write: 123760.49 bytes/s Std Dev 14233.32

Test: Speed 10000, Word size 32bit, Data length 32768 bytes, Number of iterations 5
Results: Read: 138249.18 bytes/s Std Dev 2835.41, Write: 131775.39 bytes/s Std Dev 1910.84
```
The difference in results is significant.

 I also tested the download speeds with openocd on these two processors. For a 507KB firmware, the total download time on the dual-core A7 is about 1 second slower than on the 7840H. Therefore, I believe that probe-rs has higher processor performance requirements than openocd, or it may still need optimization.

Yatekii, 2024-03-28:
Thanks a lot for the clarification! That's actually terrible! The host should never be the bottleneck! I will run some tests and try find the culprit :)

kaidegit, 2024-11-22:
Hi, how to disable log, just not use --log-file flag?

My test result is much more worse than yours. I use AMD 5700G with Windows 11 and Allwinner H618 with the ubuntu orangepi offered, use the master version and build with --release. and the results are here.

H618
```
orangepi@orangepizero2w:~$ ./probe-rs --version
probe-rs 0.24.0 (git commit: v0.1.0-266-gb15caaee)

orangepi@orangepizero2w:~$ ./probe-rs download --chip nRF52832_xxAA --binary-format hex --speed 1000 firmware/xxx.hex
      Erasing ✔ 100% [####################] 320.00 KiB @  33.09 KiB/s (took 10s)
  Programming ✔ 100% [####################] 316.00 KiB @   6.96 KiB/s (took 45s)
Finished in 46.62s

orangepi@orangepizero2w:~$ ./probe-rs benchmark --chip nRF52832_xxAA --speed 1000 --address 0x20000000
Probe: Probe type CMSIS-DAP, debug interface not specified, target chip nRF52832_xxAA

Test: Speed 1000, Word size 32bit, Data length 4 bytes, Number of iterations 5
Results: Read: 1000.78 bytes/s Std Dev 0.26, Write: 1880219.32 bytes/s Std Dev 460496.62

Test: Speed 1000, Word size 32bit, Data length 32 bytes, Number of iterations 5
Results: Read: 3999.00 bytes/s Std Dev 2.38, Write: 3997.51 bytes/s Std Dev 5.63

Test: Speed 1000, Word size 32bit, Data length 128 bytes, Number of iterations 5
Results: Read: 8000.19 bytes/s Std Dev 0.41, Write: 7996.35 bytes/s Std Dev 9.11

Test: Speed 1000, Word size 32bit, Data length 2048 bytes, Number of iterations 5
Results: Read: 12799.95 bytes/s Std Dev 0.13, Write: 12800.32 bytes/s Std Dev 0.20

Test: Speed 1000, Word size 32bit, Data length 32768 bytes, Number of iterations 5
Results: Read: 12799.96 bytes/s Std Dev 0.19, Write: 12800.30 bytes/s Std Dev 0.28
```

5700G
```
PS C:\Users\xxx\Downloads> C:\Users\xxx\Documents\GitHub\probe-rs\target\release\probe-rs.exe --version
probe-rs 0.24.0 (git commit: v0.1.0-266-gb15caaee)

PS C:\Users\xxx\Downloads> C:\Users\xxx\Documents\GitHub\probe-rs\target\release\probe-rs.exe download --chip nRF52832_xxAA --binary-format hex "xxx.hex"
      Erasing ✔ 100% [####################] 320.00 KiB @ 210.99 KiB/s (took 2s)
  Programming ✔ 100% [####################] 316.00 KiB @  23.33 KiB/s (took 14s)
     Finished in 13.86s

PS C:\Users\xxx\Downloads> C:\Users\xxx\Documents\GitHub\probe-rs\target\release\probe-rs.exe benchmark --chip nRF52832_xxAA --speed 1000 --address 0x20000000
Probe: Probe type CMSIS-DAP, debug interface not specified, target chip nRF52832_xxAA

Test: Speed 1000, Word size 32bit, Data length 4 bytes, Number of iterations 5
Results: Read: 5976.26 bytes/s Std Dev 36.19, Write: 4706959.71 bytes/s Std Dev 2198473.65

Test: Speed 1000, Word size 32bit, Data length 32 bytes, Number of iterations 5
Results: Read: 22215.48 bytes/s Std Dev 331.04, Write: 22043.68 bytes/s Std Dev 291.24

Test: Speed 1000, Word size 32bit, Data length 128 bytes, Number of iterations 5
Results: Read: 27561.87 bytes/s Std Dev 804.36, Write: 27785.35 bytes/s Std Dev 206.51

Test: Speed 1000, Word size 32bit, Data length 2048 bytes, Number of iterations 5
Results: Read: 30566.71 bytes/s Std Dev 134.70, Write: 30569.79 bytes/s Std Dev 97.80

Test: Speed 1000, Word size 32bit, Data length 32768 bytes, Number of iterations 5
Results: Read: 30662.78 bytes/s Std Dev 6.93, Write: 30608.35 bytes/s Std Dev 19.50
```

What's more, I don't notice that the h618 cpu usage would go full via `htop` when flashing, it always lower than 10% i think.

zh522130, 2024-11-22:
> Hi, how to disable log, just not use --log-file flag?
> 

Your printout indicates that logging isn't enabled. I haven't tested probe-rs for a while, but I used it in library form before. If the official binary includes a logging framework, you can usually enable logging by using `export RUST_LOG=debug` or another log level. By the way, the test results also depend on the speed of the downloader, so make sure the downloader isn't a bottleneck.

Yatekii, 2024-11-22:
What is your commit based on? Did you try an up2date master?

Also, can you `time` the run with pyocd? It combines the numbers for erase and programming, so ofc the average speed would be in the middle between erasing and programming.

`nRF52832_xxAA` should be super fast to program - I would rule out issues with the flash algorithm.

`--speed 1000` of course makes this super slow. 1MHz is nothing. Try `10000` or even higher.

What puts me off a bit is that the benchmark has such high numbers for single byte write in both cases. I would expect this to be slowest. @9names do you have any insight here? Just an anomaly?

kaidegit, 2024-11-22:
>What is your commit based on? Did you try an up 2 date master?

master version at that time

> nRF52832_xxAA should be super fast to program - I would rule out issues with the flash algorithm.
--speed 1000 of course makes this super slow. 1MHz is nothing. Try 10000 or even higher.

I have tested in faster speed, like 4M and 10M. But no quality improvement.

>Also, can you time the run with pyocd? It combines the numbers for erase and programming, so ofc the average speed would be in the middle between erasing and programming.

The number at the beginning of the log should be the time in ms. I would test it again on Monday.

kaidegit, 2024-11-25:
pyocd & 1M 
```
orangepi@orangepizero2w:~$ pyocd flash -t nrf52 -f 1M -Lpyocd.flash.flash.trace=debug firmware/xxx.hex
0003024 I Loading /home/orangepi/firmware/xxx.hex [load_cmd]
0004516 D algo init and load to 0x20000000 [flash]
0004657 D algo call init(addr=0, clock=0, op=3) [flash]
0004683 D init result = 0 [flash]
0004709 D call compute crc(20002000, 4f) [flash]
[                                                  ]   0%
0004835 D algo call init(addr=0, clock=0, op=1) [flash]
0004860 D init result = 0 [flash]
0004861 D call erase_sector(0) [flash]
0004882 D erase_sector result = 0 [flash]
[                                                  ]   0%
0004883 D call erase_sector(1000) [flash]
0004903 D erase_sector result = 0 [flash]
[                                                  ]   0%
0004904 D call erase_sector(2000) [flash]
0004924 D erase_sector result = 0 [flash]
[                                                  ]   1%
0004925 D call erase_sector(3000) [flash]
0004945 D erase_sector result = 0 [flash]
[                                                  ]   1%
0004946 D call erase_sector(4000) [flash]
0004966 D erase_sector result = 0 [flash]
[                                                  ]   1%
0004967 D call erase_sector(5000) [flash]
0004987 D erase_sector result = 0 [flash]
[                                                  ]   1%
0004988 D call erase_sector(6000) [flash]
0005008 D erase_sector result = 0 [flash]
[                                                  ]   2%
0005009 D call erase_sector(7000) [flash]
0005029 D erase_sector result = 0 [flash]
[                                                  ]   2%
0005030 D call erase_sector(8000) [flash]
0005050 D erase_sector result = 0 [flash]
[                                                  ]   2%
0005051 D call erase_sector(9000) [flash]
0005071 D erase_sector result = 0 [flash]
[=                                                 ]   2%
0005071 D call erase_sector(a000) [flash]
0005092 D erase_sector result = 0 [flash]
[=                                                 ]   2%
0005093 D call erase_sector(b000) [flash]
0005113 D erase_sector result = 0 [flash]
[=                                                 ]   3%
0005114 D call erase_sector(c000) [flash]
0005134 D erase_sector result = 0 [flash]
[=                                                 ]   3%
0005135 D call erase_sector(d000) [flash]
0005155 D erase_sector result = 0 [flash]
[=                                                 ]   3%
0005156 D call erase_sector(e000) [flash]
0005176 D erase_sector result = 0 [flash]
[=                                                 ]   3%
0005177 D call erase_sector(f000) [flash]
0005197 D erase_sector result = 0 [flash]
[=                                                 ]   3%
0005198 D call erase_sector(10000) [flash]
0005218 D erase_sector result = 0 [flash]
[=                                                 ]   4%
0005219 D call erase_sector(11000) [flash]
0005239 D erase_sector result = 0 [flash]
[=                                                 ]   4%
0005240 D call erase_sector(12000) [flash]
0005260 D erase_sector result = 0 [flash]
[==                                                ]   4%
0005261 D call erase_sector(13000) [flash]
0005281 D erase_sector result = 0 [flash]
[... 810 lines cut ...]

```
PS C:\Users\xxx\Downloads> C:\Users\xxx\Documents\GitHub\probe-rs\target\release\probe-rs.exe download --chip nRF52832_xxAA --binary-format hex  --speed 10000 "C:\Users\yekai\Downloads\xxx.hex"
      Erasing ✔ 100% [####################] 320.00 KiB @ 594.60 KiB/s (took 1s)
  Programming ✔ 100% [####################] 316.00 KiB @  48.62 KiB/s (took 6s)
     Finished in 6.63s

PS C:\Users\xxx\Downloads> C:\Users\xxx\Documents\GitHub\probe-rs\target\release\probe-rs.exe benchmark --chip nRF52832_xxAA --speed 10000 --address 0x20000000
Probe: Probe type CMSIS-DAP, debug interface not specified, target chip nRF52832_xxAA

Test: Speed 10000, Word size 32bit, Data length 4 bytes, Number of iterations 5
Results: Read: 24898.10 bytes/s Std Dev 1592.75, Write: 4477871.15 bytes/s Std Dev 2261876.30

Test: Speed 10000, Word size 32bit, Data length 32 bytes, Number of iterations 5
Results: Read: 98782.89 bytes/s Std Dev 7059.92, Write: 97373.26 bytes/s Std Dev 8019.72

Test: Speed 10000, Word size 32bit, Data length 128 bytes, Number of iterations 5
Results: Read: 138975.49 bytes/s Std Dev 8681.10, Write: 139316.00 bytes/s Std Dev 7325.05

Test: Speed 10000, Word size 32bit, Data length 2048 bytes, Number of iterations 5
Results: Read: 173595.66 bytes/s Std Dev 1168.27, Write: 166349.41 bytes/s Std Dev 2633.06

Test: Speed 10000, Word size 32bit, Data length 32768 bytes, Number of iterations 5
Results: Read: 172650.57 bytes/s Std Dev 449.05, Write: 165248.14 bytes/s Std Dev 334.75
```

Yatekii, 2024-11-25:
I believe the speed discrepancy between the two systems points to logging to an SD card potentially being the culprit? What are the setups of the two systems?

kaidegit, 2024-11-25:
kioxia 32g blue microsd card on orangpi and TiPlus5000 nvme on windows machine. If sd card cause this, why pyocd can have a better performance than probe-rs? Do pyocd cached the fw into ram and probe-rs read it every time before write page?

Yatekii, 2024-11-28:
We always log to a file on disk :) That sloes things down considerably with an SD card. can you try point the log file to a tempfs in RAM with `--log-file`? :)

kaidegit, 2024-11-28:
I find that /tmp always be full on this system, and no improvement on downloading...I would try another system later.

kaidegit, 2024-11-28:
I tried armbian and ubuntu and the result is quite the same. And /tmp would be filled after one time download...(my opi has only 1g ram and 512M /tmp)

<details>
<[summary]>log</summary>

```
orangepi@orangepizero2w:~$ python3 ./xxx/relay.py
orangepi@orangepizero2w:~$ df -Th
Filesystem     Type      Size  Used Avail Use% Mounted on
udev           devtmpfs  389M     0  389M   0% /dev
tmpfs          tmpfs      92M  1.9M   90M   3% /run
/dev/mmcblk0p1 ext4       57G  1.5G   55G   3% /
tmpfs          tmpfs     460M     0  460M   0% /dev/shm
tmpfs          tmpfs     5.0M     0  5.0M   0% /run/lock
tmpfs          tmpfs     460M     0  460M   0% /tmp
/dev/zram1     ext4       47M  1.2M   43M   3% /var/log
tmpfs          tmpfs      92M     0   92M   0% /run/user/1000
orangepi@orangepizero2w:~$ cp firmware/xxx.hex /tmp/xxx.hex
orangepi@orangepizero2w:~$ df -Th
Filesystem     Type      Size  Used Avail Use% Mounted on
udev           devtmpfs  389M     0  389M   0% /dev
tmpfs          tmpfs      92M  1.9M   90M   3% /run
/dev/mmcblk0p1 ext4       57G  1.5G   55G   3% /
tmpfs          tmpfs     460M     0  460M   0% /dev/shm
tmpfs          tmpfs     5.0M     0  5.0M   0% /run/lock
tmpfs          tmpfs     460M  852K  459M   1% /tmp
/dev/zram1     ext4       47M  1.2M   43M   3% /var/log
tmpfs          tmpfs      92M     0   92M   0% /run/user/1000
orangepi@orangepizero2w:~$ sudo ./probe-rs download --chip nRF52832_xxAA --binary-format hex --speed 10000 --log-file=/tmp/1.log /tmp/xxx.hex
[sudo] password for orangepi:
      Erasing ✔ 100% [####################] 320.00 KiB @  27.47 KiB/s (took 12s)
  Programming ✔ 100% [####################] 316.00 KiB @   6.24 KiB/s (took 51s)                                                                                                                                       Finished in 52.07s
orangepi@orangepizero2w:~$ df -Th
Filesystem     Type      Size  Used Avail Use% Mounted on
udev           devtmpfs  389M     0  389M   0% /dev
tmpfs          tmpfs      92M  1.9M   90M   3% /run
/dev/mmcblk0p1 ext4       57G  1.5G   55G   3% /
tmpfs          tmpfs     460M     0  460M   0% /dev/shm
tmpfs          tmpfs     5.0M     0  5.0M   0% /run/lock
tmpfs          tmpfs     460M  460M     0 100% /tmp
/dev/zram1     ext4       47M  1.2M   43M   3% /var/log
tmpfs          tmpfs      92M     0   92M   0% /run/user/1000
orangepi@orangepizero2w:~$

```
</details>

Yatekii, 2024-11-28:
I dont see how you put the logfile somewhere else than the SD card?

kaidegit, 2024-11-28:
This would not take effect? I see the log is generated here.

[image]

jwagen, 2025-02-07:
I am seeing a very simmilar isuse. I use a STlink V2 to connect to a STM32F205. I am using probe-rs v0.26.

I initaly used wsl and [usbipd](https://learn.microsoft.com/en-us/windows/wsl/connect-usb), and thought that might be the issue for me. But that is only partly the issue.
I tried `openocd` in wsl using usbipd passthrough. I don't know how much overhead usbipd has, but it looks to affect probe-rs the worst. Intrestingly CubeProgrammer is almost unaffected.

The probe-rs time is also quite unstable I have seen times up to 60s. v0.26 seems to work better than 0.25, where upload times was never below 30s.

Time to upload 289kB elf file.
|OS | Program |Time (s)|
|--|--|--|
|Windows 11 | STM32CubeProgrammer | 5.7  |
|Windows 11 | probe-rs | 12.59 |
|WSL ubuntu | STM32CubeProgrammer | 6.8 |
|WSL ubuntu | probe-rs | 26.39 |
|WSL ubuntu | openocd | 11.7 |

Probe-rs wsl usbipd
```
 probe-rs run --chip STM32F205VGTx Firmware.elf --speed 4600
      Erasing ✔ 100% [####################] 384.00 KiB @  62.46 KiB/s (took 6s)
  Programming ✔ 100% [####################] 289.00 KiB @  14.28 KiB/s (took 20s)
     Finished in 26.39s
```

Probe-rs Windows
```
probe-rs.exe run Firmware.elf --chip STM32F205VGTx --speed 4600
      Erasing ✔ 100% [####################] 384.00 KiB @  64.34 KiB/s (took 6s)
  Programming ✔ 100% [####################] 289.00 KiB @  43.78 KiB/s (took 7s)
     Finished in 12.59s
```

Openocd wsl usbipd
```
time openocd -f interface/stlink.cfg -f target/stm32f2x.cfg -c "adapter_khz 4000" -c "Firmware.elf verify reset exit"
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
DEPRECATED! use 'adapter speed' not 'adapter_khz'
adapter speed: 10000 kHz

Info : clock speed 10000 kHz
Info : STLINK V2J45S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.203162
Info : [stm32f2x.cpu] Cortex-M3 r2p0 processor detected
Info : [stm32f2x.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f2x.cpu on 3333
Info : Listening on port 3333 for gdb connections
[stm32f2x.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x0802e360 msp: 0x20020000
** Programming Started **
Info : device id = 0x200f6411
Info : flash size = 1024 KiB
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked

real    0m11.854s
user    0m0.159s
sys     0m0.309s
```

Windows STM32CubeProgrammer Windows
```
  09:24:52 : Opening and parsing file: Firmware.elf
  09:24:52 : Memory Programming ...
  09:24:52 :   File          : Firmware.elf
  09:24:52 :   Size          : 288.20 KB 
  09:24:52 :   Address       : 0x08040000 
  09:24:52 : Erasing memory corresponding to segment 0:
  09:24:52 : Erasing internal memory sectors [6 8]
  09:24:55 : Download in Progress:
  09:24:57 : File download complete
  09:24:57 : Time elapsed during download operation: 00:00:05.714
```

STM32CubeProgrammer wsl usbipd
```
 09:21:00 : Opening and parsing file: Firmware.elf
  09:21:00 : Memory Programming ...
  09:21:00 :   File          : Firmware.elf
  09:21:00 :   Size          : 288.20 KB 
  09:21:00 :   Address       : 0x08040000 
  09:21:00 : Erasing memory corresponding to segment 0:
  09:21:00 : Erasing internal memory sectors [6 8]
  09:21:03 : Download in Progress:
  09:21:06 : File download complete
  09:21:06 : Time elapsed during download operation: 00:00:06.878
```

Yatekii, 2025-02-07:
How are the times without usbipd? If probe-rs is mostly affected by it, I would guess we do not batch enough.

jwagen, 2025-02-07:
I updated the table in my previous comment to make it clearer. 

On with native usb on windows probe-rs takes 12.6 seconds and 26.4 on wsl with usbip. Compared with STM32CubeProgrammer which takes 5.6s and 6.8s for uploading the same file.

jwagen, 2025-02-07:
I checked with openocd on windows and I get about the same time as probe-rs when I run with native usb. The numbers are added to the tabel. 

So looks like wsl and usbip is to blame for the large difference I saw in programming speed, between probe-rs and openocd. ST's tool is still more than twice as fast, and almost the same speed on native and with usbip.

Yatekii, 2025-02-07:
ST-Links have some faster API for memory access. I am sure we can improve our code to get to similar speeds. I cannot tell off the bat where we can improve, but I am sure there is some tweaks. Would be nice to look at the disassembly of the STCubeProgrammer.