probe-rs/probe-rs #2301: Xtensa semihosting support

SergioGasquez, 2024-03-18:
Now that `semihosting` has a [PR for Xtensa support](https://github.com/taiki-e/semihosting/pull/9/files/20ff3fddb1bbbb9ecf009106d9a2298875467acb..e0900064c222c39eca2e9ae4a2d06a07513ee4b9), we would need to add support for Xtensa semihosting in `probe-rs` in order to be able to use `embedded-test` on Xtensa devices.

According to @t-moe, basically, we would need an Xtensa version of https://github.com/probe-rs/probe-rs/blob/8dc467f3e0a492d0644e183255098432fad4c0f4/probe-rs/src/architecture/riscv/mod.rs#L116C4-L162C2

bugadani, 2024-03-20:
Github must be having a bad day, this issue should have been closed by the merge. Thanks!

t-moe, 2024-03-20:
> Github must be having a bad day, this issue should have been closed by the merge. Thanks!

Not quite. Look at the target branch ;)

bugadani, 2024-03-20:
Shoot. Trap for your players...

PR #2303 (2024-03-20): Add Xtensa semihosting support
Adds semihosting support for Xtensa arch. 

Thanks @t-moe and @MabezDev for the help!

Closes #2301
closed 2024-03-26 by bugadani