probe-rs/probe-rs #2309: Cant flash ESP32-S2

SergioGasquez, 2024-03-22:
**Describe the bug**
Using an FTDI board (ESP-Prog) I cant flash an ESP32-S2

**To Reproduce**
Steps to reproduce the behavior:
1. `cargo generate esp-rs/esp-template --name probe-s2 -d mcu=esp32s2 -d advanced=false`
2. `cd probe-s2`
3. `sed -i 's/runner = "espflash flash --monitor"/runner = "probe-rs run --chip esp32s2"/g' .cargo/config.toml`
4. `cargo run`

**Expected behavior**
The app is properly flashed

**Stacktrace**
```
ERROR probe_rs::probe::common: Fewer IRs detected than TAPs
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An error with the usage of the probe occurred
    1: Invalid IR scan chain
```

**Desktop (please complete the following information):**

- Arch Linux

**Additional context**
Connections:
| ESP-Prog | ESP32-S2-DevKitC-1 v1.0 |
|:--------:|:-----------------------:|
|   VJTAG  |            5V           |
|    GND   |           GND           |
|    TMS   |            2            |
|    TCK   |            40           |
|    TDO   |            41           |
|    TDI   |            42           |

[`ESP32-S2-DevKitC-1 v1.0` Pinout](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/_images/esp32-s2-devkitc-1-v1-pinout.png)

bugadani, 2024-03-22:
Sorry @SergioGasquez but I can't immediately reproduce this issue:

```
     Running `probe-rs run --chip esp32s2 target\xtensa-esp32s2-none-elf\debug\probe-s2`
      Erasing ✔ [00:00:00] [###################################################################################################################################################################] 192.00 KiB/192.00 KiB @ 259.93 KiB/s (eta 0s )
  Programming ✔ [00:00:01] [######################################################################################################################################################################] 32.89 KiB/32.89 KiB @ 24.53 KiB/s (eta 0s )
    Finished in 2.239s
```

My setup is somewhat different, though:
- I'm using the same devkit
- I'm using an FT232H, as opposed to an FT2232H
- It seems my pinout is different than what you use:
[image]

Can you verify that your S2 is wired up correctly?

SergioGasquez, 2024-03-22:
Sorry, table was wrong (I'll fix it) but connections on my device were the one you shared.
```
probe-s2 on  main [?] is 📦 v0.1.0 via 🦀 v1.77.0-nightly
❯ cargo r
    Finished dev [optimized + debuginfo] target(s) in 0.11s
     Running `probe-rs run --chip esp32s2 target/xtensa-esp32s2-none-elf/debug/probe-s2`
ERROR probe_rs::probe::common: Fewer IRs detected than TAPs
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An error with the usage of the probe occurred
    1: Invalid IR scan chain
    2: Invalid IR scan chain

probe-s2 on  main [?] is 📦 v0.1.0 via 🦀 v1.77.0-nightly
❯ espflash erase-flash
[2024-03-22T17:12:48Z INFO ] Serial port: '/dev/ttyUSB2'
[2024-03-22T17:12:48Z INFO ] Connecting...
[2024-03-22T17:12:48Z INFO ] Using flash stub
[2024-03-22T17:12:49Z WARN ] Setting baud rate higher than 115,200 can cause issues
[2024-03-22T17:12:49Z INFO ] Erasing Flash...
[2024-03-22T17:13:07Z INFO ] Flash has been erased!

probe-s2 on  main [?] is 📦 v0.1.0 via 🦀 v1.77.0-nightly took 19s
❯ cargo r
    Finished dev [optimized + debuginfo] target(s) in 0.03s
     Running `probe-rs run --chip esp32s2 target/xtensa-esp32s2-none-elf/debug/probe-s2`
ERROR probe_rs::probe::common: Truncated IDCODE: BitSlice<u8, bitvec::order::Lsb0> { addr: 0x6486eb0eab1c, head: 011, bits: 1D } [01, 01, 01, 00, 01, 01, 01, 01, 00, 01, 01, 01, 01, 01, 01, 01, 01, 01, 01, 01, 01, 01, 01, 01, 01, 01, 00, 01, 01]
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An error with the usage of the probe occurred
    1: Invalid IDCODE
    2: Invalid IDCODE
```

Also encountered this error in other iteration:
```
probe-s2 on  main [?] is 📦 v0.1.0 via 🦀 v1.77.0-nightly
❯ espflash erase-flash
[2024-03-22T17:15:20Z INFO ] Serial port: '/dev/ttyUSB2'
[2024-03-22T17:15:20Z INFO ] Connecting...
[2024-03-22T17:15:20Z INFO ] Using flash stub
[2024-03-22T17:15:21Z WARN ] Setting baud rate higher than 115,200 can cause issues
[2024-03-22T17:15:21Z INFO ] Erasing Flash...
[2024-03-22T17:15:39Z INFO ] Flash has been erased!

probe-s2 on  main [?] is 📦 v0.1.0 via 🦀 v1.77.0-nightly took 19s
❯ cargo r
    Finished dev [optimized + debuginfo] target(s) in 0.03s
     Running `probe-rs run --chip esp32s2 target/xtensa-esp32s2-none-elf/debug/probe-s2`
ERROR probe_rs::probe::common: Truncated IDCODE: BitSlice<u8, bitvec::order::Lsb0> { addr: 0x56de072cea8f, head: 111, bits: 01 } [01]
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An error with the usage of the probe occurred
    1: Invalid IDCODE
    2: Invalid IDCODE
```

bugadani, 2024-03-22:
```
❯ espflash erase-flash
[2024-03-22T17:25:49Z INFO ] 🚀 A new version of espflash is available: v3.0.0
[2024-03-22T17:25:49Z INFO ] Serial port: 'COM46'
[2024-03-22T17:25:49Z INFO ] Connecting...
[2024-03-22T17:25:49Z INFO ] Using flash stub
[2024-03-22T17:25:51Z INFO ] Erasing Flash...
 bugad@Dani-PC  C:  _Hobby  crates  probe-s2  main ≢  ?9   1.77.0-nightly
❯ cargo run
    Finished dev [optimized + debuginfo] target(s) in 0.44s
     Running `probe-rs run --chip esp32s2 target\xtensa-esp32s2-none-elf\debug\probe-s2`
      Erasing ✔ [00:00:00] [###################################################################################################################################################################] 192.00 KiB/192.00 KiB @ 274.43 KiB/s (eta 0s )
  Programming ✔ [00:00:01] [######################################################################################################################################################################] 32.89 KiB/32.89 KiB @ 18.10 KiB/s (eta 0s )
    Finished in 2.678s
 WARN probe_rs::util::rtt: No RTT header info was present in the ELF file. Does your firmware run RTT?
 WARN probe_rs::session: Could not clear all hardware breakpoints: An Xtensa specific error occurred.

Caused by:
    0: Xtensa debug module error
    1: Error while accessing register
    2: Unexpected value
 WARN probe_rs::session: Failed to deconfigure device during shutdown: An Xtensa specific error occurred.

Caused by:
    0: Xtensa debug module error
    1: Error while accessing register
    2: Unexpected value
Error: An Xtensa specific error occurred.

Caused by:
    0: Xtensa debug module error
    1: Error while accessing register
    2: Unexpected value
error: process didn't exit successfully: `probe-rs run --chip esp32s2 target\xtensa-esp32s2-none-elf\debug\probe-s2` (exit code: 1)
```

Apart from the random errors after flashing my S2 works as expected. I'd ask you to triple check your cabling, TMS on pin 2 is definitely not correct. This devkit has a slightly misleading silkscreen where the labels are _under_ the pins as opposed to above them as one would expect.

SergioGasquez, 2024-03-22:
Wires: 
- Red: 5V
- Orange: GND
- Blue: TMS
- Purple: TCK
- Grey: TDO
- White: TDI

[image]
[image]

>  TMS on pin 2 is definitely not correct.

Im using pin number 2 from devkit (GPIO 42) as according to:
[image]

bugadani, 2024-03-22:
> Im using pin number 2 from devkit (GPIO 42) as according to:

As I said, that is not supposed to be pin 2, but pin 42 :) Your wires are shifted by one position.

closed 2024-03-22 by SergioGasquez

bugadani, 2024-03-22:
I'm glad this turned out to be an easy issue. I do wonder what monsters we'll run into the current Xtensa implementation though, so please keep reporting anything you find! :)

SergioGasquez, 2024-03-22:
>I do wonder what monsters we'll run into the current Xtensa implementation though, so please keep reporting anything you find! :)

S3 has been working like a charm for me, ESP32 and ESP32-S2 not that much, currently having some issues with the test branch: https://github.com/probe-rs/probe-rs/pull/2292#issuecomment-2014911949

jukrieger, 2024-04-03:
@SergioGasquez Would you share your config for your S3? I'm struggling getting the chip to work, with probe-rs either failing because of a too large of a binary or other errors.

SergioGasquez, 2024-04-03:
> Would you share your config for your S3? 

No real setup required, as S3 has a builtin usb-serial/jtag. I was just connecting the devkit (ESP32-S3-DevKitC-1) via USB port.

jukrieger, 2024-04-03:
@SergioGasquez are you just using `probe-rs run --chip es32s3`?

SergioGasquez, 2024-04-03:
> are you just using probe-rs run --chip es32s3?

[More or less](https://github.com/esp-rs/esp-hal/blob/main/hil-test/.cargo/config.toml#L9), my goal was to run some tests using `embedded-test`, so I was not using any `launch.json`, but something like https://docs.esp-rs.org/book/tooling/debugging/probe-rs.html#example-launchjson should do the trick

jukrieger, 2024-04-03:
Have you encountered any partition table issues, or the flash algorithm reporting mismatches?