probe-rs/probe-rs #2310: Cant flash ESP32-C2

SergioGasquez, 2024-03-22:
**Describe the bug**
Using an FTDI board (ESP-Prog) I cant flash an ESP32-C2

**To Reproduce**
Steps to reproduce the behavior:
1. `cargo generate esp-rs/esp-template --name probe-c2 -d mcu=esp32c2 -d advanced=false`
2. `cd probe-s2`
3. `sed -i 's/runner = "espflash flash --monitor"/runner = "probe-rs run --chip esp32c2"/g' .cargo/config.toml`
4. `cargo run`

**Expected behavior**
The app is properly flashed

**Stacktrace**
```
❯ cargo r
    Finished `dev` profile [optimized + debuginfo] target(s) in 0.06s
     Running `probe-rs run --chip esp32c2 target/riscv32imc-unknown-none-elf/debug/probe-c2`
 WARN probe_rs::probe::ftdi: Read 0 bytes, expected 6
 WARN probe_rs::session: Failed to deconfigure device during shutdown: A RISC-V specific error occurred.

Caused by:
    Error occurred during execution of an abstract command: HaltResume                                                                                                           Error: The flashing procedure failed for 'target/riscv32imc-unknown-none-elf/debug/probe-c2'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: An error with the usage of the probe occurred
    2: An error which is specific to the debug probe in use occurred.
    3: A timeout occurred during probe operation.
```
Sometimes its able to properly flash the app. Also, while testing https://github.com/probe-rs/probe-rs/tree/feature/testing-rebased with C2 I've encountered the following error:
```
    Finished `test` profile [optimized + debuginfo] target(s) in 6.81s
     Running tests/gpio.rs (target/riscv32imc-unknown-none-elf/debug/deps/gpio-d04d7836253eff5e)
ERROR probe_rs::probe::common: Fewer IRs detected than TAPs
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An error with the usage of the probe occurred
    1: Invalid IR scan chain
    2: Invalid IR scan chain
error: test failed, to rerun pass `--test gpio`

Caused by:
  process didn't exit successfully: `probe-rs run /home/sergio/Documents/Espressif/forks/esp-hal/hil-test/target/riscv32imc-unknown-none-elf/debug/deps/gpio-d04d7836253eff5e` (exit status: 1)
```

**Desktop (please complete the following information):**

- Arch Linux

**Additional context**
Connections:
| ESP-Prog | ESP8684-DevKitM-1 v1.1 |
|:--------:|:-----------------------:|
|   VJTAG  |            5V           |
|    GND   |           GND           |
|    TMS   |            4            |
|    TCK   |            6            |
|    TDO   |             7           |
|    TDI   |             5           |

[`ESP8684-DevKitM-1 v1.1` Pinout](https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/_images/esp8684-devkitm-1-pinout_v1.1.png)

bugadani, 2024-03-22:
Can you please try if lowering the JTAG frequency helps? (`probe-rs run --chip esp32c2 --speed 10000` for example). If you could check a few speeds, it would be helpful to determine if the issue is around these parts.

> WARN probe_rs::probe::ftdi: Read 0 bytes, expected 6

Well it's also equally possible that I managed to unfix the connection sequence [here](https://github.com/probe-rs/probe-rs/pull/2099). I didn't know where the addresses came from, and they didn't match the TRM, and I guess I was lax in testing these changes.

SergioGasquez, 2024-03-22:
Did some more testing, found out that when getting the `Fewer IRs detected than TAPs` error `espflash erase-flash` puts it on a working state again:
```
probe-c2 on  main [?] is 📦 v0.1.0 via 🦀 v1.79.0-nightly 
❯ probe-rs run --chip esp32c2 --speed 10000 target/riscv32imc-unknown-none-elf/debug/probe-c2
ERROR probe_rs::probe::common: Fewer IRs detected than TAPs
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An error with the usage of the probe occurred
    1: Invalid IR scan chain
    2: Invalid IR scan chain

probe-c2 on  main [?] is 📦 v0.1.0 via 🦀 v1.79.0-nightly 
❯ espflash erase-flash
[2024-03-22T17:05:55Z INFO ] Serial port: '/dev/ttyUSB2'
[2024-03-22T17:05:55Z INFO ] Connecting...
[2024-03-22T17:05:55Z INFO ] Using flash stub
[2024-03-22T17:05:56Z WARN ] Setting baud rate higher than 115,200 can cause issues
[2024-03-22T17:05:56Z INFO ] Erasing Flash...
[2024-03-22T17:05:58Z INFO ] Flash has been erased!

probe-c2 on  main [?] is 📦 v0.1.0 via 🦀 v1.79.0-nightly took 3s 
❯ probe-rs run --chip esp32c2 --speed 10000 target/riscv32imc-unknown-none-elf/debug/probe-c2
      Erasing ✔ [00:00:00] [#####################################################################################################################################################################################] 192.00 KiB/192.00 KiB @ 303.37 KiB/s (eta 0s )
  Programming ✔ [00:00:01] [########################################################################################################################################################################################] 26.23 KiB/26.23 KiB @ 23.87 KiB/s (eta 0s )    Finished in 1.78s
 WARN probe_rs::util::rtt: No RTT header info was present in the ELF file. Does your firmware run RTT?
^C⏎                                                            
```
Also, I was able to flash the tests, after an `erase-flash` but still getting an error (`Error: A RISC-V specific error occurred.`) but that might not be relevant to this issue.

bugadani, 2024-03-22:
Interesting, I'm not able to reproduce this issue either, but in this case - judging from the successful flash - your wiring is correct. I'll try and see if the esp-prog gives me a different result or not. I've just tried flashing a few times, and `probe-rs erase` without issue. Do you have a consistent way to get probe-rs to fail with `Fewer IRs detected than TAPs`? Is it related to the embedded-test PR/semihosting in some way?

SergioGasquez, 2024-03-25:
> Do you have a consistent way to get probe-rs to fail with Fewer IRs detected than TAPs? Is it related to the embedded-test PR/semihosting in some way?

Mostly was failing to flash anything after a failed test, I've tried reproducing it without the embedded-test, so I guess its related. Doing `espflash erase-flash` puts the chip in a flashable state by probe-rs again.

> probe-rs does not like if the device is in download mode, or if it's reset in quick succession with the reset button.

Yes, seems to be related with the boot mode.

bugadani, 2024-03-25:
Interesting, I'm not sure if we should consider this issue closed, regardless of me reproducing it or not. If issues persists with embedded-test, it may still indicate something incorrect with the C2/RISC-V-related code in probe-rs. Or in some part of the semihosting implementation.

bugadani, 2024-04-03:
Thanks, I'll add this to my list.

A few more questions that may help me reproduce this:
- what is the firmware flashed to the MCU before this error? The same embedded-test elf?
- what happens if you erase the MCU first, e.g. with `espflash erase-flash`?

SergioGasquez, 2024-04-03:
> - what is the firmware flashed to the MCU before this error? The same embedded-test elf?

The `spi_full_duplex` tests, which runs successfully.

> - what happens if you erase the MCU first, e.g. with espflash erase-flash?

Same thing:
```
❯ espflash erase-flash
[2024-04-03T10:10:53Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-04-03T10:10:53Z INFO ] Connecting...
[2024-04-03T10:10:54Z INFO ] Using flash stub
[2024-04-03T10:10:55Z WARN ] Setting baud rate higher than 115,200 can cause issues
[2024-04-03T10:10:55Z INFO ] Erasing Flash...
[2024-04-03T10:10:56Z INFO ] Flash has been erased!

❯ CARGO_BUILD_TARGET=riscv32imc-unknown-none-elf \
                        PROBE_RS_CHIP=esp32c2 \
                        cargo +nightly test --features=esp32c2 --test gpio
....
   Compiling hil-test v0.0.0 (/home/sergio/Documents/Espressif/forks/esp-hal/hil-test)
    Finished `test` profile [optimized + debuginfo] target(s) in 6.97s
     Running tests/gpio.rs (target/riscv32imc-unknown-none-elf/debug/deps/gpio-6bcc7fb3bbead046)
Using 40MHz bootloader
 WARN probe_rs::probe::ftdi: Read 0 bytes, expected 6
 WARN probe_rs::session: Failed to deconfigure device during shutdown: A RISC-V specific error occurred.

Caused by:
    Error occurred during execution of an abstract command: HaltResume                                                                                                                                                                                           Error: The flashing procedure failed for '/home/sergio/Documents/Espressif/forks/esp-hal/hil-test/target/riscv32imc-unknown-none-elf/debug/deps/gpio-6bcc7fb3bbead046'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: An error with the usage of the probe occurred
    2: An error which is specific to the debug probe in use occurred.
    3: A timeout occurred during probe operation.
error: test failed, to rerun pass `--test gpio`

Caused by:
  process didn't exit successfully: `probe-rs run /home/sergio/Documents/Espressif/forks/esp-hal/hil-test/target/riscv32imc-unknown-none-elf/debug/deps/gpio-6bcc7fb3bbead046` (exit status: 1)
```
If after the error, I try to run the `spi_full_duplex` test, it also flashes and runs successfully.

Note that in order to run the `spi_full_duplex` test, you need to wire pins 2 and 3.

bugadani, 2024-04-05:
I'm not entirely sure if I'm doing something incorrectly or if this issue went away, but...

```
    Finished `test` profile [optimized + debuginfo] target(s) in 0.78s
     Running tests\gpio.rs (target\riscv32imc-unknown-none-elf\debug\deps\gpio-43a402619ff1e1cc)
Using 40MHz bootloader
      Erasing ✔ [00:00:00] [#########################################################################################################################################################] 192.00 KiB/192.00 KiB @ 228.93 KiB/s (eta 0s )  Programming ✔ [00:00:01] [############################################################################################################################################################] 47.73 KiB/47.73 KiB @ 29.47 KiB/s (eta 0s )    Finished in 2.64s     

running 6 tests
test tests::test_async_edge     ... ok
test tests::test_a_pin_can_wait ... ok
test tests::test_gpio_input     ... ok
test tests::test_gpio_output    ... ok
test tests::test_gpio_interrupt ... ok
test tests::test_gpio_od        ... ok

test result: ok. 6 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out; finished in 5.09s

❯ probe-rs --version
probe-rs 0.23.0 (git commit: v0.22.0-210-g086823c3-modified)
```

bugadani, 2024-04-05:
Aha, progress! If a test run finishes successfully, the next one runs into the reported error.

bugadani, 2024-04-19:
I'm going to reopen this once more because I can't flash my C2 for some reason. Sorry for the churn, not sure what happened.

SergioGasquez, 2024-04-22:
I was also still getting error in some tests, see https://github.com/probe-rs/probe-rs/pull/2365#issuecomment-2061174562

SergioGasquez, 2024-04-26:
Closed by #2402. Thanks, @bugadani!

PR #2365 (2024-04-17): RISC-V reset sequences
Closes #2164
Closes #1963
Closes #2310

Closes #1964 as I believe that PR will not be necessary anymore.

Issues that may or may not be fixed:

#2317
closed 2024-04-26 by SergioGasquez