probe-rs/probe-rs #2311: Can't Flash STM32F750N8Hx

haydenridd, 2024-03-22:
**Describe the bug**
Despite having it listed as a supported chip, probe-rs fails to write flash for chip `STM32F750N8Hx` complaining that no available flash loader algorithm is available.

**To Reproduce**
Steps to reproduce the behavior:
1. Target `thumbv7em-none-eabihf`
2. Use chip `STM32F750N8Hx` for probe-rs
3. Attempt to flash a binary

**Expected behavior**
probe-rs should be able to flash the chip, given it is listed as supported!

**Stacktrace**
Logs generated when attempting to flash:
 
```
WARN probe_rs::config::target > Using custom sequence for ARMv7 STM32F750N8Hx
     Erasing sectors ⠁ [00:00:00] [###############################################################]      0 B/     0 B @      0 B/s (eta 0s )       Error failed to flash /home/hayden/Documents/rust_experiment/target/thumbv7em-none-eabi/debug/rust_experiment
             
             Caused by:
                 0: Error while flashing
                 1: Trying to write flash, but no suitable (default) flash loader algorithm is linked to the given target: STM32F750N8Hx .
 ```

**Desktop (please complete the following information):**

```
Operating System: Fedora Linux 37 (Workstation Edition)
     CPE OS Name: cpe:/o:fedoraproject:fedora:37
          Kernel: Linux 6.5.12-100.fc37.x86_64
    Architecture: x86-64```

bugadani, 2024-03-22:
Can you please try updating your probe-rs to 0.23 and retry? It should print a bit more detailed error message in case it fails, and that would be helpful.

haydenridd, 2024-03-22:
Amazing! Looks like you already found + fixed it. Still need those more detailed logs?

bugadani, 2024-03-22:
We found something that was incorrect and may be related, but I would appreciate if you could verify that your issue was indeed fixed.

haydenridd, 2024-03-22:
Alright, installed `probe-rs 0.23.0` and unfortunately still getting the same failure, albeit a slightly more detailed message:
```
    Finished dev [unoptimized + debuginfo] target(s) in 0.00s
      Config default
      Target /home/hayden/Documents/rust_experiment/target/thumbv7em-none-eabihf/debug/rust_experiment
       Error The flashing procedure failed for '/home/hayden/Documents/rust_experiment/target/thumbv7em-none-eabihf/debug/rust_experiment'.
             
             Caused by:
                 Trying to write to flash region 0x8000000..0x8010000, but no suitable (default) flash loader algorithm is linked to the given target: STM32F750N8Hx.
                 ```

bugadani, 2024-03-22:
Thanks, that verifies the theory for sure :) You'll need to install probe-rs from github to get the fix, though, 0.23 was released a month or so ago. I believe you can do something like `cargo install probe-rs --git https://github.com/probe-rs/probe-rs.git`

haydenridd, 2024-03-22:
Awesome! Thanks so much yall :)

bugadani, 2024-03-22:
Apologies, the correct command is `cargo install probe-rs --git https://github.com/probe-rs/probe-rs.git --features cli`

haydenridd, 2024-03-22:
Okay updated and we made it past the loader! However... Looks like something different now:
```
WARN probe_rs::probe::arm_debug_interface: DAP FAULT                                                                                        WARN probe_rs::probe::arm_debug_interface: Writing DAP register failed. Ctrl/Stat register value is: Ctrl {
    .0: 4160753539,
    csyspwrupack: true,
    csyspwrupreq: true,
    cdbgpwrupack: true,
    cdbgpwrupreq: true,
    cdbgrstack: true,
    c_dbg_rst_req: false,
    errmode: false,
    trn_cnt: 0,
    mask_lane: 15,
    w_data_err: true,
    read_ok: false,
    sticky_err: false,
    stick_cmp: false,
    trn_mode: 0,
    sticky_orun: true,
    orun_detect: true,
}                                                                                                                                            WARN probe_rs::session: Could not clear all hardware breakpoints: An ARM specific error occurred.

Caused by:
    0: Error using access port
    1: Failed to write register CSW at address 0x00000000
    2: An error occurred in the communication with an access port or debug port.
    3: Target device responded with a FAULT response to the request.                                                                         WARN probe_rs::probe::arm_debug_interface: DAP FAULT                                                                                        WARN probe_rs::probe::arm_debug_interface: Writing DAP register failed. Ctrl/Stat register value is: Ctrl {
    .0: 4160753539,
    csyspwrupack: true,
    csyspwrupreq: true,
    cdbgpwrupack: true,
    cdbgpwrupreq: true,
    cdbgrstack: true,
    c_dbg_rst_req: false,
    errmode: false,
    trn_cnt: 0,
    mask_lane: 15,
    w_data_err: true,
    read_ok: false,
    sticky_err: false,
    stick_cmp: false,
    trn_mode: 0,
    sticky_orun: true,
    orun_detect: true,
}                                                                                                                                            WARN probe_rs::session: Failed to deconfigure device during shutdown: An ARM specific error occurred.

Caused by:
    0: Error using access port
    1: Failed to write register CSW at address 0x00000000
    2: An error occurred in the communication with an access port or debug port.
    3: Target device responded with a FAULT response to the request.                                                                               Error The flashing procedure failed for '/home/hayden/Documents/rust_experiment/target/thumbv7em-none-eabihf/debug/rust_experiment'.
             
             Caused by:
                 0: Something during the interaction with the core went wrong
                 1: An ARM specific error occurred.
                 2: Error using access port
                 3: Failed to write register DRW at address 0x0000000c
                 4: An error occurred in the communication with an access port or debug port.
                 5: Target device responded with a FAULT response to the request.
```

And just to confirm this is a functional JLink that can flash binaries using the "regular" methods, here's a log of `JLinkExe` flashing the same .elf successfully:
```
SEGGER J-Link Commander V7.84f (Compiled Feb  7 2023 16:51:58)
DLL version V7.84f, compiled Feb  7 2023 16:51:27

Connecting to J-Link via USB...O.K.
Firmware: J-Link V11 compiled Jan 30 2023 11:22:20
Hardware version: V11.00
J-Link uptime (since boot): 0d 00h 05m 35s
[... 61 lines cut ...]
FPS24= 00000000, FPS25= 00000000, FPS26= 00000000, FPS27= 00000000
FPS28= 00000000, FPS29= 00000000, FPS30= 00000000, FPS31= FFFFFFFF
FPSCR= 00000000
Reset delay: 0 ms
Reset type NORMAL: Resets core & peripherals via SYSRESETREQ & VECTRESET bit.
Reset: Halt core after reset via DEMCR.VC_CORERESET.
Reset: Reset device via AIRCR.SYSRESETREQ.
'loadfile': Performing implicit reset & halt of MCU.
Reset: Halt core after reset via DEMCR.VC_CORERESET.
Reset: Reset device via AIRCR.SYSRESETREQ.
Downloading file [target/thumbv7em-none-eabihf/debug/rust_experiment.elf]...
Comparing flash   [100%] Done.
Erasing flash     [100%] Done.
Programming flash [100%] Done.
J-Link: Flash download: Bank 0 @ 0x08000000: 1 range affected (32768 bytes)
J-Link: Flash download: Total: 0.505s (Prepare: 0.035s, Compare: 0.009s, Erase: 0.373s, Program & Verify: 0.069s, Restore: 0.017s)
J-Link: Flash download: Program & Verify speed: 463 KB/s
O.K.
Reset delay: 0 ms
Reset type NORMAL: Resets core & peripherals via SYSRESETREQ & VECTRESET bit.
Reset: Halt core after reset via DEMCR.VC_CORERESET.
Reset: Reset device via AIRCR.SYSRESETREQ.
OnDisconnectTarget() start
OnDisconnectTarget() end
```

bugadani, 2024-03-22:
Unfortunately I don't have one of these devices so I won't be able to debug this.

haydenridd, 2024-03-22:
No worries! Let me know how I can help! Newbie to Rust for sure, but _intimately_ knowledgeable about this chip. If you point me in the right direction for a source file I can start twiddling with for what's actually writing stuff to chip (via JLink) I can give it a look

bugadani, 2024-03-22:
I haven't been following the STM32 scene in a few years, I don't really know where to start. What I can see is that the errors provided by probe-rs aren't helpful at all. But it's not clear to me whether programming went through correctly (you kind of suggested that, but I'm not sure if I interpreted the comment corectly), and where the process gets stuck. I personally wouldn't want to waste your time by suggesting red herrings to change, so maybe someone more knowledgeable can chime in.

haydenridd, 2024-03-23:
Ahh, sorry to misconstrue. Nope, `probe-rs` was NOT successfully able to flash the image. I just flashed the same `.elf` file using Segger's CLI that interacts w/ the JLink as a sanity check that there wasn't something wrong with hardware/.elf/etc.

haydenridd, 2024-05-03:
Update:
Tried with the latest commit to `master` [cf6ae5e](https://github.com/probe-rs/probe-rs/commit/cf6ae5e5761731c228a2753617b46dd03869e8ac) and still the same thing, but I grabbed the super detailed log for it. FWIW I'm happy to hop on a Zoom or something and live debug this on one of these chips if it's helpful!

[flash_results_log.zip](https://github.com/probe-rs/probe-rs/files/15203857/flash_results_log.zip)

haydenridd, 2024-05-03:
Even more info! Was able to successfully flash + RTT log on an STLinkV2, so can confirm that this bug is somehow related to the JLink's implementation

haydenridd, 2024-05-03:
Attaching both of those logs here since the comparison may be useful:
[log_comparison.zip](https://github.com/probe-rs/probe-rs/files/15205508/log_comparison.zip)

bugadani, 2024-05-04:
Can you try j-link with [this PR](https://github.com/probe-rs/probe-rs/pull/2415) please?

haydenridd, 2024-05-04:
Interesting! Different behavior (erases successfully, then hangs for a while on the programming stage before erroring out), but still not dice. Error message from running:
```
probe-rs download test_boot_internal.elf --chip STM32F750N8Hx --protocol swd --log-file jlink_flash.log
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
 INFO probe_rs::session: Core status: LockedUp
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
 WARN probe_rs::probe::arm_debug_interface: DAP FAULT                                                                                                                                                     WARN probe_rs::probe::arm_debug_interface: Writing DAP register failed. Ctrl/Stat register value is: Ctrl {
    .0: 4160753539,
    csyspwrupack: true,
    csyspwrupreq: true,
    cdbgpwrupack: true,
    cdbgpwrupreq: true,
    cdbgrstack: true,
    c_dbg_rst_req: false,
    errmode: false,
    trn_cnt: 0,
    mask_lane: 15,
    w_data_err: true,
    read_ok: false,
    sticky_err: false,
    stick_cmp: false,
    trn_mode: 0,
    sticky_orun: true,
    orun_detect: true,
}                                                                                                                                                                                                         WARN probe_rs::session: Could not clear all hardware breakpoints: An ARM specific error occurred.

Caused by:
    0: Error using access port
    1: Failed to write register CSW at address 0x00000000
    2: An error occurred in the communication with an access port or debug port.
    3: Target device responded with a FAULT response to the request.                                                                                                                                      WARN probe_rs::probe::arm_debug_interface: DAP FAULT                                                                                                                                                     WARN probe_rs::probe::arm_debug_interface: Writing DAP register failed. Ctrl/Stat register value is: Ctrl {
    .0: 4160753539,
    csyspwrupack: true,
    csyspwrupreq: true,
    cdbgpwrupack: true,
    cdbgpwrupreq: true,
    cdbgrstack: true,
    c_dbg_rst_req: false,
    errmode: false,
    trn_cnt: 0,
    mask_lane: 15,
    w_data_err: true,
    read_ok: false,
    sticky_err: false,
    stick_cmp: false,
    trn_mode: 0,
    sticky_orun: true,
    orun_detect: true,
}                                                                                                                                                                                                         WARN probe_rs::session: Failed to deconfigure device during shutdown: An ARM specific error occurred.

Caused by:
    0: Error using access port
    1: Failed to write register CSW at address 0x00000000
    2: An error occurred in the communication with an access port or debug port.
    3: Target device responded with a FAULT response to the request.                                                                                                                                      WARN probe_rs::probe::arm_debug_interface: DAP FAULT                                                                                                                                                     WARN probe_rs::probe::arm_debug_interface: Writing DAP register failed. Ctrl/Stat register value is: Ctrl {
    .0: 4160753539,
    csyspwrupack: true,
    csyspwrupreq: true,
    cdbgpwrupack: true,
    cdbgpwrupreq: true,
    cdbgrstack: true,
    c_dbg_rst_req: false,
    errmode: false,
    trn_cnt: 0,
    mask_lane: 15,
    w_data_err: true,
    read_ok: false,
    sticky_err: false,
    stick_cmp: false,
    trn_mode: 0,
    sticky_orun: true,
    orun_detect: true,
}                                                                                                                                                                                                        Error: The flashing procedure failed for 'test_boot_internal.elf'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: An ARM specific error occurred.
    2: Error using access port
    3: Failed to write register DRW at address 0x0000000c
    4: An error occurred in the communication with an access port or debug port.
    5: Target device responded with a FAULT response to the request.
    ```
    And detailed log attached:
    
[jlink_failed_flash_pr_try.zip](https://github.com/probe-rs/probe-rs/files/15210144/jlink_failed_flash_pr_try.zip)

haydenridd, 2024-05-04:
The odd thing is it seems to program ~1KB of flash before it hangs + errors out:
[image]

PR #2312 (2024-03-22): Fix F750 flashing algo flash range
Closes #2311