probe-rs/probe-rs #2317: Panic in `jtag_drm.rs` when using `probe-rs` in combination with `ESP32-C3` and `ESP-Prog`

juliankrieger, 2024-03-25:
**Describe the bug**
After a couple of days of trying to get probe-rs to work with my chip and JTAG adapter, I'm now at a point where I can successfully enter a "pseudo-debugging" state. VSCode enters the debugger, but upon stepping, I'm getting a panic with a stack trace. 

**To Reproduce**
Steps to reproduce the behavior:

1. Download the `esp-blinky` example
2. Run probe-rs with the following configuration:
```
[build]
target = "riscv32imc-esp-espidf"

[target.riscv32imc-esp-espidf]
linker = "ldproxy"
#runner = "espflash flash --monitor" # Select this runner for espflash v2.x.x\
runner = "probe-rs run --chip ESP32C3 --speed 10000 --protocol jtag" # Select this runner for probe-rs
rustflags = [ "--cfg",  "espidf_time64"] # Extending time_t for ESP IDF 5: https://github.com/esp-rs/rust/issues/110

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32c3"
# Note: this variable is not used by the pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v5.1.3"
CRATE_CC_NO_DEFAULTS="1"
```
4. Use `cargo run`

**Expected behavior**
The debugger hits the breakpoint

**Stacktrace**
```
0x105004560 - std::panicking::begin_panic_handler::{{closure}}::h26c73eb75670321f
   8:        0x105002e60 - std::sys_common::backtrace::__rust_end_short_backtrace::hef4773cf0f80485b
   9:        0x105004338 - _rust_begin_unwind
  10:        0x105185aa8 - core::panicking::panic_fmt::h10bd3bfe5bb30d67
  11:        0x104ad45e8 - core::ops::function::FnOnce::call_once::hc66ed890add480ac
  12:        0x1049f6840 - probe_rs::probe::common::<impl probe_rs::probe::JTAGAccess for Probe>::write_register_batch::hc063a306d3fa80dd
  13:        0x104aeb310 - <probe_rs::architecture::riscv::dtm::jtag_dtm::JtagDtm as probe_rs::architecture::riscv::dtm::dtm_access::DtmAccess>::execute::h28718779f760ce86
  14:        0x104aeafe0 - <probe_rs::architecture::riscv::dtm::jtag_dtm::JtagDtm as probe_rs::architecture::riscv::dtm::dtm_access::DtmAccess>::read_deferred_result::hc7f8b50f6b4526ab
  15:        0x104a31854 - probe_rs::architecture::riscv::communication_interface::RiscvCommunicationInterface::read_dm_register::hc0b5771450cdb168
  16:        0x104bc9e80 - <probe_rs::architecture::riscv::Riscv32 as probe_rs::core::CoreInterface>::core_halted::hfa651aeee88430da
  17:        0x104c01704 - probe_rs::session::Session::clear_all_hw_breakpoints::h2fffaf05455d6dbd
  18:        0x104c0e1bc - <probe_rs::session::Session as core::ops::drop::Drop>::drop::h6d4b84f60c4010d0
  19:        0x104e5b228 - core::
ptr::drop_in_place<probe_rs::session::Session>::h62b3dc4874b2aa31
  20:        0x104e5f3b0 - core::ptr::drop_in_place<probe_rs::cmd::dap_server::server::session_data::SessionData>::h7fe1ecafc91c7716
  21:        0x104e76bbc - probe_rs::cmd::dap_server::server::debugger::Debugger::debug_session::hced9676169181247
  22:        0x104e20524 - probe_rs::cmd::dap_server::server::startup::debug::h4a4b53bf2cc69a48
  23:        0x104d93758 - probe_rs::cmd::dap_server::run::he5937137663cb1c2
  24:        0x104cd2af8 - probe_rs::main::h892f4e91d1c28778
  25:        0x104c3ee48 - std::sys_common::backtrace::__rust_begin_short_backtrace::h139193254363bc21
  26:        0x104e59484 - std::rt::lang_start::{{closure}}::hadeb1f7ad991546e
  27:        0x104ff7674 - std::rt::lang_start_internal::haeade8bb0e866fa6
  28:        0x104cdc7fc - _main

thread 'main' panicked at library/core/src/panicking.rs:144:5:
panic in a destructor during cleanup
thread caused non-unwinding panic. aborting.
```

**Desktop (please complete the following information):**

- macOS Sonoma

**Additional context**
There is a lot of confusion around useage of the ESP32-C3. Perhaps we should consider opening a tracking issue?

bugadani, 2024-03-25:
Is an esp-idf binary required to trigger this issue?

Vollbrecht, 2024-03-25:
Could you also clarify what "jtag adapter" are you using? a ftdi one ore something else? Did you burn the internal FUSE to disable the internal jtag debugger?

juliankrieger, 2024-03-25:
@bugadani I am unsure what you mean. I would figure that a flashed binary would be necessary to trigger the debugger?

@Vollbrecht I am using the `ESP-Prog` as per title. I will include it in the issue :) 
And yes, I've burned the FUSE as per documentation. I've also not installed any further ESP-Prog firmware, as per recent documentation Apple ships those by default now.

juliankrieger, 2024-03-25:
> Is an esp-idf binary required to trigger this issue?

I get what you mean now. I've not tried an esp-idf no_std binary. 

OpenOCD works fine with native CPP Esp-idf binaries and arduino based binaries, both in esp-idf and platformio.

juliankrieger, 2024-03-25:
[dmi_timeout.log](https://github.com/probe-rs/probe-rs/files/14750061/dmi_timeout.log)
[fewer_irs_than_taps.log](https://github.com/probe-rs/probe-rs/files/14750074/fewer_irs_than_taps.log)
[flash_alg_fail_verify.log](https://github.com/probe-rs/probe-rs/files/14750097/flash_alg_fail_verify.log)

Here is a couple of error logs I've encountered along the way

juliankrieger, 2024-03-25:
@Vollbrecht I've included the exact chip in the issue. It's the Espressif ESP32-C3-Devkit-M1

juliankrieger, 2024-03-26:
I've been able to get into a debugging session today, but stepping forward still results in a DMI Timeout.

[image]

[image]

bugadani, 2024-03-26:
@MabezDev since you've been playing around with related stuff, do you think these issues will be solvable in the forseeable future? My best guess is that we're still bumping up against some evil setup done by esp-idf and I haven't been following closely to know if anything was done to remedy this.

juliankrieger, 2024-03-26:
@bugadani Just out of personal interest, is there a board where probe-rs support is considered to be better?

MabezDev, 2024-03-26:
@juliankrieger Could you test out https://github.com/probe-rs/probe-rs/pull/2164, you can do so by running `cargo install probe-rs --features cli --git https://github.com/mabezdev/probe-rs --branch memprot`

juliankrieger, 2024-03-26:
> In my experience C6-based boards work better.
Alright. I will look into buying one.

> > is there a recommendation on using cargo embed and a remote gdb debugging session
> 
> I think these should work about the same. The low-level debug drivers are the same.
Strangely enough, I've been facing a different issue where the debugger halts a couple of times on SIGTRAP and then terminates with the same DMI Access Timeout.

juliankrieger, 2024-03-26:
@MabezDev I've been getting two different errors, depending on how I start probe-rs:

## Erasing flash beforehand

1 - Running `cargo run` which leads to `probe-rs run --chip ESP32C3 --speed 10000 --protocol jtag` results in a flashed binary, a warning about no RTT headers and the command line process hanging.
2 - Running the `launch` task in VSCode results in the debugger hitting my breakpoint, but I still get the same error of a DMI Access Timeout. 
[dmi_access_after_patch.log](https://github.com/probe-rs/probe-rs/files/14757219/dmi_access_after_patch.log)

## Not Erasing the flash beforehand

1 - Running `cargo run` as above results in an immediate DMI access timeout.

2 - Running the `launch` task results in an immediate DMI access timeout without the debugger hitting the breakpoint.
[launch_no_erase.log](https://github.com/probe-rs/probe-rs/files/14757233/launch_no_erase.log)

Vollbrecht, 2024-03-26:
@juliankrieger it could be that you are hitting iwdt that are reseting your mcu after hitting the breakpoint. That is because a) esp-idf has code that detects if a debugger is connected b) if so it trys to make a "semihosting" call through the debugger to probe-rs to tell probe-rs "please set the breakpoint for me", but if it fails it will continoue to do so and than after half a second resets the chip. 

Currently probe-rs doesnt support the semihosting answering of that call. To test out the hypotesis first try to disable the watchdog interrupt timer by setting the appropriate sdkconfig and test with that binary.

Vollbrecht, 2024-03-26:
the sdkconfig option is "CONFIG_ESP_INT_WDT=n" .

juliankrieger, 2024-03-26:
On it.

juliankrieger, 2024-03-26:
What's a bit weird to me is that `cargo espflash` has no problem erasing flash. `Probe-rs` however fails, even when given the following arguments:

[image]

juliankrieger, 2024-03-26:
@Vollbrecht running `cargo run` and specifying a log file gives the following error:
[image]

MabezDev, 2024-03-26:
When using probe-rs run, can you also supply `--catch-reset`? That should hopefully solve the DMI timeout issues (if you're still running my branch). I'm not sure if we have an equivalent option for the vscode config.

juliankrieger, 2024-03-26:
No change using attach:
[image]

No change using run:
[image]

No change using run with chip-erase and verify. Sadly, I can not attach a log file, my `RUST_LOG=error` env var is not respected...

Interestingly enough, passing `verify` after a flash erasure fails with `Flash content verification failed`

@MabezDev Can I supply command line options to `probe-rs` in the `launch.json` file?

bugadani, 2024-03-26:
> Can I supply command line options to probe-rs in the launch.json file?

Do you expect any difference with the debugger as opposed to running probe-rs from cli? If yes, why?

juliankrieger, 2024-03-26:
Well for once, I can not test stepping the debugger if I just use `probe-rs run` in my console.

juliankrieger, 2024-03-26:
Right. If you're in need of hardware, I will see what I can do.

bugadani, 2024-03-26:
You can try walking through [this guide](https://docs.esp-rs.org/book/tooling/debugging/openocd.html) while skipping the xtensa-specific info in there.

juliankrieger, 2024-03-26:
@bugadani I there anything left I can do to help? I could clone probe-rs, or try to use it as a library and step through the code there to get to the bottom of this issue.

any more ideas?

lure23, 2024-08-08:
..and/or https://github.com/probe-rs/probe-rs/pull/2748 @juliankrieger

PR #2365 (2024-04-17): RISC-V reset sequences
Closes #2164
Closes #1963
Closes #2310

Closes #1964 as I believe that PR will not be necessary anymore.

Issues that may or may not be fixed:

#2317
closed 2025-03-17 by Yatekii