probe-rs/probe-rs #2322: cargo-embed produces broken output, probe-rs does not

FlareFlo, 2024-03-26:
**Describe the bug**
Cargo-embed produces different defmt output than probe-rs does

**To Reproduce**
Steps to reproduce the behavior:

1. Clone [the project of mine that has the issue](https://github.com/FlareFlo/stm32-rtic/tree/defmt-rtt-issue) 
2. Run `cargo-embed --release`
3. Run `probe-rs run target/thumbv7em-none-eabihf/release/stm32-rtic --chip STM32F411CEUb`

**Expected behavior**
cargo-embed produces non-broken output as probe-rs does.

**Desktop (please complete the following information):**

- Linux (6.8.1-zen1-1-zen)

**Additional context**
cargo-embed run:
[image]
probe-rs run:
[image]

I hope my project is minimal enough to help with reproducing the issue. The hardware used is an STM32F411CEb with the STLink V2.0.
Cargo-embed is on version 0.13.0
probe-rs is on version 0.23.0

Yatekii, 2024-03-26:
Hmm, I think this might be because we use a blanket println and not the one that respects the TUI. Just a wild guess right now though.

bugadani, 2024-03-26:
Notice how the first print (the warning) doesn't show up in cargo embed. This is something else I believe, though the issue isn't immediately obvious to me. I'll try and investigate.

FlareFlo, 2024-03-26:
I was able to observe that when i use `warn` or similar that only a single character is printed, usually just `~`.

bugadani, 2024-03-26:
I believe that single character is one byte of the raw defmt data, though I'm not sure why it's mixed into the output. I'll try to repro this with hardware I have and see where that takes me. In the meantime, have you tried the master branch?

FlareFlo, 2024-03-26:
I added a loop of warnings to exaggerate the effect
```diff
+ for i in 0..100 {
+    warn!("Warn with {}.", i);
+ }
```
This is what i get out of it:
https://asciinema.org/a/KQEaudyXfHypQL6JxEcsa24D3

FlareFlo, 2024-03-26:
> I believe that single character is one byte of the raw defmt data, though I'm not sure why it's mixed into the output. I'll try to repro this with hardware I have and see where that takes me. In the meantime, have you tried the master branch?

By this you mean, install the master branch of probe-rs/cargo/embed?

bugadani, 2024-03-26:
Yes, with `cargo install probe-rs --git https://github.com/probe-rs/probe-rs.git --features cli`. There have been some changes to cargo embed recently and I'd like to know if they broke this even further :)

FlareFlo, 2024-03-26:
Well this is interesting: Cargo-embed produces this output now: `Trying to process defmt data but table or locations could not be loaded.`
probe-rs works fine
i made sure to clean the build dir before re-running with the master-build using:
`cargo install --git https://github.com/probe-rs/probe-rs.git probe-rs --features cli`

bugadani, 2024-03-26:
Thanks, so I did break something there. I'll spend some time tomorrow to hunt these down and report back.

FlareFlo, 2024-03-26:
It sounds like this error occurs on the latest release, but the master-version actually recognizes that it may have failed to load the tables (which the latest fails to report/tell)?

bugadani, 2024-03-26:
If probe-rs run works, cargo embed has to work, too. The info is there, it's just missed. Also I believe cargo embed tries to process partial defmt frames and throws them away, hence the weird output.

PR #2323 (2024-03-27): Fix cargo-embed's defmt processing
This PR removes code that interfered with the defmt data processing. The PR also stips ansi escape sequences from the log output because we don't currently process them, they interfere with text wrapping and look bad when not interpreted.

This PR intends to fix #2322. @FlareFlo can I ask you to test that your issue is no longer present with this? ~Your broken output does not reproduce on my F429, so I'm not sure if it was fixed already or I'm just lucky.~ The broken output reproduces with probe-rs v0.23.0, but not with this PR for me.
closed 2024-03-27 by Yatekii