probe-rs/probe-rs #2337: stm32f105rb works with probe-rs version 0.21.1 but not with 0.23.0

HausBJB, 2024-04-02:
**Describe the bug**
I can flash my application with version 0.21.1 to the device without any problems, but as soon as I use a new version, I get an error message

**To Reproduce**
Steps to reproduce the behavior:

1. create a new application for the device stm32f105rb with the target = "thumbv7m-none-eabi"
2. try to flash the program with the command "probe-rs run --chip stm32f105rb"
3. The following error message appears

**Expected behavior**
The program is flashed onto the device

**Stacktrace**
warning: unnecessary `unsafe` block
  --> src\main.rs:36:25
   |
36 |     gpioc.crl.write(|w| unsafe {
   |                         ^^^^^^ unnecessary `unsafe` block
   |
   = note: `#[warn(unused_unsafe)]` on by default

warning: `stm32f105rbt6_pac` (bin "stm32f105rbt6_pac") generated 1 warning
    Finished dev [unoptimized + debuginfo] target(s) in 0.09s
     Running `probe-rs run --chip stm32f105rb target\thumbv7m-none-eabi\debug\stm32f105rbt6_pac`
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
[... 98 lines cut ...]
  10: BaseThreadInitThunk
  11: RtlUserThreadStart
error: process didn't exit successfully: `probe-rs run --chip stm32f105rb target\thumbv7m-none-eabi\debug\stm32f105rbt6_pac` (exit code: 1)

**Desktop (please complete the following information):**
- VS Code ()
- rustup 1.27.0 (bbb9276d2 2024-03-08)
- Windows 11
- rust-analyzer
v0.3.1906

**Additional context**
Add any other context about the problem here.

toml
[dependencies]
cortex-m = { version = "0.7.7", features = ["critical-section-single-core"] }
cortex-m-rt = "0.7.3"
defmt = "0.3.6"
defmt-rtt = "0.4"
panic-probe = { version = "0.3.1", features = ["print-defmt"] }

[dependencies.stm32f1]
version = "0.15.1"
features = ["stm32f107", "rt"]

bugadani, 2024-04-02:
> features = ["stm32f107", "rt"]

Are you sure your project is properly set up? I'm not sure what the difference is between the 105 and the 107 but that's a glaring difference without any comment on it.

HausBJB, 2024-04-02:
Hello yes the 105 and the 107 are both the same and as I said if I install the version 0.21.1 of probe-rs with the command "cargo install probe-rs --locked --features cli --version 0.21.1" then it works and I can flash the programme the led flashes and the programme runs correctly. I use the ST-LINK V2.J35.S7 JTAG

HerrMuellerluedenscheid, 2024-08-18:
FWIW: I run into the same issue using the NUCLEO L031K6 trying to flash the blinky example from latest master from [embassy](https://github.com/embassy-rs/embassy/blob/main/examples/stm32l0/src/bin/blinky.rs). Tried probe-rs from latest master after I saw #2660 closed. But the error persists in this setup.

HerrMuellerluedenscheid, 2024-08-19:
Ah I configured the wrong chip. Before anybody jumps to action, let me try again with correct settings (tomorrow).

HerrMuellerluedenscheid, 2024-08-19:
No luck, unfortunately. I'm getting the same error.

HerrMuellerluedenscheid, 2024-08-19:
Turned out it was a warning, not an error. The flashing worked after fixing a memory issue. Sorry about the noise!

katyo, 2025-01-05:
I have same issue with STM32F103RBxx.

Yatekii, 2025-02-22:
Can we have some more info on your setup?

Either way, I believe this to be a UX but which does not provide any useful info to the user.