probe-rs/probe-rs #2338: Flashing `esp32-s3` causes flash algorithm verification fail

jukrieger, 2024-04-03:
**Describe the bug**
A clear and concise description of what the bug is.

Setting up `espup` and generating a basic example project like so: `cargo generate esp-rs/esp-idf-template cargo` will result in an example project. I chose Esp-Idf version 5.

Running a basic example project with the ESP32-S3-Devkit-C1 will unfortunately cause an error like this:

```
 WARN probe_rs::util::rtt: No RTT header info was present in the ELF file. Does your firmware run RTT?
 WARN probe_rs::config::sequences::esp: Unknown flash capacity byte: 0
ERROR probe_rs::flashing::flasher: Failed to verify flash algorithm. Data mismatch at address 0x40380400
ERROR probe_rs::flashing::flasher: Original instruction: 0x40001f74
ERROR probe_rs::flashing::flasher: Readback instruction: 0xd23ea1f6
ERROR probe_rs::flashing::flasher: Original: [40001f74, 40000aec, 3fcb0004, 3fcbaaf4, 3fcbaaf8, 3fcbaafc, 3fcbab00, 3fcb0000, aac0ffee, 3fcb0000, aac0ffee, 40000a08, 3fcb0000, aac0ffee, 4000099c, 3fcb0000, aac0ffee, 3fcbaaf8, 403806c8, 3fcbaafc, 3fcbab00, 3fcb0004, 3fcbaaf4, 8000, 3fcb2af4, 40000828, fffff060, 3fcb0000, aac0ffee, 403806c8, 40000a5c, 3fcbaaf4, 3fcbaaf8, 3fcbaafc, 3fcb2af4, 40000a14, 3fcd0000, 6dffff11, 8d037d02, 49504, 41f0062d, 0, 6dfffa11, 71502, 41f0062d, 0, d5fff611, 62d0008, 41f0, 6dfff311, 8d037d02, 99504, 41f0062d, 0, 6dffee11, 1a5502, 41f0062d, 0, 81004136, 8e0ffc5, 81020c00, 2bdffc4, 810008e0, 829ffc3, 29ffc381, ffc28108, 899f97c, 29ffc281, ffc18108, c1810829, ffc191ff, f01d0899, 81004136, 888ffc0, 97ffc091, 81c1198, 20400800, bd8191a0, 8e0ff, f01d0a2d, 1d17ac22, f0, 81004136, 888ffb9, 97ffb991, b8810998, 8e0ff, f01d0a2d, 1d17ac22, f0, 8100a136, 888ffb4, 97ffb491, 3b0c0c98, 161084b0, ac220098, 22f01d13, f01d17ac, 88ffaf51, ffb07105, 91ffb261, a0cffad, 419921a9, b9431827, 9e071, 220443f6, f01d14ac, 92030482, 4a20204, 104b200, a4c10529, d921d8ff, 11bb800c, 20aba0, 9a901199, 1888020, 89208980, ff9f8107, 6d908d9, 4bfcc332, 8871b844, 8981607, 20632830, 799c098, 33871a0c, b90abd01, ff983171, 81ff9971, 6189ff95, 89ff9881, 16ac8251, f87c1189, 12163189, 5e1aa606, 6888129, a1a971a8, 99c09380, 7a01a991, 20c1c2e8, a824c1f2, dd04bd61, e0518807, 81980008, 8806b297, 4611a806, 6880004, 88ba91b8, 449a0689, 58c02290, 4a1621, 98370a5d, e04188b4, a8d0008, 319805ad, 91a62987, 289aff7f, ac22f01d, 82f01d15, 8a8a30a8, 80319fa0, f01d1029, 61006136, 688ff79, 97ff7991, 78812698, 8e0ff, 149140c, 7631570c, 88050cff, 10884001, a2013816, 7bd00c1, 160003e0, 1c6feca, 17aca200, ad0000c6, 2d065905, f01d0a, 91004136, 978ff6c, 88ff6c81, ff6b6108, 20c0658, e7160929, b1a85a00, 6981ff69, e007cdff, a2d0008, 689875a, f01d]
ERROR probe_rs::flashing::flasher: Readback: [d23ea1f6, 1d00dfe5, f0, 520020c0, a0420028, 8d152701, 520020c0, 95270128, e504ad04, 40c00e6, ffde86, a1004136, afb2d233, c8e5ff, e5004256, a2dffdb, 3004c232, 726520a3, 16a24801, c2a20054, 171a518, bdd23241, e504ad03, d282016b, 20c001, 30500832, 13667430, c0030c07, 48320020, d22c3150, 880020c0, a5188c03, eb90ff31, 49d9003, 80d21b81, 20c0a099, 27099800, 20c06199, bc033800, 3eb2013, 31042d20, 2230d215, 20c0a0, d28c0228, c1d256d1, 59b1d259, d222a1d2, a004d265, ada003eb, f675e504, a5d20da1, f01d00d3, c0d20c21, 4480020, 280020c0, 10942702, 2003eb20, 2280042d, 20c0a0, f5860239, ff8265ff, fff406, b003ebb0, a04204bd, 1a0a200, b0209440, 9980839a, 20c0a0, 99270998, 20c0b2, 12470328, 3eb200e, 20042d20, d6a593a4, ffe6c600, 3003eb30, a230043d, a0aa8093, 290020c0, ffe1c60a, 56004136, cb2501a2, 202aa0ff, b2011a56, fad1cfa2, d232c1d1, e0d1fba1, c86511bb, 34c22204, f01d, e6004136, e3a12013, ffafb2d1, 5600b4e5, c7e50052, 202aa0ff, a1a03320, 6342d1de, 16635215, 1d00c765, f0, ad004136, e6020c02, 9a270b13, ffc56502, 22a033a0, f01d1523, ad004136, 1042cc02, c4252011, 112a22ff, f01d, 8d004136, e6020c02, ca210a28, a08820d1, 280020c0, f01d08, a0004136, d2210063, 20c0d1, 120c0288, eb80889c, 48d8003, 90d1c291, 20c0a088, 82089800, 289000a0, 80222093, e0d20081, f01d0008, ac004136, fffc6592, d100ea56, 3c1d203, d1c9a1d2, 2579a6b2, b5a104bc, a5fb7cd1, 2ad00a9, 2500a0b2, b1a1ff0c, bca5d1, a003eba0, 5e2504ad, f01df6, e5006136, 2aa0fff8, ea1620, c1d1f6d1, f6b1d1f6, d1b9a1d1, a104b825, afb2d1a6, a5a5ff, 4003eb40, a131044d, 1154e0d1, c0a03430, 3680020, 20c0660b, c0036900, 3380020, 9c3193cc, 20c0d1, 93560338, d198a10a, 1d00b665, f0, 32032732, c3a20323, 14be518, badb34b, 4b6501b9, d19d8101, 20c0b3a8, b8089800, 4b9a701, a90020c0, d18d8108, 80a0aaa0, 4425a0aa, 3ebc001, b204cdc0, b3a81123, e0251d0c, 41a66fe, a90020c0, 20c006, a8560788]
Error: The flashing procedure failed for 'target/xtensa-esp32s3-espidf/debug/esp32'.

Caused by:
    The RAM contents did not match the expected contents after loading the flash algorithm.

```

**Expected behavior**
Probe-rs should successfully flash the board.

**Desktop (please complete the following information):**

- macOS Sonoma

**Additional context**
***Things tried***
[x] running with `--chip-erase`
[x] running with `speed 10000`
[x] running with `protocol jtag`
[x] running with `allow-erase-all`

Log file:
[log.json](https://github.com/probe-rs/probe-rs/files/14849585/log.json)

`.cargo/config.toml`
```
[build]
target = "xtensa-esp32s3-espidf"

[target.xtensa-esp32s3-espidf]
linker = "ldproxy"
# runner = "espflash --monitor" # Select this runner for espflash v1.x.x
runner = "probe-rs run --chip esp32s3" # Select this runner for espflash v2.x.x
rustflags = [ "--cfg",  "espidf_time64"] # Extending time_t for ESP IDF 5: https://github.com/esp-rs/rust/issues/110

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32s3"
# Note: this variable is not used by the pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v5.1.3"
CRATE_CC_NO_DEFAULTS="1"

```

`rust-toolchain.toml`
```
[toolchain]
channel = "esp"
```

`sdkconfig.defaults`
```
# Rust often needs a bit of an extra main task stack size compared to C (the default is 3K)
CONFIG_ESP_MAIN_TASK_STACK_SIZE=8000

# Use this to set FreeRTOS kernel tick frequency to 1000 Hz (100 Hz by default).
# This allows to use 1 ms granuality for thread sleeps (10 ms by default).
#CONFIG_FREERTOS_HZ=1000

# Workaround for https://github.com/espressif/esp-idf/issues/7631
#CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=n
#CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL=n
```

`Cargo.toml`
```
[package]
name = "test"
version = "0.1.0"
authors = ["jkrieger <julian.krieger@hm.edu>"]
edition = "2021"
resolver = "2"
rust-version = "1.71"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "embassy", "esp-idf-svc/native"]

pio = ["esp-idf-svc/pio"]
std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
alloc = ["esp-idf-svc/alloc"]
nightly = ["esp-idf-svc/nightly"]
experimental = ["esp-idf-svc/experimental"]
embassy = ["esp-idf-svc/embassy-sync", "esp-idf-svc/critical-section", "esp-idf-svc/embassy-time-driver"]

[dependencies]
log = { version = "0.4", default-features = false }
esp-idf-svc = { version = "0.48", default-features = false }

[build-dependencies]
embuild = "0.31.3"
```

bugadani, 2024-04-03:
> I chose Esp-Idf version 5.

Thank you, this is where the problem lies. Esp-idf does something fishy that we're not prepared to deal with.

jukrieger, 2024-04-03:
Hold on, I'm trying V4 and will report back.

bugadani, 2024-04-03:
You should instead try and put your board into Download mode before flashing: press and hold Reset, press and hold Boot0, release Reset, release Boot0. You'll probably be able to flash the board, but I don't think you'll be able to debug it yet. I'll have to look into it some more, I've largely been ignoring this issue but we know about it because the devkits ship with some esp-idf based default firmware.

jukrieger, 2024-04-03:
No change with either V4 or V5. 
I get the following error.

```
 INFO probe_rs::session: Core status: Halted(Breakpoint(Software))
 INFO probe_rs::session: Core status: Halted(Breakpoint(Software))
 WARN probe_rs::util::rtt: No RTT header info was present in the ELF file. Does your firmware run RTT?
 INFO probe_rs::session: Core status: Halted(Breakpoint(Software))
 INFO probe_rs::session: Core status: Halted(Breakpoint(Software))                                                                                                                                                         Error: The flashing procedure failed for 'target/xtensa-esp32s3-espidf/debug/esp32'.

Caused by:
    0: Failed to erase flash sector at address 0x00010000.
    1: The execution of 'erase_sector' failed with code 65536. This might indicate a problem with the flash algorithm.
```

Do we have a tracking issue for this board / ESP-IDF v5? Or is it just the XTENSA tracking issue?

jukrieger, 2024-04-03:
After trying the `ESP32-C3` in a past issue and now with this board failing, this sadly marks the end of my rust debugging journey on espressif board for now.

I will buy the C6 and report back.

bugadani, 2024-04-03:
> I will buy the C6 and report back.

Please do not waste your money when the issue is specific to the software framework you're trying to use. Also please read my last message on what you may try. You can also try bare-metal development with `esp-hal`, if you can get past replacing the esp-idf based default firmware on your devkit it should be a much smoother experience.

jukrieger, 2024-04-03:
Whoops, error on my part. I have tried entering Download mode exactly how you told me. I see no difference in behaviour.

jukrieger, 2024-04-03:
It's also not my money since my employer pays for the boards, so I see no issue in wasting a bit of money lol.

bugadani, 2024-04-03:
Download mode doesn't have a visual indicator, unfortunately.

One more thing you may try is erasing the board with `espflash erase-flash`. If probe-rs fails you, you can always fall back to espflash. This will get eventually resolved but it's not really anyone's priority as far as I can tell.

jukrieger, 2024-04-03:
This has been my workflow for the past weeks already. `Openocd` also works without issue, but I can't get debugging symbols to be correct when using `cppdbg` and a gdb server with my rust projects.

jukrieger, 2024-04-03:
Are there any `beginner` Issues I could look at? If anything, I have a lot of time since this project is partly relevant to my PhD.  I would love to contribute, if possible, but I would still consider myself a beginner in embedded development.

jukrieger, 2024-04-03:
Running with `--chip-erase` also gives the following output, if it helps.
```
probe-rs run --chip esp32s3 target/xtensa-esp32s3-espidf/debug/esp32 --chi
p-erase
 WARN probe_rs::util::rtt: No RTT header info was present in the ELF file. Does your firmware run RTT?
ERROR nusb::platform::macos_iokit::transfer: Failed to submit transfer on endpoint 2: e000404f                                 WARN probe_rs::session: Could not clear all hardware breakpoints: An error with the usage of the probe occurred

Caused by:
    0: USB Communication Error
    1: unknown error                                                                                                          ERROR nusb::platform::macos_iokit::transfer: Failed to submit transfer on endpoint 2: e000404f                                 WARN probe_rs::session: Failed to deconfigure device during shutdown: An error with the usage of the probe occurred

Caused by:
    0: USB Communication Error
    1: unknown error                                                                                                          Error: The flashing procedure failed for 'target/xtensa-esp32s3-espidf/debug/esp32'.

Caused by:
    0: Failed to erase the whole chip.
    1: Something during the interaction with the core went wrong
    2: An error with the usage of the probe occurred
    3: USB Communication Error
    4: unknown error
```

I have not yet figured out how to run probe-rs without erasing first (since I do that with espflash already)

bugadani, 2024-04-03:
I have a few [assigned to me](https://github.com/probe-rs/probe-rs/issues/assigned/bugadani) that seemed like low hanging fruit :)

None of these are relevant to your issue, unfortunately.

This current issue is similar to #1963 and @MabezDev is also working on https://github.com/probe-rs/probe-rs/pull/2164 which is intended to solve similar issues, hopefully finishing that PR will help.

jukrieger, 2024-04-03:
Well I did ask you to run `espflash flash-erase` for a reason, and not `probe-rs` with whatever chip-erase it can do :)

Yes :) Sorry if I was a bit unclear, running `espflash flash-erase` and then trying to flash with probe-rs does not work either. I just wanted to provide more error outputs to be helpful. 

> None of these are relevant to your issue, unfortunately.

I will try to look into them to learn more about how `probe-rs` works :) 

> This current issue is similar to #1963 and @MabezDev is also working on #2164 which is intended to solve similar issues, hopefully finishing that PR will help.

Thanks for the guidance, I'm looking forward to it

lure23, 2024-08-02:
This was very interesting to read.

>Disclaimer: I came here because of the `WARN probe_rs::util::rtt: No RTT header info was present in the ELF file. Does your firmware run RTT?` mention and misread "S3" as "C3".  My boards are C3 (and one C6) and I'd really like to be able to use `probe-rs` with the C3. Also, my use case is `no_std` Embassy, so slightly different from @juliankrieger 's

Some tidbits:

>I will buy the C6 and report back.

For me C6 (ESP32-C6-DevKitM-01) works perfectly with `probe-rs` (haven't tested debugging), so I'd say worth a try

>This will get eventually resolved but it's not really anyone's priority

That's great to hear!  For Espressif's Rust efforts, I feel this (not working on S3 and C3) could be pretty high on the agenda.

Is there a `probe-rs`-compatibility-with-C3 issue somewhere? I haven't found one.

bugadani, 2024-08-02:
> Any pointers, how I could do that?

espflash should work for that (and more)

closed 2024-08-13 by bugadani

Erhannis, 2025-10-01:
I had the same (very similar?) problem, with a new ESP32-S3-N16R8.  I held the boot button and reset the device (for download mode), flashed some code, reset it (into normal mode), and now it behaves like every other ESP32-S3 I have, like it just needed a bootloader or something.

PR #2738 (2024-08-13): ESP32 Xtensa reset sequences
Closes #2338

Algorithms, binaries are ported from OpenOCD.