probe-rs/probe-rs #2351: probe-rs info Failed to open the debug probe.

Jet-29, 2024-04-05:
**Describe the bug**
The issue is that it is unable to create a debug probe for the microbit controller.
It occurs when in wsl using usbipd to attach it from the windows side.
Running lsusb in wsl shows the device.
Running directly on a linux box or windows, the cargo-embed command works as expected and can flash to the chip. The issue seems to be from the fact that somewhere, either the usbipd or wsl is not supported.

**To Reproduce**
Steps to reproduce the behavior:

1. Plug in microbit
2. Use usbipd on windows to attach to wsl
3. Open wsl
4. Check using lsusb it shows up.
5. Check permissions ls -l /dev/bus/usb/.../... has the crw-rw-rw-
6. Install latest probe-rs using cargo install probe-rs --features cli
7. run probe-rs info

**Expected behavior**
I expect it to show the chip information like it does if run from windows.

**Desktop (please complete the following information):**

Windows 10 running wsl2 with ubuntu

**Additional context**
I am not fussy with any solutions. If there is a different tool that may work, I don't need a lot of functionality just need it to flash.
Other options could be different methods to attach the USB to wsl, if usbipd isn't the greatest.
I have had a search around and have not found anything though.

9names, 2024-04-13:
Someone might fix this eventually (if it is fixable), but the quickest solution to this is to not use WSL and instead use the native rust toolchain on your Windows machine to do your builds, and a native build of probe-rs to do your debugging.

Jet-29, 2024-04-13:
I would do that, but running cargo embed on native Windows takes upwards of 5 minutes for apparently no reason. Spinning up a barebones Ubuntu VM installing probe-rs and running the embed command from there is nearly instant so I have no idea why it's so slow. In the meantime, I have found a solution involving a separate VM (so I don't have to wait a long time) and just ssh in and embed. I've written tools to streamline the process so it's no longer an issue. Ty for the reply

9names, 2024-04-13:
If cargo embed runs that slow natively on windows, that is a bug and we should fix that too!
I'll see if I can reproduce the bad performance locally. Thanks for providing the context for your issue, much appreciated

AkhilTThomas, 2024-04-13:
Can you try running as `sudo` ? I also faced the same issue and found that normal user did not have the permissions to access it. With `sudo` it worked fine.

Jet-29, 2024-04-16:
[image]
I have added my user to the permission list and even tried sudoing it, I can confirm that sudo is not the issue. I was digging into the code and found where it was not working. When sudoed it would not have permissions, but when sudoed it would find the chip but fail some checks on the usb protocol or something related. It was a couple of days ago so I don't remember the specifics.

AkhilTThomas, 2024-04-16:
What is the output of `ls -ls /dev/bus/usb/yourdevice`

Jet-29, 2024-04-18:
Update, I was using a microbit v1.38 which works fine on linux and windows but not when passed through to wsl. I recently got a new one v2.21 and plugging it in and passing it through to wsl it just works. So somewhere the issue is between the v1.38, which is 8 or so years old, and wsl.

9names, 2024-04-18:
Just to clarify, did you try updating the firmware on your microbit?

https://microbit.org/get-started/user-guide/firmware/

dboycepa, 2024-05-01:
> I would do that, but running cargo embed on native Windows takes upwards of 5 minutes for apparently no reason. Spinning up a barebones Ubuntu VM installing probe-rs and running the embed command from there is nearly instant so I have no idea why it's so slow.

It might not be the reason for your slow running, but it's worth mentioning that I've seen this kind of behaviour with other build systems such as CMake;  it turned out to be an anti-malware check being invoked on every (windows) user process.  For an individual process it's not particularly noticeable, but when a process-heavy command such as cmake is run on a large workspace, you really notice it.

dboycepa, 2024-05-01:
Now for the real reason I came here: I've been trying to find a way to run `probe-rs info` within a docker container, and although I've made some progress, I have to pass the following arguments to `docker run` to succeed:

```bash
--device=/dev/stlink2-1_          # what StLink calls the device
-v /dev/bus/usb:/dev/bus/usb      # needed for nusb to find the device for opening
--privileged                      # also needed to open the device!
```
(I'm particularly unhappy about `--privileged`. Users of SELinux will also want to be reminded to append either `:z` or `:Z` to the -v argument.)

How is this relevant to WSL? Simply because this is also a containerised environment, and I suspect the same kind of issues apply, although someone with more experience of WSL(2) may be able to comment further.

Note that a big part of the problem is the inability for nusb (used by probe-rs) to actually open a usb device without access to the /dev/bus/usb/ hierarchy, which is why we have to jump through these hoops for a container.  It may be that it is inescably part of modern Linux.

kevinmehall, 2024-05-01:
What is the `/dev/stlink2-1_` device node? A serial port?

probe-rs is using nusb to access another interface of the device, so it's actually doing all its communication through the `/dev/bus/usb` usbfs node.

It also needs access to `/sys` to find the device, which is probably what `--privileged` is getting you.

But I don't think this is relevant to WSL2, which uses usbip to forward Windows devices into the VM and is not a container.

dboycepa, 2024-05-02:
> What is the `/dev/stlink2-1_` device node? A serial port?

It's a symlink to `/dev/sdb`, a usb storage device representing the debug port on an STM Nucleo board.  This is reported as being on bus 3 device 5 (relevant below).

> probe-rs is using nusb to access another interface of the device, so it's actually doing all its communication through the `/dev/bus/usb` usbfs node.

Yes.

> It also needs access to `/sys` to find the device, which is probably what `--privileged` is getting you.

No, I don't think that can be the issue.  At the point of failure, nusb has already found the device and is trying to `open(/dev/bus/usb/003/005, )` returning an `EPERM` error.  `--privileged` allows this call to succeed.

> But I don't think this is relevant to WSL2, which uses usbip to forward Windows devices into the VM and is not a container.

Fair enough.  I just thought there might be some mileage in sharing what I'd found as a possibility to consider.

Erhannis, 2024-07-26:
I'm posting this here so people searching can find it - I had a problem where `probe-rs list` showed a debug probe, but `probe info` showed
```
Error: Failed to open the debug probe.

Caused by:
    0: Probe could not be created
    1: Probe was not found.
```
I skimmed a bunch of github issues; running with RUST_LOG=trace didn't show any errors that stood out to me, but eventually I found @AkhilTThomas 's comment about sudo, and that seems to have been the problem.  I `chown -R root:plugdev /dev/bus/usb`, and `probe-rs info` instead started saying: 
```
Probing target via JTAG

Error identifying target using protocol JTAG: An error with the usage of the probe occurred

Probing target via SWD

Error showing ARM chip information: An ARM specific error occurred.

Caused by:
    0: An ARM specific error occurred.
    1: Failed to connect to the debug port. Please check the debug cable and target power. If SWD multi-drop is used, ensure the correct TARGETSEL value is used.
Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```
This was confusing for a while, but then I went back to VSCode (where I started) and tried debugging my code, and this time it flashed and ran.

akauppi, 2024-08-07:
Tested on WSL2, just to give more data:

- board ESP32-C3-DevKitC-02
- ..with JTAG/USB cable (soldered)
- `probe-rs` with [latest patch](https://github.com/probe-rs/probe-rs/pull/2748): installed by `cargo install --git https://github.com/probe-rs/probe-rs probe-rs-tools --force`
- `usbipd` v.4.3.0
- WSL 2.2.4.0

```
$ probe-rs info --protocol jtag
Probing target via JTAG

No DAP interface was found on the connected probe. ARM-specific information cannot be printed.
RISC-V Chip:
  IDCODE: 0000005c25
    Version:      0
    Part:         5
    Manufacturer: 1554 (Espressif Systems (Shanghai)  Co Ltd)
Xtensa Chip:
  IDCODE: 0000000000
    Version:      0
    Part:         0
    Manufacturer: 0 (Unknown Manufacturer Code)
```

= no problems detected