probe-rs/probe-rs #2362: Memory leak when running embedded-test

t-moe, 2024-04-11:
When running an (embedded-test) integration test on the esp32c6 with probe-rs master ([df2dcf8](https://github.com/probe-rs/probe-rs/commit/df2dcf88e26795188547ca11162f7bdb986a249b))  I sometimes run into a memory leak.

Unfortunately I cannot reliably reproduce it yet.
The leak happens a the start of a test case. Sometimes on the first test I run, sometimes it happens on one of the later tests in the suite.

The result is that the kernel kills probe-rs eventually:
[image]

I will report back wih a reproducer.

t-moe, 2024-04-11:
@bugadani 
Could this be related to rtt attach?  (just a hunch....)

bugadani, 2024-04-11:
I don't know how RTT would eat up memory in the 100MB/s ballpark. The interesting part is the "sometimes" in your report - is this really manifesting randomly?

t-moe, 2024-04-15:
The most interesting bugs just disappear, once you start investigating them :disappointed: .....

(I'll reopen this once it surfaces again, and I have a reproducer..)

closed 2024-04-15 by t-moe