probe-rs/probe-rs #2376: RP2040 - Core gets stuck after reflashing

brt-adam-snaider, 2024-04-16:
**Describe the bug**

Occasionally (~10% of the time) when I use probe-rs to reflash my chip with new software, the new software is reprogrammed but the chip appears to be stuck. The only way to get out of this state is to physically reset the chip.

```
probe-rs 0.23.0
git commit: 426750d
```

**To Reproduce**
Use probe-rs to reflash an RP2040 chip. This issue doesn't always manifest. I've noticed that it mostly happens with larger payloads

**Expected behavior**
Chip is reflashed and begins running the new software

**Desktop (please complete the following information):**

Ubuntu 22.04

**Additional context**

I'm using a CMSIS-DAP probe.

This is what I see after reflashing (again it's almost like the chip is running no software after reflashing since neither my old software nor the new one appears to be executing)

```
      Erasing ✔ [00:00:00] [###########################################################################################################################################################################################################################################################################################################] 56.00 KiB/56.00 KiB @ 71.87 KiB/s (eta 0s )
  Programming ✔ [00:00:01] [###########################################################################################################################################################################################################################################################################################################] 56.00 KiB/56.00 KiB @ 36.67 KiB/s (eta 0s )
    Finished in 2.337s
        Done processing config default
 WARN probe_rs::architecture::arm::core::armv6m: Core is running, but we expected it to be halted
```

xingrz, 2024-04-18:
Same on STM32F030CC. It seems that the chip is not reset automatically by probe-rs after flash done, unless manually run `probe-rs reset`. I don't know if it's a expected behavior.

brt-adam-snaider, 2024-04-19:
probe-rs reset doesn't seem to be working for me

```
$ probe-rs reset --chip RP2040
 WARN probe_rs::architecture::arm::core::armv6m: The core is in locked up status as a result of an unrecoverable exception
```

akhilles, 2024-10-04:
Was running into a similar issue when reflashing an stm32h7 via an rp2040 probe running https://github.com/probe-rs/rusty-probe-firmware. Adding this delay between programming and the reset_and_halt seems to have fixed it: https://github.com/akhilles/probe-rs/commit/349ec7f1e93a1d238943db2258549edbcc631b93. Using the `--verify` flag, which effectively also adds a delay, also works.