probe-rs/probe-rs #2379: RP2040 Can't attach to running probe

brt-adam-snaider, 2024-04-17:
**Describe the bug**
I flashed some software to an RP2040 chip with probe-rs using a CMSIS-DAP debug probe. I'm trying to attach it with `probe-rs attach ../../../target/thumbv6m-none-eabi/release/udp_test --chip RP2040` but I get the following error

```
Error: An ARM specific error occurred.

Caused by:
    Timeout occurred during operation.
```

**To Reproduce**
Flash software to an RP 2040, Ctrl-C, try to attach to the RTT log

**Expected behavior**
Command succeeds and proceeds to print the DEFMT log

**Stacktrace**

**Desktop (please complete the following information):**

- Ubuntu 22.04

**Additional context**
```
$ probe-rs --version
probe-rs 0.23.0
git commit: 426750d
```

jenshnielsen, 2024-04-20:
I think this was resolved after 0.23.0 see https://github.com/probe-rs/probe-rs/issues/2183

brt-adam-snaider, 2024-04-22:
That was my bad. I actually opened that ticket with my personal account and forgot about it. I'll close as duplicate!

closed 2024-04-22 by brt-adam-snaider