probe-rs/probe-rs #2396: cargo embed rtt BinaryLE channel does not work properly

ruabmbua, 2024-04-21:
**Describe the bug**

It seems like the *socket* feature does not work properly when setting up a RTT channel in the *BinaryLE* format. According to wireshark, it does not even attempt to establish a connection. The *String* format works fine.

Also, for me the *cargo-embed* console display of BinaryLE RTT data is not line wrapped. For reference, I am sending 8 byte messages in one call to the rtt_target *UpChannel*.

**To Reproduce**
Steps to reproduce the behavior:
1) Configure *UpChannel 1* with *NoBlockSkip* on both ends, and set *socket* to localhost + some port, then listen on it with e.g. netcat.
2.) Run cargo-embed in a project that sends 8byte messages as rtt data on channel 1.   

**Expected behavior**
You will see that there is no attempt at connecting the socket. Also the *cargo-embed* interface for channel 1 will not line-wrap the binary data. 

**Desktop (please complete the following)**
- OS: Linux (Arch)
- Target: RPI Pico (RP2040)
- Debugger: Raspberry PI debug probe
- probe-rs version: master (0.23)

bugadani, 2024-04-21:
Both of these issues were supposed to be fixed by #2249

closed 2024-08-06 by bugadani

ruabmbua, 2024-08-06:
I can confirm that this seems to be working now!