probe-rs/probe-rs #2425: probe-rs run leaves RTT in blocking mode when disconnecting

9names, 2024-05-04:
**Describe the bug**
With probe-rs 0.23, pressing CTRL+C while running probe-rs run leaves RTT in non-blocking mode allowing the firmware to continue running indefinitely. This is (I think) the behaviour that we want.

With the current master (rev cf6ae5e5761731c228a2753617b46dd03869e8ac) it does not, which means that when the RTT buffer eventually fills up the running firmware hangs.

**To Reproduce**
Install probe-rs cli from git.
Add defmt/rtt to your crate.
Log inside of a loop with an LED blink so that progress is observable without RTT
Run with probe-rs run from git
Observe that the program never locks up.
Press CTRL+C to disconnect debugger.
Wait for firmware to lock up

**Expected behavior**
RTT would switch to non-blocking and firmware would never lock up

bugadani, 2024-05-04:
> With probe-rs 0.23, pressing CTRL+C while running probe-rs run leaves RTT in non-blocking mode allowing the firmware to continue running indefinitely. This is (I think) the behaviour that we want.

Well, 0.23 didn't transition to blocking mode, so sort-of-maybe-not-exacly. Not transitioning back, however, is a mistake and thanks for noticing it.

bugadani, 2024-05-15:
Hmm I can't find where 0.23 changes channel mode, except in the dap server. Either way, this should be fixed by #2466 if I managed to find all points where we need to restore the flag :) 🤞

PR #2466 (2024-05-16): Restore RTT channel mode if we've updated it
Closes #2425
closed 2024-05-16 by Yatekii