probe-rs/probe-rs #2445: Design a lower level interface

Yatekii, 2024-05-10:
When we attach, we straight get a session. This is very limiting in several usecases:

- Target autodetect
- More complex unlocking sequences
- Recover MCU state

and many more.

We want more low level interface access that is more granular & specific than the current Session.
We still want to keep the Session and its interface, but we want an interface point before the Session where we can do more low level operations.

bugadani, 2024-05-10:
Lower level is a bit vague IMO. We already have system-level (Session) and core-level APIs, I'm not sure if we can go lower than core-level. On the other hand, the current model is slightly insufficient, as the Core model is entangled with the debug modules, which behave differently between architectures.

Yatekii, 2024-05-10:
Lower means AP level here kind of. I added a comment :)