probe-rs/probe-rs #2446: Multicore support

Yatekii, 2024-05-10:
We already support multicore chips, but the support is very limited.
Specifically we always try to unlock both cores, which is often the wrong choice and can lead to unexpected issues.

We should be able to cope with locked cores and just work with the unlocked core, instead of requiring all to be unlocked from a library perspective.
We want to be able to selectively unlock cores manually without just doing this automatically.

bugadani, 2024-05-10:
What does unlocking mean in this context? We support architectures that don't understand this term.

Yatekii, 2024-05-10:
Well, whatever it means within the arch. unlocking/enabling is used interchangeably. We will have to work out the exact API/requirements.

bugadani, 2024-05-31:
An interesting first question: can we define halting and resuming (also, system reset [and halt]) in a multi-core system (both the expected semantics, as well as per-system properties)? Currently we print warnings if we find that a core's state is not as we expect, but that isn't an error if halting one core halt every core.