probe-rs/probe-rs #2454: Clean out debugger code

Yatekii, 2024-05-10:
The debugger code has grown tremendously and we really need to clean it up.
Currently not even `cargo fmt` understands it :O

bugadani, 2024-05-10:
We should define some goals here, or we'll have another eternally open issue.

Yatekii, 2024-05-10:
Yeah, the idea is that we kind of make sub issues (or more precise ones). Those are only for a rough overview :)

bugadani, 2024-05-11:
Fair, but cleaning up is a fractal task. No matter what we do, there's always more.

Yatekii, 2024-05-11:
Oh yeah it would be very nice to define some metrics.
If you have good ideas that would be nice, but I would say of the top of my head for me that would be:
- Functions 30-50 lines max
- Max 2 indentations excluding trait and function impl base ones.