probe-rs/probe-rs #2456: Stable scripting API in the CLI

Yatekii, 2024-05-10:
For 3rd party CI we want a stable CLI API that does not change after a specification round that people can use to automate things.

HalfSweet, 2025-09-02:
Hi, I would like to know if there is any progress on this work so far. I now have a number of actions that need to be scripted to accomplish a series of actions, in openocd I can use tcl and in jlink there are similar scripts.
Currently I'm trying to wrap a program using probe-rs to the rust library, but this is too cumbersome

Yatekii, 2025-10-05:
Why is using probe-rs as a library too cumbersome? It should be *easier* than wrapping the CLI.

If you are missing some functionality we should expose that.