probe-rs/probe-rs #2477: Fail to compile via cargo `error: failed to run custom build command for libudev-sys v0.1.4`

soypat, 2024-05-18:
**Describe the bug**
Tried to install via cargo.

**To Reproduce**
Steps to reproduce the behavior:
```sh
# Install fresh rust on a new Linux Mint PC
$ curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh
$ rustc --version
rustc 1.78.0 (9b00956e5 2024-04-29)
$ cargo --version
cargo 1.78.0 (54d8815d0 2024-03-26)

# First thing I did was try install probe-rs to use embassy
$ cargo install probe-rs --features cli
```

<details><summary>Logs</summary>

```
$ cargo install probe-rs --features cli
   Updating crates.io index
  Downloaded probe-rs v0.23.0
  Downloaded 1 crate (4.0 MB) in 3.16s
  Installing probe-rs v0.23.0
    Updating crates.io index
  Downloaded Inflector v0.11.4
  Downloaded ahash v0.7.8
  Downloaded anstream v0.6.14
  Downloaded adler v1.0.2
  Downloaded ahash v0.8.11
  Downloaded addr2line v0.21.0
  Downloaded allocator-api2 v0.2.18
  Downloaded anstyle v1.0.7
  Downloaded bitflags v1.3.2
  Downloaded backtrace-ext v0.2.1
  Downloaded arrayvec v0.7.4
  Downloaded bytes v1.6.0
  Downloaded atomic-waker v1.1.2
  Downloaded anstyle-query v1.0.3
  Downloaded autocfg v1.3.0
  Downloaded bitfield v0.14.0
  Downloaded bytecheck_derive v0.6.12
  Downloaded cfg_aliases v0.1.1
  Downloaded array-init v2.1.0
  Downloaded cassowary v0.3.0
  Downloaded castaway v0.2.2
  Downloaded bytecheck v0.6.12
  Downloaded cargo-platform v0.1.8
  Downloaded equivalent v1.0.1
  Downloaded darling_macro v0.14.4
  Downloaded cfg-if v1.0.0
  Downloaded camino v1.1.7
  Downloaded byte-unit v5.1.4
  Downloaded borsh-derive v1.5.0
  Downloaded block-buffer v0.10.4
  Downloaded anstyle-parse v0.2.4
  Downloaded zero v0.1.3
  Downloaded boolean_expression v0.3.11
  Downloaded bytemuck_derive v1.6.0
  Downloaded clap_derive v4.5.4
  Downloaded bincode v1.3.3
  Downloaded bytesize v1.3.0
  Downloaded colored v2.1.0
  Downloaded cpufeatures v0.2.12
  Downloaded defmt-parser v0.3.4
  Downloaded concurrent-queue v2.5.0
  Downloaded directories v5.0.1
  Downloaded either v1.12.0
  Downloaded dirs-sys v0.4.1
[... 373 lines cut ...]
**Desktop (please complete the following information):**
```
$ neofetch
             ...-:::::-...                 pato@qtpi 
          .-MMMMMMMMMMMMMMM-.              --------- 
      .-MMMM`..-:::::::-..`MMMM-.          OS: Linux Mint 21.3 x86_64 
    .:MMMM.:MMMMMMMMMMMMMMM:.MMMM:.        Host: B660M DS3H DDR4 
   -MMM-M---MMMMMMMMMMMMMMMMMMM.MMM-       Kernel: 6.5.0-35-generic 
 `:MMM:MM`  :MMMM:....::-...-MMMM:MMM:`    Uptime: 14 hours, 33 mins 
 :MMM:MMM`  :MM:`  ``    ``  `:MMM:MMM:    Packages: 2728 (dpkg), 22 (flatpak) 
.MMM.MMMM`  :MM.  -MM.  .MM-  `MMMM.MMM.   Shell: bash 5.1.16 
:MMM:MMMM`  :MM.  -MM-  .MM:  `MMMM-MMM:   Resolution: 1920x1080, 1920x1080 
:MMM:MMMM`  :MM.  -MM-  .MM:  `MMMM:MMM:   DE: Cinnamon 6.0.4 
:MMM:MMMM`  :MM.  -MM-  .MM:  `MMMM-MMM:   WM: Mutter (Muffin) 
.MMM.MMMM`  :MM:--:MM:--:MM:  `MMMM.MMM.   WM Theme: Mint-Y-Dark-Purple (Mint-Y 
 :MMM:MMM-  `-MMMMMMMMMMMM-`  -MMM-MMM:    Theme: Mint-Y-Dark-Purple [GTK2/3] 
  :MMM:MMM:`                `:MMM:MMM:     Icons: Mint-Y-Purple [GTK2/3] 
   .MMM.MMMM:--------------:MMMM.MMM.      Terminal: gnome-terminal 
     '-MMMM.-MMMMMMMMMMMMMMM-.MMMM-'       CPU: 12th Gen Intel i5-12400F (12) @ 
       '.-MMMM``--:::::--``MMMM-.'         GPU: AMD ATI Radeon RX 6800/6800 XT  
            '-MMMMMMMMMMMMM-'              Memory: 5960MiB / 31928MiB 
               ``-:::::-``
                                                                   
                                                                   
```

bugadani, 2024-05-19:
I would like to ask you to please follow the [install instructions](https://probe.rs/docs/getting-started/installation/#linux%2C-macos). `cargo install` is not recommended (despite it still being a listed method) and won't be available starting from the next release. Please let us know if you have further issues with the documented instructions.

soypat, 2024-05-19:
OK, running the suggested script solved it! Thank you.

closed 2024-05-19 by Yatekii