probe-rs/probe-rs #2491: STM32H745IIx RTT does not work unless both cores are running.

liarokapisv, 2024-05-23:
The STM32h745IIx is a cortex-m7/cortex-m4 dual-core variant.
It appears that unless both cores are up and running "properly" then I cannot get any RTT connection. Attempting to connect fails with `failed to attach to RTT`. I am not sure what "properly" entails in this case, just flashing a simple endless loop in the CM4 core allows me to properly get output from the CM7 core. Just thought I should report this, it was very annoying to figure out and the workaround is still very annoying.

maxgr0, 2024-06-04:
Hey @liarokapisv how are you specifying the target core to which the given program gets flashed to when using probe-rs? We're facing a similar situation right now, the solution with the loop on the M4 might be hacky but is at least a really quick solution.
Cheers!🍻

liarokapisv, 2024-06-04:
I just change the address in the memory.rs file and also the cortex m variant for my hal. I am currently working on the cm7 core so I did this only once and haven't had an issue since. I will need to revisit this once I get back to working with the m4 core.