probe-rs/probe-rs #2516: Insufficient DWARF info with recent nightly

ia0, 2024-05-29:
**Describe the bug**
When using nightly-2024-05-24, I get `Insufficient DWARF info`, and indeed I don't get log locations. With nightly-2024-01-19, there is no problem.

**To Reproduce**
Steps to reproduce the behavior (you need a nRF52840-dk connected):

1. `git clone -b probe-rs https://github.com/ia0/wasefire.git`
2. `cd wasefire`
3. `./script/setup.sh`
4. `cargo xtask applet rust hello runner nordic`

The output is:
```
 WARN probe_rs::util::rtt: Insufficient DWARF info; compile your program with `debug = 2` to enable location info.
<lvl> 0.097538: hello world
└─ <mod path> @ └─ <invalid location: defmt frame-index: 23>:0   
```

**Expected behavior**
I would expect to see the log level and location as with the previous compiler (for which I used 0.22.0 because 0.24.0 doesn't compile with it).

**Desktop (please complete the following information):**

- Linux Debian

**Additional context**

Also, when I compare the binary size, it is indeed very different. Before it was 10M and now it is 500K.

Tiwalun, 2024-05-29:
That seems like an issue with rustc or defmt, both of which are not handled by this project. If debug info is missing, which seems to be the case based on the size of the binary, there is nothing we can do.

The Rust project is probably the place to ask for help.

Yatekii, 2024-05-29:
Is the debug info on linux split on that nightly maybe? I cannot explain the drop otherwise.

ia0, 2024-05-30:
I also thought it was `split-debuginfo` and tested it (and forgot to mention in the issue), but that's not it. However, I ended up figuring out what's going on. It's https://github.com/rust-lang/cargo/pull/13257 meant to fix https://github.com/rust-lang/cargo/issues/4122 which seems to only check whether debug is enabled at cargo level and not through RUSTFLAGS, so cargo was stripping debuginfo thinking the user didn't want debug. I'm fixing this in https://github.com/google/wasefire/pull/499.

The current hint "compile your program with `debug = 2`" is technically correct when done in the Cargo.toml. It simply is not enough when using RUSTFLAGS, where `-C debuginfo=2 -C strip=none` is actually needed.

Thanks for the rubberducking!

closed 2024-05-30 by ia0

bugadani, 2024-05-30:
Well the hint is `debug = 2` not `-C debuginfo=2` ;)

ia0, 2024-05-30:
Yes that's why it's technically correct :-)

(Writing "enable debug" would not have been technically correct, it needs to be "enable debug in cargo".)