probe-rs/probe-rs #2518: More advice if device not found (error messages)

akauppi, 2024-05-29:
**Is your feature request related to a problem? Please describe.**

I'm just getting started with `probe-rs` and have not been able to make it see my ESP32-C3-DEVKIT-M1, yet. There may be myriad reasons what's wrong, but also the CLI could lead the way. Now I get:

```
$ probe-rs info --connect-under-reset
Error: No connected probes were found.
```

Once I've nailed this, I could tell what it might have pointed to. Not yet.

**Describe the solution you'd like**

If a probe isn't seen by `probe-rs`, it would give ideas, what to check. Maybe a link to a specific page (URL), if there's a lot to consider.

**Describe alternatives you've considered**

I keep reading all of the documentation. 

In my specific case, I'm connecting via USBIP which may affect things. The `/etc/udev/rules.d/69-probe-rs.rules` expects `303a:1001` (I think) but I get:

```
$ lsusb
...
Bus 001 Device 002: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
...
```  

>However, please consider this issue is not about helping me reach the device; but about not giving any lead on where to look. :)

**Additional context**

bugadani, 2024-05-29:
probe-rs can not communicate over a USB serial port, you'll need to use espflash for that device I'm afraid. That being said, this is a common enough issue that we might provide some more useful feedback.

akauppi, 2024-06-02:
Hi @bugadani

This is a *bit* confusing, and I'm hoping to clarify things for others in the process. Thanks for your help! 

The page ESP32-C3 > [JTAG Debugging](https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-guides/jtag-debugging/index.html) (looks authoritative to me) mentions:

>Debugging through the USB interface implemented in ESP32-C3 requires to have a chip with revision 0.3 or newer. Please use other debugging options (e.g., with ESP-Prog) for chip revisions 0.1 and 0.2. [...]

My board has "v.1.1" printed on the bottom side. If this is the "chip revision" I would.. expect the USB port to have JTAG functionality.

bugadani, 2024-06-02:
Don't confuse the chip version with the development board's version. You can see the chip's version info if you run `espflash board-info` (which is, indeed, a bit of a misleadingly named command).

The chip's USB hardware may indeed support JTAG, but if you look at the development board's [schematics](https://dl.espressif.com/dl/schematics/SCH_ESP32-C3-DEVKITM-1_V1_20200915A.pdf) you can see that, by default, the USB port is not connected to the ESP32 chip.

akauppi, 2024-06-03:
Thanks @bugadani 

I was able to get "println" output yesterday, using `espflash`, not `probe-rs`. It will likely take me a while to get my head truly wrapped around these tools. That's fine. In case you are curious - the work is available at [ESP32-Mac](https://github.com/lure23/ESP32-Mac).

As to this ticket, let's keep it to the title, i.e. a request for an error message that provides some pointers instead of just saying "no connection".

akauppi, 2024-09-18:
Closing, since the suggestion on more descriptive error messages didn't get much support (perhaps it's a non-issue?). Also, I've moved on and no longer suffer from the newbie problems.

For future visitors:

- you want to have `lsusb` report something with `JTAG` in its name. Then `probe-rs` will find it, just fine.

amadeu01, 2025-10-20:
I cannot find my device with probe-rs. I'm using omarchy linux. I was able to find the esp32 using `espflash`

```
cobot-rs main  ? ❯ espflash board-info
[2025-10-20T16:01:08Z INFO ] Serial port: '/dev/ttyUSB0'
[2025-10-20T16:01:08Z INFO ] Connecting...
[2025-10-20T16:01:12Z INFO ] Using flash stub
Chip type:         esp32 (revision v3.1)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC address:       fc:b4:67:f1:90:c0
Security features: None
```

```
$ lsusb
Bus 001 Device 005: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
```

akauppi, 2025-10-20:
@amadeu01 You did notice that this is a closed issue, did you?  I'll open it now, but please keep it brief. :)

Also, `probe-rs` requires a JTAG/USB port I think whereas `espflash` works with both normal and JTAG. Do you have two ports?

closed 2025-10-20 by bugadani

amadeu01, 2025-10-20:
I'm sorry. I think the problem is indeed related to the JTAG/USB, is that possible to use probe-rs with USB ?

how did you make this ?

> you want to have lsusb report something with JTAG in its name. Then probe-rs will find it, just fine.

bugadani, 2025-10-20:
https://docs.espressif.com/projects/rust/book/getting-started/tooling/probe-rs.html

probe-rs requires an external debug probe to work with an ESP32