probe-rs/probe-rs #2519: Raspberry Pi Pico + pico-debug: device already open

boguscoder, 2024-05-30:
**Describe the bug**
I'm using "virtual probe" for rp2040 [pico-debug ](https://github.com/majbthrd/pico-debug/tree/master) that was also mentioned in [issue1253](https://github.com/probe-rs/probe-rs/issues/1253) couple years ago
probe works fine with `openocd`, `gdb` connects fine and debugging works so all things seem to be in place, however `probe-rs` can't seem to do anything except list it , everything else ends up failing on access to hid device:
```
❯ probe-rs list
The following debug probes were found:
[0]: CMSIS-DAP -- 1209:2488:DF62504213401B37 (CMSIS-DAP)
```

```
❯ cargo run --release --bin test
    Finished `release` profile [optimized] target(s) in 0.68s
     Running `probe-rs run --chip RP2040 target/thumbv6m-none-eabi/release/test`
Error: Failed to open the debug probe.

Caused by:
    0: The debug probe could not be created.
    1: An HID API occurred.
    2: hidapi error: hid_open_path: failed to open IOHIDDevice from mach entry: (0xE00002C5) (iokit/common) exclusive access and device already open
```

**To Reproduce**

1. Install pico-debug by flashing the provided pico-debug-gimmecache.uf2 to a Raspberry Pi Pico.
2. Use openocd with virtual probe `openocd -f board/pico-debug.cfg`
3. Use any probe-rs command other than list a la `probe-rs debug`

**Expected behavior**
probe-rs theoretically should "just work" since CMSIS-DAP interface is available, would you mind shedding light if this is probe-rs access issue, my macos setup issue or pico-debug issue

**Desktop (please complete the following information):**
- macOS Big Sur

bugadani, 2024-06-01:
I'm honestly not entirely sure what you want to achieve. If you start openocd and it takes control over the USB device, you can't use the same device from a different software. I think the `exclusive access and device already open` error is clear enough in this regard.

Where did you find instructions to start openocd before probe-rs?

boguscoder, 2024-06-01:
Now when you replied Ive understood my confusion, I've clobbered instructions from pico-debug and probe-rs, together, I _assumed_ probe-rs would connect _via_ openocd but I totally see I was wrong. Without openocd running I get more progress (still issues but ill ask/report separately if I cant get through)

Appreciate time response, it helped!

closed 2024-06-01 by boguscoder