probe-rs/probe-rs #2537: v0.24: Flashing is broken with j-link on AT32F437VMT7

skibon02, 2024-06-04:
**Describe the bug**
After upgrade from v0.23 to stable version v0.24 using `curl ...` command I can no longer download firmware to my microcontroller.
I use JLink and SWD interface to flash AT32F437VMT7 microcontroller.
Downgrade to v0.23 immediately fix the problem.

**To Reproduce**
Steps to reproduce the behavior:

1. Install v0.24 version using attached `curl` script in github releases section 
2. Download udev rules file and apply it
3. Wire up J-Link to AT32F437VMT7 microcontroller using SWD interface, connect J-Link to pc
4. Clone git repository [emcell](https://github.com/skibon02/emcell.git)
5. Navigate to emcell/example/cell1 and execute `run.sh`

**Expected behavior**
Successful erace and download operation, no warnings or errors emitted

**Stacktrace**
Here is the RUST_LOG=trace output fox probe-rs download command:

<details>
<summary>Console output</summary>

```
(base) ~/projects/plc-runtime-rs/rte git:[dev]
probe-rs download --chip AT32F437VMT7 --probe 1366:0105 ../target/thumbv7em-none-eabihf/release/rte
      Erasing ✔ [00:00:06] [######################################################################################################################################################] 344.00 KiB/344.00 KiB @ 51.73 KiB/s (eta 0s )
  Programming ✔ [00:00:02] [#--------------------------------------------------------------------------------------------------------------------------------------------------------] 1.00 KiB/343.00 KiB @    489 B/s (eta 0s )
Error: The flashing procedure failed for '../target/thumbv7em-none-eabihf/release/rte'.

Caused by:
    0: The page write of the page at address 0x08001000 failed.
    1: Something during the interaction with the core went wrong
    2: A timeout occurred.
```
</details>

<details>
<summary>Another console output</summary>

```
    Finished `release` profile [optimized] target(s) in 10.44s
     Running `probe-rs download --chip AT32F437VMT7 --probe '1366:0105' ../target/thumbv7em-none-eabihf/release/cell1`
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
Error: The flashing procedure failed for '../target/thumbv7em-none-eabihf/release/cell1'.

Caused by:
    0: The page write of the page at address 0x08000000 failed.
    1: The core entered an unexpected status: LockedUp.
```
</details>

<details>
<summary>Console output (RUST_LOG=debug)</summary>

File is too big, so I attach it as a file:
[debug-output.log](https://github.com/user-attachments/files/15550356/output.log)

</details>

<details>
<summary>Console output (RUST_LOG=trace)</summary>

File is too big, so I attach it as a file:
[trace-output.log](https://github.com/user-attachments/files/15550387/trace-output.log)

</details>

**Desktop:**
- Linux Mint

**Additional context**
Tested with different projects, they all use `cortex-m-rt` and my library `emcell` for linker scripts generation.

skibon02, 2024-06-10:
I've just got it working witn `--disable-double-buffering` option.

bugadani, 2024-06-10:
Maybe we have incorrect memory regions for this chip, too. Do you have a datasheet for it?

skibon02, 2024-06-10:
yes, datasheet and reference manual are avaliable [here](https://www.arterychip.com/en/product/AT32F437.jsp)
(registration not required)

skibon02, 2024-06-10:
[image]

skibon02, 2024-06-10:
Here are values for flash_properties struct:
[image]

skibon02, 2024-06-10:
Looks like `AT32F4_Series.yaml` wasn't changed between 0.23 and 0.24 versions of probe-rs

bugadani, 2024-06-11:
THe problem is that these chips have user-configurable memory layouts. This issue must have started happening when we moved the flash algorithm stack to the end of memory. This can probably be fixed if we limit the stack size these devices use for flashing.

bugadani, 2024-06-11:
I think this issue can be solved by specifying `data_load_address` for these targets. The reason: we place data at the end of the memory, but we don't know if we can actually use that memory. However, we place the stack before the memory, so if the target forces `data_load_address` inside the first 128K, everything should be placed in the first 128K which should be safe RAM.

bugadani, 2024-06-11:
As I don't have one of these devices, please test https://github.com/probe-rs/probe-rs/pull/2561 for me.

skibon02, 2024-06-11:
Thanks for your help on this issue!
I've tested your fork with AT32F437VMT7 and it does indeed fix the flash programming issue.

skibon02, 2024-06-11:
Is IRAM1 range used for anything other than default `data_load_address` calculation? In yaml file the maximum possible RAM range is specified, but not default one, and there is a high probability for mismatch of range.end of `IRAM1` and actual end of RAM on MCU. For example, AT32F437 is marketed to have 512kB of RAM, but it is true when you manually reprogram user system data to select this mode. Default mode used in most cases includes only 384KB of RAM making accesses beyond that incorrect.

PR #2561 (2024-06-12): Specify data load address for AT32F4 devices
We currently place stack and data at the end of the memory to allow maximum stack use automatically. This causes problems for devices that have more exotic options like RAM that can be mapped to act as 0 wait cycle flash.

This PR will place the data blocks to the end of the smallest ram in the family. Stack is placed below data, and all flash algos should have around 4-5KB worth of stack.

Closes #2537
closed 2024-06-12 by Tiwalun