probe-rs/probe-rs #2541: Prerequisites are not correct

SergioGasquez, 2024-06-05:
**Describe the bug**
`cmake` is a required prerequisite.

**To Reproduce**
Steps to reproduce the behavior: Follow the instructions in https://probe.rs/docs/getting-started/installation/#installing-from-source-(cargo-install)

1. `sudo apt install -y pkg-config libudev-dev`
2. `cargo install probe-rs-tools@0.24.0`

**Expected behavior**
Installation succeeds. Installing `cmake` solves the issue, so updating the prerequisites to: `sudo apt install -y pkg-config libudev-dev cmake` solves it. 

There are several places that need to be updated:
- https://probe.rs/docs/getting-started/installation/#debian-based-linux
- https://github.com/probe-rs/probe-rs?tab=readme-ov-file#building
  - Here `pkg-config` is not listed either.
  
Additionally, having binaries for `armv7-unknown-linux-gnueabihf` would be very helpful for us. As building it from source in a RPi takes lots of time, I could open a feature request for this and help, unless there is any reason not to add this.

**Desktop (please complete the following information):**

- Raspbian 12 in RPi (`armv7-unknown-linux-gnueabihf`)

**Additional context**
<details open>
<summary>Full installation log</summary>

```
espressif@BrnoRPIRS03:/home/sergio $ cargo install probe-rs-tools@0.24.0
    Updating crates.io index
  Downloaded probe-rs-tools v0.24.0
  Downloaded 1 crate (202.9 KB) in 0.13s
  Installing probe-rs-tools v0.24.0
    Updating crates.io index
  Downloaded bzip2 v0.4.4
  Downloaded borsh-derive v1.5.1
  Downloaded ansi-parser v0.9.1
  Downloaded bytemuck_derive v1.7.0
  Downloaded aes v0.8.4
  Downloaded zstd-sys v2.0.10+zstd.1.5.6
  Downloaded probe-rs v0.24.0
  Downloaded linux-raw-sys v0.4.14
  Downloaded espflash v3.1.0
  Downloaded libz-ng-sys v1.1.15
  Downloaded zstd-safe v7.1.0
  Downloaded zstd v0.13.1
  Downloaded zip v1.1.4
  Downloaded zerocopy-derive v0.7.34
  Downloaded unicode-width v0.1.13
  Downloaded unicode-truncate v1.0.0
  Downloaded toml v0.8.14
  Downloaded thiserror-impl v1.0.61
  Downloaded thiserror v1.0.61
  Downloaded strum_macros v0.26.4
  Downloaded sha1 v0.10.6
  Downloaded serde_spanned v0.6.6
  Downloaded ratatui v0.26.3
  Downloaded probe-rs-target v0.24.0
  Downloaded probe-rs-mi v0.1.0
  Downloaded polling v3.7.1
  Downloaded pbkdf2 v0.12.2
  Downloaded parking_lot v0.12.3
  Downloaded num_enum_derive v0.7.2
  Downloaded num_enum v0.7.2
  Downloaded miniz_oxide v0.7.3
  Downloaded lzma-rs v0.3.0
  Downloaded libc v0.2.155
  Downloaded jobserver v0.1.31
  Downloaded insta v1.39.0
  Downloaded inout v0.1.3
  Downloaded hmac v0.12.1
  Downloaded figment v0.10.19
  Downloaded event-listener-strategy v0.5.2
  Downloaded event-listener v5.3.1
  Downloaded either v1.12.0
  Downloaded deflate64 v0.1.8
  Downloaded crossbeam-utils v0.8.20
  Downloaded crossbeam-channel v0.5.13
  Downloaded crc32fast v1.4.2
  Downloaded crc-catalog v2.4.0
  Downloaded crc v3.2.1
  Downloaded constant_time_eq v0.3.0
  Downloaded cmake v0.1.50
  Downloaded cipher v0.4.4
  Downloaded cfg_aliases v0.2.1
  Downloaded cc v1.0.98
  Downloaded bzip2-sys v0.1.11+1.0.8
  Downloaded borsh v1.5.1
  Downloaded async-lock v3.4.0
  Downloaded async-io v2.3.3
  Downloaded 56 crates (15.9 MB) in 3.13s (largest was `espflash` at 5.4 MB)
   Compiling proc-macro2 v1.0.85
   Compiling unicode-ident v1.0.12
   Compiling serde v1.0.203
   Compiling cfg-if v1.0.0
   Compiling libc v0.2.155
   Compiling version_check v0.9.4
   Compiling once_cell v1.19.0
   Compiling byteorder v1.5.0
   Compiling quote v1.0.36
   Compiling syn v2.0.66
   Compiling autocfg v1.3.0
   Compiling jobserver v0.1.31
   Compiling syn v1.0.109
   Compiling cc v1.0.98
   Compiling memchr v2.7.2
   Compiling pkg-config v0.3.30
   Compiling typenum v1.17.0
   Compiling generic-array v0.14.7
   Compiling equivalent v1.0.1
   Compiling hashbrown v0.14.5
   Compiling indexmap v2.2.6
   Compiling itoa v1.0.11
   Compiling thiserror v1.0.61
   Compiling ahash v0.8.11
   Compiling fnv v1.0.7
   Compiling bitflags v2.5.0
   Compiling ryu v1.0.18
   Compiling zerocopy v0.7.34
   Compiling toml_datetime v0.6.6
   Compiling rustversion v1.0.17
   Compiling winnow v0.5.40
   Compiling crypto-common v0.1.6
   Compiling aho-corasick v1.1.3
   Compiling num-traits v0.2.19
   Compiling serde_derive v1.0.203
   Compiling thiserror-impl v1.0.61
   Compiling crossbeam-utils v0.8.20
   Compiling smallvec v1.13.2
   Compiling regex-syntax v0.8.3
   Compiling allocator-api2 v0.2.18
   Compiling cmake v0.1.50
   Compiling heck v0.5.0
   Compiling pin-project-lite v0.2.14
   Compiling libz-ng-sys v1.1.15
   Compiling regex-automata v0.4.6
   Compiling log v0.4.21
   Compiling regex v1.10.4
   Compiling block-buffer v0.10.4
   Compiling subtle v2.5.0
   Compiling lazy_static v1.4.0
   Compiling unicode-width v0.1.13
   Compiling digest v0.10.7
   Compiling tracing-core v0.1.32
error: failed to run custom build command for `libz-ng-sys v1.1.15`

Caused by:
  process didn't exit successfully: `/tmp/cargo-installMunk5k/release/build/libz-ng-sys-8af4ffc2d5da5254/build-script-build_zng` (exit status: 101)
  --- stdout
  CMAKE_TOOLCHAIN_FILE_armv7-unknown-linux-gnueabihf = None
  CMAKE_TOOLCHAIN_FILE_armv7_unknown_linux_gnueabihf = None
  HOST_CMAKE_TOOLCHAIN_FILE = None
  CMAKE_TOOLCHAIN_FILE = None
  CMAKE_GENERATOR_armv7-unknown-linux-gnueabihf = None
  CMAKE_GENERATOR_armv7_unknown_linux_gnueabihf = None
  HOST_CMAKE_GENERATOR = None
  CMAKE_GENERATOR = None
  CMAKE_PREFIX_PATH_armv7-unknown-linux-gnueabihf = None
  CMAKE_PREFIX_PATH_armv7_unknown_linux_gnueabihf = None
  HOST_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_armv7-unknown-linux-gnueabihf = None
  CMAKE_armv7_unknown_linux_gnueabihf = None
  HOST_CMAKE = None
  CMAKE = None
  running: cd "/tmp/cargo-installMunk5k/release/build/libz-ng-sys-edcf3cd7fcb80358/out/build" && CMAKE_PREFIX_PATH="" "cmake" "/home/espressif/.cargo/registry/src/index.crates.io-1cd66030c949c28d/libz-ng-sys-1.1.15/src/zlib-ng" "-DBUILD_SHARED_LIBS=OFF" "-DZLIB_COMPAT=OFF" "-DZLIB_ENABLE_TESTS=OFF" "-DWITH_GZFILEOP=ON" "-DCMAKE_INSTALL_PREFIX=/tmp/cargo-installMunk5k/release/build/libz-ng-sys-edcf3cd7fcb80358/out" "-DCMAKE_C_FLAGS= -ffunction-sections -fdata-sections -fPIC -march=armv7-a -mfpu=vfpv3-d16" "-DCMAKE_C_COMPILER=/usr/bin/cc" "-DCMAKE_CXX_FLAGS= -ffunction-sections -fdata-sections -fPIC -march=armv7-a -mfpu=vfpv3-d16" "-DCMAKE_CXX_COMPILER=/usr/bin/c++" "-DCMAKE_ASM_FLAGS= -ffunction-sections -fdata-sections -fPIC -march=armv7-a -mfpu=vfpv3-d16" "-DCMAKE_ASM_COMPILER=/usr/bin/cc" "-DCMAKE_BUILD_TYPE=Release"

  --- stderr
  thread 'main' panicked at /home/espressif/.cargo/registry/src/index.crates.io-1cd66030c949c28d/cmake-0.1.50/src/lib.rs:1098:5:

  failed to execute command: No such file or directory (os error 2)
  is `cmake` not installed?

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: failed to compile `probe-rs-tools v0.24.0`, intermediate artifacts can be found at `/tmp/cargo-installMunk5k`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
```

</details>

bugadani, 2024-06-05:
I don't think we depend on libz-ng-sys anymore, so at least that part of the problem should be gone. Could you please try installing from git?

The rest is a constant headache 😅

SergioGasquez, 2024-06-05:
> Could you please try installing from git?

I guess you mean without having `cmake` installed, right? I'll probably try it tomorrow morning as, as I mentioned, it takes 30/40 minutes to install `probe-rs` and during that time we cant merge anything into `esp-hal`...

bugadani, 2024-06-05:
Fair enough, although I don't need the process to complete just to know if it tries to compile libz-ng-sys or gets farther.

Regarding prebuilts, we're supposed to have binaries for `aarch64-unknown-linux-gnu` which is supposed to be compatible with raspberries, but we build with an incompatible glibc unfortunately and cargo-dist isn't smart enough to realize this. We've merged https://github.com/probe-rs/probe-rs/pull/2502 that hopefully will result in an actually running binary.

SergioGasquez, 2024-06-06:
Just did the test and I can confirm that `cmake` is not needed in `main`:

<details>
<summary>Full Log</summary>

```
espressif@BrnoRPIRS03:/home/sergio $ sudo apt remove cmake
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  cmake-data libjsoncpp25 librhash0 libuv1
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  cmake
0 upgraded, 0 newly installed, 1 to remove and 28 not upgraded.
After this operation, 23.3 MB disk space will be freed.
Do you want to continue? [Y/n] y
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = "UTF-8",
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
(Reading database ... 70791 files and directories currently installed.)
Removing cmake (3.25.1-1) ...
Processing triggers for man-db (2.11.2-2) ...
espressif@BrnoRPIRS03:/home/sergio $ cmake --version
bash: cmake: command not found
espressif@BrnoRPIRS03:/home/sergio $ cargo install probe-rs-tools --git https://github.com/probe-rs/probe-rs
    Updating git repository `https://github.com/probe-rs/probe-rs`
  Installing probe-rs-tools v0.24.0 (https://github.com/probe-rs/probe-rs#cf3b4181)
    Updating crates.io index
  Downloaded zip v2.1.3
  Downloaded object v0.36.0
  Downloaded 2 crates (418.4 KB) in 0.67s
   Compiling proc-macro2 v1.0.85
   Compiling unicode-ident v1.0.12
   Compiling serde v1.0.203
   Compiling cfg-if v1.0.0
   Compiling libc v0.2.155
   Compiling version_check v0.9.4
   Compiling once_cell v1.19.0
   Compiling byteorder v1.5.0
   Compiling autocfg v1.3.0
   Compiling syn v1.0.109
   Compiling quote v1.0.36
   Compiling memchr v2.7.2
   Compiling syn v2.0.66
   Compiling itoa v1.0.11
   Compiling thiserror v1.0.61
   Compiling ahash v0.8.11
   Compiling bitflags v2.5.0
   Compiling fnv v1.0.7
   Compiling ryu v1.0.18
   Compiling zerocopy v0.7.34
   Compiling rustversion v1.0.17
   Compiling crossbeam-utils v0.8.20
   Compiling hashbrown v0.14.5
   Compiling allocator-api2 v0.2.18
   Compiling equivalent v1.0.1
   Compiling indexmap v2.2.6
   Compiling aho-corasick v1.1.3
   Compiling typenum v1.17.0
   Compiling pin-project-lite v0.2.14
   Compiling heck v0.5.0
[... 233 lines cut ...]
   Compiling capstone v0.12.0
   Compiling insta v1.39.0
   Compiling svd-parser v0.14.5
   Compiling addr2line v0.22.0
   Compiling indicatif v0.17.8
   Compiling rustyline v14.0.0
   Compiling libtest-mimic v0.7.3
   Compiling ratatui v0.26.3
   Compiling goblin v0.8.2
   Compiling tracing-appender v0.2.3
   Compiling cargo_metadata v0.18.1
   Compiling itm v0.9.0-rc.1
   Compiling directories v5.0.1
   Compiling probe-rs-mi v0.1.0 (/home/espressif/.cargo/git/checkouts/probe-rs-92d520f86d5696fc/cf3b418/probe-rs-mi)
   Compiling ansi-parser v0.9.1
   Compiling clap_complete v4.5.2
   Compiling cargo-config2 v0.1.26
   Compiling sanitize-filename v0.5.0
   Compiling termtree v0.4.1
   Compiling fastrand v2.1.0
   Compiling bytesize v1.3.0
   Compiling urlencoding v2.1.3
    Building [=======================> ] 397/398: probe-rs(bin)
```
</details>

Vollbrecht, 2024-06-06:
>  we're supposed to have binaries for aarch64-unknown-linux-gnu which is supposed to be compatible with raspberries

that is only true for the current raspberry pi's running the 64bit version. Many sbc like the 3b+ and other uses the thumbv7 targets

bugadani, 2024-06-06:
> > we're supposed to have binaries for aarch64-unknown-linux-gnu which is supposed to be compatible with raspberries
> 
> that is only true for the current raspberry pi's running the 64bit version. Many sbc like the 3b+ and other uses the thumbv7 targets

Well I've been happily running aarch64-unknown-linux-gnu binaries on my 3b+ so I guess that's not entirely correct :) thumbv7 is mainly cortex-m, though, even though the cortex-a cpus eat it I don't think I've seen software commonly compiled for that.

bugadani, 2024-06-06:
One more thing: you can also [cross-compile](https://probe.rs/docs/library/crosscompiling/#using-cross) probe-rs.
cargo-dist doesn't let us cross-compile yet, so it's unlikely we can release prebuilts for anything other than aarch64, and considering their installer issue it's not a smooth experience either.

Vollbrecht, 2024-06-06:
for my teleprobe fork that worked with an older version of probe-rs, cross was the only options since i had not enough ram on a 3b+ to compile it natively ;D 

But yeah you seams to be right, i double checked my cross images and they were using aarch64, but yeah it only works if you are using a current 64bit OS otherwise you still would need something like "armv7-unknown-linux-gnueabihf".

I think the two pain points at that time where the vendored openssl + getting an old enough libc compilation. Musl never worked for me here.

SergioGasquez, 2024-06-11:
Since https://github.com/probe-rs/webpage/pull/133 got merged, I think this issue can be closed. If we need to further discuss the binaries, we can open another issue for it.

closed 2024-06-11 by SergioGasquez