probe-rs/probe-rs #2544: Got "Unexpected semihosting command" when running embedded-test on NUCLEO-H743ZI

mahu-wm, 2024-06-05:
**Describe the bug**

Running an embedded-test on a NUCLEO-H743ZI with probe-rs (0.24.0 and master 5800d517d97caa61763468c56a7f31f4ab104c71) reports "Error: Unexpected semihosting command".

Evaluation of the trace (--log-file) did indicate, that in test_run_mode.rs/list_tests() did receive the GetCommandLine request. From the Error I assume, that somehow the operation of the following SemihostingCommand was set to 0.

**To Reproduce**

Steps to reproduce the behavior:

1. Install probe-rs 0.24.0 (or 5800d517d97caa61763468c56a7f31f4ab104c71)
2. Connect NUCLEO-H743ZI
3. Use embedded-test to create a test binary and run it via probe-rs. You can run `cargo test --example_test` in [this example project](https://github.com/weinmann-emt/embedded-test-example/tree/nucleo-h743zi). Or `probe-rs run --chip STM32H743ZITx --list <ELF>`

**Expected behavior**

Running of the tests and reporting failure/success.

**Stacktrace**

This is the output of `probe-rs run --chip STM32H743ZITx <ELF>`, after successful programming:

```
Frame 0: syscall_readonly @ 0x080085fa
       ~\.cargo\registry\src\index.crates.io-6f17d22bba15001f\semihosting-0.1.10\src\sys\arm_compat\syscall/arm.rs:88:9
Frame 1: sys_exit_extended @ 0x080086cc
       ~\.cargo\registry\src\index.crates.io-6f17d22bba15001f\semihosting-0.1.10\src\sys\arm_compat/mod.rs:198:2
Frame 2: <unknown function @ 0x00020026> @ 0x00020026
Error: Unexpected semihosting command Unknown(UnknownCommandDetails { operation: 0, parameter: 536996204 }) cmdline_requested: true
```

**Desktop (please complete the following information):**

- Windows 10

**Additional context**

The problem disappears when probe-rs is downgraded to 204a9f7c30b7196b0d85eecf5782c772a49b1254 of pull request #2292. The following two commits (a4789d542805330f55a20a08451c8b6af94fc34e and 3b6a3b189206f57a10fc3c460419ca5fe9dc2fc1) seem to cause the problem. When these are reverted on current master (https://github.com/weinmann-emt/probe-rs/tree/fix-unexpected-semihosting-command), probe-rs testing works as expected.

In the pull request, it is mentioned that Xtensa and RISC-V targets were tested, probably this bug exists only for cortex-m?

closed 2024-07-01 by bugadani

t-moe, 2024-07-01:
hi @mahu-wm ,
sorry we didn't reach out earlier. Can you check whether the issue is fixed on the master branch?

mahu-wm, 2024-07-01:
Thanks, yes, the issue is fixed with the current master (c3a767bdc0827f3b4c5db6d6628c2aa7d7730023)