probe-rs/probe-rs #2546: HIL tests fail on H2 with probe-rs 0.24.0

SergioGasquez, 2024-06-06:
**Describe the bug**
`ecc` tests are failing in a weird way in our custom runners (RPi): https://github.com/esp-rs/esp-hal/actions/runs/9363281580/job/25832239222. Running the test suite locally on my machine works fine.

**To Reproduce**
Steps to reproduce the behavior:

1. `cargo install probe-rs-tools@0.24.0`
2. Run [HIL workflow](https://github.com/esp-rs/esp-hal/actions/workflows/hil.yml)
3. Test fails: https://github.com/esp-rs/esp-hal/actions/runs/9363281580/job/25832239222

**Expected behavior**
Test succeeds.

**Desktop (please complete the following information):**

- Raspbian 12 in RPi (`armv7-unknown-linux-gnueabihf`)

**Additional context**
- If I execute the test suite on my machine, it does not fail
- All the other targets (C3, C6, S2 and S3) work fine with `probe-rs@0.24.0`. Only H2 fails, and always in the same way.
- The same error happens with `main` (rev `cf3b418`): https://github.com/esp-rs/esp-hal/actions/runs/9398579732

bugadani, 2024-06-06:
Please retry with https://github.com/probe-rs/probe-rs/pull/2547

SergioGasquez, 2024-08-12:
Using `bba1bb5` works fine in H2, so I'll mark the issue as resolved!

closed 2024-08-12 by SergioGasquez

bugadani, 2024-08-12:
Interesting, I'm not aware of relevant changes, but I'll take it :D