probe-rs/probe-rs #2557: SWD support for FTDI adapter based debugging

Roquelobster, 2024-06-09:
**Is your feature request related to a problem? Please describe.**
I fired up my FTDI cable (C232HM-DDHSL-0 -- 0403:6014:FTUJK2JU (FTDI)) connected to my RP2040 but it gives the error that SWD isn't implemented and I can see the swd_io() function in the FTDI module is just a stub returning the unsupported error.

**Describe the solution you'd like**
To have SWD support for FTDI devices.

**Describe alternatives you've considered**
-

**Additional context**
-

bugadani, 2024-06-10:
"FTDI devices" is impossible because SWD support needs external hardware. We don't currently have configuration options for probe drivers (e.g. we can't tell the driver which pin is SWD direction select). What we can do is collect a few popular implementations and create a custom driver for them - but since I don't have any, I don't think I'll be able to do this.

bugadani, 2024-06-11:
The resistor method may work but is not universal. For example, https://github.com/probe-rs/probe-rs/issues/2403 has an external hardware that uses something else built into the probe itself. I would like to avoid adding 3 different drivers for the different FTDI-based probes.

Roquelobster, 2024-06-16:
Understood, which single FTDI based probe do you think is a candidate?

mpictor, 2024-06-26:
KT-Link is based on the FTDI 2232 and supports SWD.

PID/VID are 0x0403 and 0xBBE2, respectively. [source](https://github.com/openocd-org/openocd/blob/ad87fbd1cf28760795c4e18f3318a2d720e5a8a6/tcl/interface/ftdi/kt-link.cfg#L11)

I'm not sure if kristech still sells their device, but Dangerous Prototypes' BusBlaster v2 and later are inexpensive and can emulate KT-Link including SWD. 

Note that emulation isn't complete - the real KT-Link can do SWD _and_ SWV, while BBv2 is limited to the former.  It's possible that later versions of BusBlaster can do SWV; mine's v2 and I haven't dug into the capabilities of later versions.

----

Some way of configuring probe-rs to work with unknown hardware would be nice, though that seems like a fairly complex configuration that would need to be implemented... :thinking: