probe-rs/probe-rs #2563: Problem with reset of Cortex-M4 based  MCU

robamu, 2024-06-12:
Hi,

I am trying to make probe-rs work for a Cortex-M4 based MCU. I managed to generate the yaml file from the vendor provided CMSIS pack, but I have issues flashing the device.

I now traced down the issue to not being able to reset the MCU properly. The vendor only supported Keil MDK, so debugging this is a bit tricky, especially because I am not an expert in low-level ARM access port handling.

I have an application like this:

```rust
use std::fs::File;
use std::path::Path;
use std::time::Duration;

use probe_rs::config::add_target_from_yaml;
//use probe_rs::flashing::DownloadOptions;
use probe_rs::probe::list::Lister;
use probe_rs::{flashing, MemoryInterface, Permissions, Session};

fn main() {
    tracing_subscriber::fmt::init();

    let lister = Lister::new();

    // Get a list of all available debug probes.
    let probes = lister.list_all();

    // Use the first probe found.
    let probe = probes[0].open().expect("failed to open probe");
    println!("probe: {:?}", probe);

    add_target_from_yaml(
        File::open(Path::new("../va416xx-rs/scripts/VA416xx_Series.yaml"))
            .expect("opening VA416xx description file failed"),
    )
    .expect("failed to add VA416xx target from description file");

    let mut session = probe
        .attach("VA416xx", Permissions::default())
        .expect("failed to attach to chip");

    // Select a core.
    let mut core = session.core(0).unwrap();

    // Disable ROM protection
    core.write_32(...).unwrap();

    //core.reset().unwrap();
    core.reset_and_halt(Duration::from_millis(2000)).unwrap();
    // drop(core);
}
```

I can see that AP accesses are possible when using DEBUG tracelevel. I also get 100000 lines of code when using `RUST_LOG=debug`.. are there some smart ways to reduce this? I am a total beginner when it becomes to tracing, so some tips would be appreciated

In any case, I think the reset fails because the command is not acknowledged. Maybe this sounds really stupid, but could it be that the reset commands is immediately executed without an acknowledgement being issue? If this is the case,
is there some simple way to skip the reply check for the reset command? Unless I can make this simple reset command work, I can not really continue.

I managed to flash the board with Keil MDK and with a combination of a JLink GDB server and gdb-multiarch without any issues.

`probe-rs info` output:

```
va416xx-rs on  main [$!?] via 🦀 v1.78.0 took 14s
❯ probe-rs info
Probing target via JTAG

Error identifying target using protocol JTAG: An error with the usage of the probe occurred

Probing target via SWD

ARM Chip with debug port Default:
Debug Port: DPv1, DP Designer: ARM Ltd
└── 0 MemoryAP
    └── ROM Table (Class 1), Designer: ARM Ltd
        ├── Cortex-M4 SCS   (Generic IP component)
        │   └── CPUID
        │       ├── IMPLEMENTER: ARM Ltd
        │       ├── VARIANT: 0
        │       ├── PARTNO: Cortex-M4
        │       └── REVISION: 1
        ├── Cortex-M3 DWT   (Generic IP component)
        └── Cortex-M3 FBP   (Generic IP component)

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.

```

Partial Keil Output when resetting for flashing:

```
* JLink Info: Found SW-DP with ID 0x2BA01477
* JLink Info: DPIDR: 0x2BA01477
* JLink Info: Scanning AP map to find all available APs
* JLink Info: AP[1]: Stopped AP scan as end of AP map has been reached
* JLink Info: AP[0]: AHB-AP (IDR: 0x24770011)
* JLink Info: Iterating through AP map to find AHB-AP to use
* JLink Info: AP[0]: Core found
* JLink Info: AP[0]: AHB-AP ROM base: 0xE00FF000
* JLink Info: CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
* JLink Info: Found Cortex-M4 r0p1, Little endian.
* JLink Info: FPUnit: 6 code (BP) slots and 2 literal slots
* JLink Info: CoreSight components:
* JLink Info: ROMTbl[0] @ E00FF000
* JLink Info: ROMTbl[0][0]: E000E000, CID: B105E00D, PID: 000BB00C SCS-M7
* JLink Info: ROMTbl[0][1]: E0001000, CID: B105E00D, PID: 003BB002 DWT
* JLink Info: ROMTbl[0][2]: E0002000, CID: B105E00D, PID: 002BB003 FPB
ROMTableAddr = 0xE00FF000
* JLink Info: Reset: Halt core after reset via DEMCR.VC_CORERESET.
* JLink Info: Reset: Reset device via AIRCR.SYSRESETREQ.
* JLink Info: Reset: Core did not halt after reset, trying to disable WDT.
* JLink Info: Reset: Halt core after reset via DEMCR.VC_CORERESET.
* JLink Info: Reset: Reset device via AIRCR.SYSRESETREQ.
* JLink Info: AfterResetTarget() start
* JLink Info: AfterResetTarget()
* JLink Info: AfterResetTarget() end
```

Output when running the JLink GDB server:

```
va416xx-rs on  main [$!?] via 🦀 v1.78.0
❯ ./jlink-gdb.sh
SEGGER J-Link GDB Server V7.96k Command Line Version

JLinkARM.dll V7.96k (DLL compiled Jun  5 2024 13:26:06)

Command line: -select USB -device Cortex-M4 -if SWD -speed 2000 -LocalhostOnly -vd -jlinkscriptfile ./jlink/JLinkSettings.JLinkScript
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               on
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 ./jlink/JLinkSettings.JLinkScript
J-Link settings file:          none
------Target related settings------
Target device:                 Cortex-M4
Target device parameters:      none
Target interface:              SWD
Target interface speed:        2000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link OB-STM32F103 V1 compiled Oct 30 2023 12:12:03
Hardware: V1.00
S/N: 660016022
Checking target voltage...
Target voltage: 3.30 V
Listening on TCP/IP port 2331
Connecting to target...
Halting core...
Connected to target
Waiting for GDB connection...
```

A special JLink script is necessary for direct flashing into RAM, but I do not think this information is relevant when I can not even make a simple reset work.. Unfortunately, both Keil and SEGGER tools are proprietary, so I can not really check what they do.

Kind Regards
Robin

noppej, 2024-06-12:
Your description is very detailed, but I don't see that you've included the actual command and error message from `probe-rs` ... did I just miss that?

To filter RUST_LOG, instead of setting the value to `debug`, try setting it to `module=debug`, where module is one of the modules reported in the logs when you see the error, e.g.  `probe_rs::flashing=debug`

robamu, 2024-06-12:
Yeah, you're right. I'll include the output above, but without the spam (logging level INFO).

robamu, 2024-06-13:
I have adapted the `cortex_m_reset_system` function like this so that the application will continue even if the AIRCR write does not work properly:

```rust
/// ResetSystem for Cortex-M devices
fn cortex_m_reset_system(interface: &mut dyn ArmProbe) -> Result<(), ArmError> {
    use crate::architecture::arm::core::armv7m::{Aircr, Dhcsr};

    let mut aircr = Aircr(0);
    aircr.vectkey();
    aircr.set_sysresetreq(true);

    if let Err(e) = interface.write_word_32(Aircr::get_mmio_address(), aircr.into()) {
        tracing::warn!("resetting the system returned with ARM error: {:?}", e);
    }

    let start = Instant::now();

    while start.elapsed() < Duration::from_millis(5000) {
        let dhcsr = match interface.read_word_32(Dhcsr::get_mmio_address()) {
            Ok(val) => Dhcsr(val),
            // Some combinations of debug probe and target (in
            // particular, hs-probe and ATSAMD21) result in
            // register read errors while the target is
            // resetting.
            Err(ArmError::AccessPort {
                source: AccessPortError::RegisterRead { .. },
                ..
            }) => continue,
            Err(ArmError::AccessPort {
                source: AccessPortError::RegisterWrite { .. },
                ..
            }) => continue,
            Err(err) => return Err(err),
        };
        println!("read dhcsr: {:#?}", dhcsr);

        // Wait until the S_RESET_ST bit is cleared on a read
        if !dhcsr.s_reset_st() {
            return Ok(());
        }
    }

    Err(ArmError::Timeout)
}

```

I now simply run into a timeout. Reading any register directly before writing to AIRCR before the reset works without any issues. Considering that it always comes that far, something else probably has to be done to be able to write to/ read from the registers again?

robamu, 2024-06-13:
I am now completely unable to flash anything even with the JLinkGDB server. Even repeatedly power-cycling the hardware  does not help..

robamu, 2024-06-28:
I opened the J-Link GDB server with the `-Log` option and received the following output:

```txt
SEGGER J-Link GDB Server V7.96k LogFile
Logging started @ 2024-06-28 23:25
03-00000000-00-00000000-01B4: -----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               on
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms

03-00000000-00-00000000-018D: ------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 ./jlink/JLinkSettings.JLinkScript
J-Link settings file:          none
------Target related settings------
Target device:                 Cortex-M4
Target device parameters:      none
Target interface:              SWD
Target interface speed:        2000kHz
Target endian:                 little

03-00000000-00-00000000-0001: 

03-00000000-00-00000000-0018: Connecting to J-Link...

02-00000000-00-00000012-0056: T533FE640 000:017.881   Firmware: J-Link OB-STM32F103 V1 compiled Oct 30 2023 12:12:03
02-00000000-00-00000012-0056: T533FE640 000:018.415   Firmware: J-Link OB-STM32F103 V1 compiled Oct 30 2023 12:12:03
02-00000000-00-00000014-0027: T533FE640 000:019.967   Hardware: V1.00
02-00000000-00-00000014-0026: T533FE640 000:019.979   S/N: 660016022
02-00000000-00-00000014-0023: T533FE640 000:019.986   OEM: SEGGER
02-00000000-00-00000014-0028: T533FE640 000:019.992   Feature(s): None
02-00000000-00-00000014-002F: T533FE640 000:020.384   Bootloader: 2013 Nov 11
02-00000000-00-00000014-003E: T533FE640 000:020.689   USB speed mode: Full speed (12 MBit/s)
02-00000000-00-00000015-0043: T533FE640 000:020.992   TELNET listener socket opened on port 19021
02-00000000-00-00000015-004A: T533FE640 000:021.134   WEBSRV WEBSRV_Init(): Starting webserver thread(s)
02-00000000-00-00000015-0044: T533FE640 000:021.250   WEBSRV Webserver running on local port 19080
02-00000000-00-00000015-0067: T533FE640 000:021.285   Looking for J-Link GUI Server exe at: /opt/SEGGER/JLink_V796k/JLinkGUIServerExe
02-00000000-00-00000015-005C: T533FE640 000:021.312   Forking J-Link GUI Server: /opt/SEGGER/JLink_V796k/JLinkGUIServerExe
02-00000000-00-00000061-004B: T533FE640 000:067.850   J-Link GUI Server info: "J-Link GUI server V7.96k "
02-00000000-00-00000064-002F: T533FE640 000:070.080 - 63.917ms returns "O.K."
03-00000000-00-00000064-0015: J-Link is connected.

02-00000000-00-00000064-0044: T533FE640 000:070.119 JLINK_DEVICE_GetIndex(sDeviceName = Cortex-M4)
02-00000000-00-00000092-002B: T533FE640 000:098.533 - 28.413ms returns 21
02-00000000-00-00000092-003C: T533FE640 000:098.574 JLINK_DEVICE_GetInfo(DeviceIndex = 21)
02-00000000-00-00000092-0029: T533FE640 000:098.579 - 0.005ms returns 0
02-00000000-00-00000092-0044: T533FE640 000:098.620 JLINK_ExecCommand("device = Cortex-M4", ...). 
02-00000000-00-00000095-0034: T533FE640 000:100.920   Device "CORTEX-M4" selected.
02-00000000-00-00000097-002C: T533FE640 000:103.158 - 4.532ms returns 0x00
02-00000000-00-00000097-0060: T533FE640 000:103.177 JLINK_ExecCommand("ScriptFile = ./jlink/JLinkSettings.JLinkScript", ...). 
02-00000000-00-00000097-002C: T533FE640 000:103.253 - 0.076ms returns 0x00
02-00000000-00-00000097-0032: T533FE640 000:103.264 JLINK_GetFirmwareString(...)
02-00000000-00-00000097-001F: T533FE640 000:103.267 - 0.004ms
03-00000000-00-00000097-003F: Firmware: J-Link OB-STM32F103 V1 compiled Oct 30 2023 12:12:03

02-00000000-00-00000097-0030: T533FE640 000:103.283 JLINK_GetHardwareVersion()
02-00000000-00-00000097-002D: T533FE640 000:103.286 - 0.003ms returns 10000
03-00000000-00-00000097-0010: Hardware: V1.00

02-00000000-00-00000097-0023: T533FE640 000:103.295 JLINK_GetSN()
02-00000000-00-00000097-0031: T533FE640 000:103.299 - 0.004ms returns 660016022
02-00000000-00-00000097-002D: T533FE640 000:103.302 JLINK_GetOEMString(...)
[... 85 lines cut ...]
03-00000000-00-00000112-0010: Halting core...

02-00000000-00-00000112-0022: T533FE640 000:118.640 JLINK_Halt()
02-00000000-00-00000116-002C: T533FE640 000:122.453 - 3.812ms returns 0x00
02-00000000-00-00000116-0038: T533FE640 000:122.466 JLINK_GetDebugInfo(0x01 = Unknown)
02-00000000-00-00000116-0028: T533FE640 000:122.470   Value=0x00000001
02-00000000-00-00000116-0029: T533FE640 000:122.473 - 0.006ms returns 0
02-00000000-00-00000118-0026: T533FE640 000:124.749 JLINK_IsHalted()
02-00000000-00-00000118-002C: T533FE640 000:124.760 - 0.011ms returns TRUE
03-00000000-00-00000118-0014: Connected to target

03-00000000-00-00000118-001D: Waiting for GDB connection...
02-00000000-00-00002201-002C: T52BFD640 002:206.939 JLINK_GetHWStatus(...)
02-00000000-00-00002201-0029: T52BFD640 002:207.275 - 0.336ms returns 0
03-00000000-00-00002533-0038: Restoring target state and closing J-Link connection...

02-00000000-00-00002533-0024: T533FE640 002:539.329 JLINK_IsOpen()
02-00000000-00-00002533-002C: T533FE640 002:539.341 - 0.012ms returns 0x01
02-00000000-00-00002543-0023: T533FE640 002:549.411 JLINK_Close()
02-00000000-00-00002543-0039: T533FE640 002:549.622   CPU_ReadMem(4 bytes @ 0xE0001000)
03-00000000-00-00002555-0011: Shutting down...

```

not sure if that helps in any way though.. I don't see any reset command, just a Halt command?

robamu, 2024-06-30:
Hello,

The flash process works without issues if I remove the `reset_and_halt` method call at https://github.com/probe-rs/probe-rs/blob/97fe2a74d464e24ff1314fcb7f6fec217562e7ef/probe-rs/src/flashing/flasher.rs#L164 .

I built a new local version of the probe-rs-tools locally and I was able to flash the MCU, but a second reset after
the flash process is now problematic. Resetting the device seems to be problematic in general. When I look at the cortex-debug configuration I managed to make work, it looks like this:

```json
        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug LED Blinky",
            "servertype": "jlink",
            "jlinkscript": "${workspaceFolder}/jlink/JLinkSettings.JLinkScript",
            "cwd": "${workspaceRoot}",
            "device": "Cortex-M4",
            "svdFile": "${workspaceFolder}/va416xx/svd/va416xx.svd.patched",
            "preLaunchTask": "blinky-example",
            "overrideLaunchCommands": [
                "monitor halt",
                "monitor reset",
                "load",
            ],
            "executable": "${workspaceFolder}/target/thumbv7em-none-eabihf/debug/examples/blinky",
            "interface": "swd",
            "runToEntryPoint": "main",
            "rttConfig": {
                "enabled": true,
                // Have to use exact address unfortunately. "auto" does not work for some reason..
                "address": "0x1fff8000",
                "decoders": [
                    {
                        "port": 0,
                        "type": "console"
                    }
                ]
            }
        },
```

where I have also overriden the default launch commands-

robamu, 2024-06-30:
Okay, this is really tricky. I executed the following test application:

```rust
use std::fs::File;
use std::path::Path;
use std::time::Duration;

use probe_rs::config::add_target_from_yaml;
use probe_rs::probe::list::Lister;
use probe_rs::probe::DebugProbeInfo;
use probe_rs::{flashing, CoreInterface, MemoryInterface, Permissions, Session};

fn main() {
    tracing_subscriber::fmt::init();

    let lister = Lister::new();

    // Get a list of all available debug probes.
    let probes = lister.list_all();
    va416xx_test(probes);
    let probes = lister.list_all();
    reconnect_and_read_rtt_block(probes);
}

fn va416xx_test(probes: Vec<DebugProbeInfo>) {
    // Use the first probe found.
    let probe = probes[0].open().expect("failed to open probe");

    add_target_from_yaml(
        File::open(Path::new("../va416xx-rs/scripts/VA416xx_Series.yaml"))
            .expect("opening VA416xx description file failed"),
    )
    .expect("failed to add VA416xx target from description file");

    let mut session = probe
        .attach("VA416xx", Permissions::default())
        .expect("failed to attach to chip");

    // Select a core.
    let mut core = session.core(0).unwrap();

    // Disable ROM protection
    core.write_32(0x4001_0010, &[0x000_0001]).unwrap();
    // Disable watchdog
    // WDOGLOCK = 0x1ACCE551
    core.write_32(0x400210C0, &[0x1ACCE551]).unwrap();
    // WDOGCONTROL = 0x0 (diable)
    core.write_32(0x40021008, &[0]).unwrap();

    core.halt(Duration::from_millis(200))
        .expect("halting processor failed");

    drop(core);

    flashing::download_file(
        &mut session,
        "../va416xx-rs/target/thumbv7em-none-eabihf/debug/examples/rtt-log",
        flashing::Format::Elf,
    )
    .expect("downloading file failed");

    let mut core = session.core(0).unwrap();
    core.reset_and_halt(Duration::from_millis(200));
}

fn reconnect_and_read_rtt_block(probes: Vec<DebugProbeInfo>) {
    // Use the first probe found.
    let probe = probes[0].open().expect("failed to open probe");

    add_target_from_yaml(
        File::open(Path::new("../va416xx-rs/scripts/VA416xx_Series.yaml"))
            .expect("opening VA416xx description file failed"),
    )
    .expect("failed to add VA416xx target from description file");

    let mut session = probe
        .attach("VA416xx", Permissions::default())
        .expect("failed to attach to chip");

    // Select a core.
    let mut core = session.core(0).unwrap();

    let mut rtt_block: [u8; 16] = [0; 16];
    core.read_8(0x1fff_8000, &mut rtt_block).unwrap();
    println!("read rtt block: {:?}", rtt_block);
    let rtt_str = core::str::from_utf8(&rtt_block);
    println!("RTT string: {:?}", rtt_str);
}

```

With this, I am able to flash the chip and read the RTT block:

```txt
probe-rs-testing on  va416xx-testing [$!] is 📦 v0.1.0 via 🦀 v1.78.0
❯ cargo run

    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.09s
     Running `target/debug/probe-rs-testing`
2024-06-30T14:56:42.430624Z  WARN probe_rs::session: Could not clear all hardware breakpoints: An ARM specific error occurred.

Caused by:
    0: Error using access port.
    1: Failed to write register CSW at address 0x00000000
    2: An error occurred in the communication with an access port or debug port.
    3: Target device did not respond to request.
2024-06-30T14:56:42.803858Z  WARN probe_rs::session: Failed to deconfigure device during shutdown: An ARM specific error occurred.

Caused by:
    0: Error using access port.
    1: Failed to write register CSW at address 0x00000000
    2: An error occurred in the communication with an access port or debug port.
    3: Target device did not respond to request.
2024-06-30T14:56:43.481163Z  WARN probe_rs::architecture::arm::communication_interface: Failed to stop DP Default
read rtt block: [83, 69, 71, 71, 69, 82, 32, 82, 84, 84, 0, 0, 0, 0, 0, 0]
RTT string: Ok("SEGGER RTT\0\0\0\0\0\0")

```

The errors that happen here come from the `reset_and_halt`, but I am not unwrapping here. I do the whole re-connection process again to be able to read the memory again.

bugadani, 2024-06-30:
> I do the whole re-connection process again to be able to read the memory again.

This suggests that resetting the chip breaks the debug connection. You can work around this by implementing a debug sequence, where `reset_system` executes the reset then returns `ArmError::ReAttachRequired` (which I don't know if we handle currently!) or does the reconnection itself.

robamu, 2024-06-30:
Thanks, it works now! :) I'll prepare a PR with vendor specific support.

closed 2024-06-30 by bugadani