probe-rs/probe-rs #2567: Connection to a Cortex-M0 based Chip fails

robamu, 2024-06-15:
I am trying to probe information for a Cortex-M0 based MCU using `probe-rs`.

I tried running `probe-rs info` for this but there seems to be some communication issue. I can connect with a JLinkServer without issues.

JLinkServer Output of command `JLinkGDBServer -select USB -device Cortex-M0 -endian little -if JTAG-speed auto -LocalhostOnly`:

```txt
va108xx-rs on  update-va108xx [$»+⇡] via 🦀 v1.78.0 took 32s
❯ ./jlink-gdb.sh
SEGGER J-Link GDB Server V7.96k Command Line Version

JLinkARM.dll V7.96k (DLL compiled Jun  5 2024 13:26:06)

WARNING: Unknown command line parameter auto found.
Command line: -select USB -device Cortex-M0 -endian little -if JTAG-speed auto -LocalhostOnly
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 Cortex-M0
Target device parameters:      none
Target interface:              JTAG
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link OB-STM32F103 V1 compiled Oct 30 2023 12:12:03
Hardware: V1.00
S/N: 660006392
Checking target voltage...
Target voltage: 3.30 V
Listening on TCP/IP port 2331
Connecting to target...

J-Link found 2 JTAG devices, Total IRLen = 9
JTAG ID: 0x0BA01477 (Cortex-M0)
Halting core...
Connected to target
Waiting for GDB connection...^CRestoring target state and closing J-Link connection...
Shutting down...
```

Output of `probe-rs info`

```
va108xx-rs on  update-va108xx [$»+⇡] via 🦀 v1.78.0
❯ probe-rs info
Probing target via JTAG

 INFO probe_rs::probe::jlink: J-Link: Firmware version: J-Link OB-STM32F103 V1 compiled Oct 30 2023 12:12:03
 INFO probe_rs::probe::jlink: J-Link: Hardware version: J-Link 1.0.0
 INFO probe_rs::probe::jlink: J-Link: Target voltage: 3.30 V
 INFO probe_rs::probe::common: Found IDCODE: 0x040037E3 (Silicon Space Technology)
 INFO probe_rs::probe::common: Found IDCODE: 0x0BA01477 (ARM Ltd)
 INFO probe_rs::probe::common: JTAG DR scan complete, found 2 TAPs. [Some(IdCode { .0: 67123171, version: 0, part_number: 16387, manufacturer: 1009, manufacturer_continuation: 7, manufacturer_identity: 113, lsbit: true }), Some(IdCode { .0: 195040375, version: 0, part_number: 47617, manufacturer: 571, manufacturer_continuation: 4, manufacturer_identity: 59, lsbit: true })]
 INFO probe_rs::probe::common: IR lengths are unambiguous: [5, 4]
[... 171 lines cut ...]
DEBUG debug_port_start:write_dp_register: nusb::platform::linux_usbfs::device: Submitted URB 0x5af19111fc90 on ep 81     dp=Default register=Select { .0: 0, ap_sel: 0, ap_bank_sel: 0, dp_bank_sel: 0 }
DEBUG nusb::platform::linux_usbfs::device: Handling events for device 0
DEBUG nusb::platform::linux_usbfs::device: URB 0x5af19111fc90 for ep 81 completed, status=0 actual_length=10
ERROR debug_port_start:write_dp_register: probe_rs::probe::arm_debug_interface: Unexpected DAP response: 0 dp=Default register=Select { .0: 0, ap_sel: 0, ap_bank_sel: 0, dp_bank_sel: 0 }
DEBUG debug_port_start:write_dp_register: probe_rs::probe::common: Write DR: [7, 0, 0, 0, 0, 0, 0, 0], len=35 dp=Default register=Select { .0: 0, ap_sel: 0, ap_bank_sel: 0, dp_bank_sel: 0 }
DEBUG debug_port_start:write_dp_register: nusb::platform::linux_usbfs::device: Submitted URB 0x5af19111fc90 on ep 2     dp=Default register=Select { .0: 0, ap_sel: 0, ap_bank_sel: 0, dp_bank_sel: 0 }
DEBUG nusb::platform::linux_usbfs::device: Handling events for device 0
DEBUG nusb::platform::linux_usbfs::device: URB 0x5af19111fc90 for ep 2 completed, status=0 actual_length=14
DEBUG debug_port_start:write_dp_register: nusb::platform::linux_usbfs::device: Submitted URB 0x5af19111fc90 on ep 81     dp=Default register=Select { .0: 0, ap_sel: 0, ap_bank_sel: 0, dp_bank_sel: 0 }
DEBUG nusb::platform::linux_usbfs::device: Handling events for device 0
DEBUG nusb::platform::linux_usbfs::device: URB 0x5af19111fc90 for ep 81 completed, status=0 actual_length=6
ERROR debug_port_start:write_dp_register: probe_rs::probe::arm_debug_interface: Unexpected DAP response: 4 dp=Default register=Select { .0: 0, ap_sel: 0, ap_bank_sel: 0, dp_bank_sel: 0 }
DEBUG debug_port_start:write_dp_register: probe_rs::probe::common: Write DR: [3, 0, 0, 0, 0, 0, 0, 0], len=35 dp=Default register=Select { .0: 0, ap_sel: 0, ap_bank_sel: 0, dp_bank_sel: 0 }
DEBUG debug_port_start:write_dp_register: nusb::platform::linux_usbfs::device: Submitted URB 0x5af19111fc90 on ep 2     dp=Default register=Select { .0: 0, ap_sel: 0, ap_bank_sel: 0, dp_bank_sel: 0 }
DEBUG nusb::platform::linux_usbfs::device: Handling events for device 0
DEBUG nusb::platform::linux_usbfs::device: URB 0x5af19111fc90 for ep 2 completed, status=0 actual_length=14
DEBUG debug_port_start:write_dp_register: nusb::platform::linux_usbfs::device: Submitted URB 0x5af19111fc90 on ep 81     dp=Default register=Select { .0: 0, ap_sel: 0, ap_bank_sel: 0, dp_bank_sel: 0 }
DEBUG nusb::platform::linux_usbfs::device: Handling events for device 0
DEBUG nusb::platform::linux_usbfs::device: URB 0x5af19111fc90 for ep 81 completed, status=0 actual_length=6
....
repeats very often...
....
```

This MCU is from the same vendor as the one mentioned in this issue: https://github.com/probe-rs/probe-rs/issues/2563

bugadani, 2024-06-15:
The VA106 is a custom target, right? You might need to write a custom debug sequence to support it fully. It's interesting that we can read incorrect IDCODE register values from it 🤔

robamu, 2024-06-15:
Hmm that would be something I have never done. I have found this link: https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/dbg_debug_sqns.html

I actually have a *.pack file for that target. Would that, or a working Keil / Segger JLINK connection help with reverse engineering the sequence?

robamu, 2024-06-15:
The pack archive does not appear to contain anything useful unfortunately.. At least no debug sequence files or something similar.

robamu, 2024-06-17:
I read a bit about [ADIv5](https://developer.arm.com/documentation/ihi0031/a/Introduction) and tried to find our where exactly the code fails, set up a small testing app like this and started stepping through the `probe-rs` code to trace the issue a bit.

```rs
use std::fs::File;
use std::path::Path;
use std::time::Duration;

use probe_rs::config::add_target_from_yaml;
//use probe_rs::flashing::DownloadOptions;
use probe_rs::probe::list::Lister;
use probe_rs::probe::DebugProbeInfo;
use probe_rs::{flashing, MemoryInterface, Permissions, Session};

fn main() {
    tracing_subscriber::fmt::init();

    let lister = Lister::new();

    // Get a list of all available debug probes.
    let probes = lister.list_all();

    // Use the first probe found.
    let mut probe = probes[0].open().expect("failed to open probe");
    println!("probe: {:?}", probe);

    probe
        .select_protocol(probe_rs::probe::WireProtocol::Jtag)
        .expect("failed to select protocol");

    let mut session = probe
        .attach("Cortex-M0", Permissions::default())
        .expect("failed to attach to chip");
}
```

The error happens when [`initialize`](https://github.com/probe-rs/probe-rs/blob/88d09eb2045f2b2d89feefc9c31cbd4896397efc/probe-rs/src/session.rs#L217) ing the interface
More specifically, it fails at the [`debug_port_start` function](https://github.com/probe-rs/probe-rs/blob/88d09eb2045f2b2d89feefc9c31cbd4896397efc/probe-rs/src/architecture/arm/sequences.rs#L572) while trying to attach.

It fails on the first call where JTAG replies are read back in that function when issuing a [SELECT command](https://github.com/probe-rs/probe-rs/blob/88d09eb2045f2b2d89feefc9c31cbd4896397efc/probe-rs/src/architecture/arm/sequences.rs#L588). It does not really matter whether I remove the abort sequence or not. If I understood correctly, there is no reply parsing necessary for the abort sequence so that runs without issues.

What confuses me here is that the `IDCODE`  register can be read as part of [`scan_chain`](https://github.com/probe-rs/probe-rs/blob/88d09eb2045f2b2d89feefc9c31cbd4896397efc/probe-rs/src/probe/common.rs#L633). Doesn't that mean that the communication with the JTAG works in principle? Why is the access with the DPACC so problematic then?

Any pointers and help would be very much appreciated. It would be very nice to get `probe-rs` to work because using `defmt`  with the default RTT viewer does not really work..

bugadani, 2024-06-17:
> J-Link found 2 JTAG devices, Total IRLen = 9

Do you know why there are two devices? Maybe the one you're looking for isn't even the first one. It's **very** hard to guess what's wrong without even knowing what manufacturer this question is about, let alone which MCU.

robamu, 2024-06-17:
I have no idea why there are two devices and I was  very confused as well. Tried to check whether I have any other JTAG devices connected, but I have not.  `JLinkGDBSever` detects two devices as well. Maybe it is related to the on-board JTAG of the target board?

Target YAML:

```yml
name: VA108xx Series
generated_from_pack: true
pack_file_release: 1.4.0
variants:
- name: VA108xx
  cores:
  - name: main
    type: armv6m
    core_access_options: !Arm
      ap: 0
      psel: 0x0
  memory_map:
  - !Ram
    name: IRAM1
    range:
      start: 0x10000000
      end: 0x10008000
    cores:
    - main
  - !Nvm
    name: IROM1
    range:
      start: 0x0
      end: 0x20000
    is_boot_memory: true
    cores:
    - main
  flash_algorithms:
  - va108xx_fm25v20a_fram_128kb_prog
  - va108xx_m95m01_128kb_prog
  - va108xx_mr25h10_1mb_prog
  - va108xx_ttflash_prog
flash_algorithms:
   ... Some algorithms here ...
```

robamu, 2024-06-17:
Thanks a lot! I added `probe.select_jtag_tap(1);` right after `probe.attach_to_unspecified()?;` inside the `attach_arm` method as a quick hacky solution. It seems to get further now:

```txt
probe: Probe { inner: JLink, attached: false }
2024-06-17T09:18:47.044483Z  INFO probe_rs::config::target: Using sequence Arm(DefaultArmSequence(()))
2024-06-17T09:18:47.045129Z  INFO probe_rs::probe::jlink: J-Link: Firmware version: J-Link OB-STM32F103 V1 compiled Oct 30 2023 12:12:03
2024-06-17T09:18:47.045244Z  INFO probe_rs::probe::jlink: J-Link: Hardware version: J-Link 1.0.0
2024-06-17T09:18:47.045362Z  INFO probe_rs::probe::jlink: J-Link: Target voltage: 3.30 V
2024-06-17T09:18:47.114236Z  INFO probe_rs::probe::common: Found IDCODE: 0x040037E3 (Silicon Space Technology)
2024-06-17T09:18:47.114369Z  INFO probe_rs::probe::common: Found IDCODE: 0x0BA01477 (ARM Ltd)
2024-06-17T09:18:47.114398Z  INFO probe_rs::probe::common: JTAG DR scan complete, found 2 TAPs. [Some(IdCode { .0: 67123171, version: 0, part_number: 16387, manufacturer: 1009, manufacturer_continuation: 7, manufacturer_identity: 113, lsbit: true }), Some(IdCode { .0: 195040375, version: 0, part_number: 47617, manufacturer: 571, manufacturer_continuation: 4, manufacturer_identity: 59, lsbit: true })]
2024-06-17T09:18:47.131821Z  INFO probe_rs::probe::common: IR lengths are unambiguous: [5, 4]
2024-06-17T09:18:47.131883Z  INFO probe_rs::probe::common: Found 2 TAPs on reset scan
2024-06-17T09:18:47.131926Z  INFO probe_rs::probe: JtagChainItem { idcode: Some(IdCode { .0: 67123171, version: 0, part_number: 16387, manufacturer: 1009, manufacturer_continuation: 7, manufacturer_identity: 113, lsbit: true }), irlen: 5 }
2024-06-17T09:18:47.131947Z  INFO probe_rs::probe: JtagChainItem { idcode: Some(IdCode { .0: 195040375, version: 0, part_number: 47617, manufacturer: 571, manufacturer_continuation: 4, manufacturer_identity: 59, lsbit: true }), irlen: 4 }
2024-06-17T09:18:47.131975Z  INFO probe_rs::probe: JtagChainItem { idcode: Some(IdCode { .0: 67123171, version: 0, part_number: 16387, manufacturer: 1009, manufacturer_continuation: 7, manufacturer_identity: 113, lsbit: true }), irlen: 5 }
2024-06-17T09:18:47.131992Z  INFO probe_rs::probe: JtagChainItem { idcode: Some(IdCode { .0: 195040375, version: 0, part_number: 47617, manufacturer: 571, manufacturer_continuation: 4, manufacturer_identity: 59, lsbit: true }), irlen: 4 }
2024-06-17T09:18:48.935807Z  INFO probe_rs::session: Core status: Halted(Unknown)
2024-06-17T09:18:49.851564Z  INFO session_drop: probe_rs::session: Core status: Halted(Unknown)
2024-06-17T09:18:50.668332Z  INFO probe_rs::architecture::arm::sequences: Powering down debug port Default
2024-06-17T09:18:50.800594Z  INFO probe_rs::architecture::arm::sequences: Powering down debug port Default
```

robamu, 2024-06-17:
I have found some more valuable information in the datasheet
[VA10800_VA10820_PG_Small.pdf](https://github.com/user-attachments/files/15865442/VA10800_VA10820_PG_Small.pdf)

robamu, 2024-06-17:
I did some more testing related to the weird error above.

I tried writing and reading single `u32` words to and from RAM, and this worked without issues, so I was really confused.
I stepped through the flasher code to [this line](https://github.com/probe-rs/probe-rs/blob/88d09eb2045f2b2d89feefc9c31cbd4896397efc/probe-rs/src/flashing/flasher.rs#L168) where the flashing algorithm is flashed.

When I read back a single u32 word, I can see the correct value in the buffer. Anything larger than one is problematic.
The core read sequence is [here](https://github.com/probe-rs/probe-rs/blob/88d09eb2045f2b2d89feefc9c31cbd4896397efc/probe-rs/src/architecture/arm/memory/adi_v5_memory_interface.rs#L313).. so for some reason, repeated AP reads are problematic?

robamu, 2024-06-18:
Some more poking and twiddling to the best of my ability and I have figured out a way to make the regular multi-word read work. I think the whole flashing process now works as well! At least the custom blinky I tried to flash all the time seems to run after a reboot now :)

In the [`raw_read_block`](https://github.com/probe-rs/probe-rs/blob/7a5dc319f23a866b4d566c05376e38616e358dc8/probe-rs/src/probe/arm_debug_interface.rs#L983) low level function, I saw the transfers vector which is constructed has an [`idle_cycles_after`](https://github.com/probe-rs/probe-rs/blob/7a5dc319f23a866b4d566c05376e38616e358dc8/probe-rs/src/probe/arm_debug_interface.rs#L592) parameter. I set that to two, and now everything is working. I have no idea why this is needed. I started to read a bit of the [`perform_transfers`](https://github.com/probe-rs/probe-rs/blob/7a5dc319f23a866b4d566c05376e38616e358dc8/probe-rs/src/probe/arm_debug_interface.rs#L352) function which comes right after that..

Do yo have any ideas why this fixes the issue?

bugadani, 2024-06-18:
What happens if you leave the initial value at 0, but remove the write check from L507?

robamu, 2024-06-18:
Does not work. It runs through that path though, and I can see that the idle cycles are incremented by two.

robamu, 2024-06-21:
Any other ideas? I think not much is missing to make this work for the VA108xx family :)

robamu, 2024-06-22:
Just one idea (please tell me if this is wrong): The register is in auto-increment mode. Could it be that those registers are auto-incremented even if a WAIT response is returned? Which would then mean that we read past the written bytes for the second (successfull) attempt? So if the error handling was moved to an upper layer where the TAR/CSW registers are set properly again, that might solve the issue?

Tiwalun, 2024-06-22:
I'm not entirely sure about JTAG, but if the behaviour is the same as when SWD is used, then the WAIT should not affect the TAR register.

The WAIT means that the Debug Port cannot accept the new transfer because the previous one wasn't complete. So it shouldn't affect the memory AP, it never gets that far.

robamu, 2024-06-22:
Doesn't the transfer complete eventually?

I've read the WAIT segment at https://developer.arm.com/documentation/ddi0413/c/debug-access-port/jtag-dp/dr-scan-chain-and-dr-registers , so the implementation definitely looks correct..

bugadani, 2024-07-09:
I think I have made a bad assumption [here](https://github.com/probe-rs/probe-rs/blob/d418d16b4ccbc75a54054d0d17437f4687f1fd59/probe-rs/src/probe/common.rs#L615). I believe this condition, which was meant to be an optimisation, is not compatible with the ARM IR commands. This `if` prevents repeated reads from completing, so that's why the same value is read over and over and over again.

robamu, 2024-07-11:
What would be the name of the YAML config key to specify the TAP to use? (even if it is not supported yet). Then I could already add it to the target file.