probe-rs/probe-rs #2573: Can't erase/flash nRF52840

30350n, 2024-06-17:
**Describe the bug**
A simple `probe-rs erase --chip nrf52840_xxAA` always fails with `ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception`.

**Desktop:**
- Windows 10
- Linux (NixOS)

**Additional context**
Could be related to #2537 (from what I can tell, `data_load_address` doesn't get defined in `probe-rs/targets/nRF52_Series.yaml`.

bugadani, 2024-06-17:
> Could be related to https://github.com/probe-rs/probe-rs/issues/2537 (from what I can tell, data_load_address doesn't get defined in probe-rs/targets/nRF52_Series.yaml

Do you know of any reason why the end of the RAM wouldn't be usable on an nRF52840?

You can try and revert https://github.com/probe-rs/probe-rs/pull/2375 to see if it helps. If it does, we can look into alternatives.

30350n, 2024-06-17:
> Do you know of any reason why the end of the RAM wouldn't be usable on an nRF52840?

I have absolutely no clue tbh, the issue just looked somewhat similar. I honestly don't really get what's happening in #2537, the error message doesn't really relate to the actual issue at all, right?

Regarding versions: I just tested through the last few versions in Windows real quick and everything until 0.23.0 (0.21.1 and 0.22.0) seems to work. 0.23.0 and 0.24.0 are broken.

30350n, 2024-07-09:
Okay, I think this was just on me, it's started working with the newest version too now.

Sorry for the trouble!

closed 2024-07-09 by 30350n

bmwebster, 2024-08-21:
Hi, I'm getting this very repeatably on 0.24.0, to the point where I couldn't get anything running via flashing. I'm just trying to use the nrf-softdevice examples on a microbit-v2, so using nRF52833_xxAA. If I downgrade to 0.22.0 everything works fine.