probe-rs/probe-rs #2574: Generic riscv chip via JTAG uses ARM driver

andreasWallnerIFX, 2024-06-17:
**Describe the bug**
I'm trying to debug an in-house RISC-V core via JTAG, using the generic "riscv" chip setting. `probe-rs info...` does not work for me, looking at the logs it seems it's using the ARM driver. (also looking at the logic analyzer trace the commands written to IR)

**To Reproduce**

I'm using an FT4232 as my JTAG adapter. The chip in question does not have a TRST.
Running `RUST_LOG=debug probe-rs info --chip riscv --protocol jtag` (specifiying the protocol because of #522) I get:

<details>
<summary>Log output</summary>
```
... some probing
DEBUG probe_rs::probe::ftdi::ftdaye: scanning 4 interfaces
DEBUG probe_rs::probe::ftdi::ftdaye: Interface is #0
DEBUG probe_rs::probe::ftdi::ftdaye: interface 1 does not match requested interface A
DEBUG probe_rs::probe::ftdi::ftdaye: interface 2 does not match requested interface A
DEBUG probe_rs::probe::ftdi::ftdaye: interface 3 does not match requested interface A
DEBUG nusb::platform::windows_winusb::device: Claiming device DevInst(1) interface 0 with interface path `\\?\USB#VID_0403&PID_6011#9&133b0cf5&0&3#{a5dcbf10-6530-11d2-901f-00c04fb951ed}`
DEBUG probe_rs::probe::ftdi::ftdaye: Opened FTDI device: Some(FT4232H)
DEBUG probe_rs::probe::ftdi: opened probe: FtdiProbe { adapter: JtagAdapter { device: Device { chip_type: Some(FT4232H) }, speed_khz: 1000, command: None { tms: false }, commands: [], in_bit_counts: [], in_bits: BitVec<u8, bitvec::order::Lsb0> { addr: 0x1, head: 000, bits: 0, capacity: 0 } [], ftdi: FtdiProperties { buffer_size: 4096, max_clock: 30000, has_divide_by_5: true } }, jtag_state: JtagDriverState { state: Reset, current_ir_reg: 1, max_ir_address: 15, expected_scan_chain: None, scan_chain: [], chain_params: ChainParams { irpre: 0, irpost: 0, drpre: 0, drpost: 0, irlen: 0 }, jtag_idle_cycles: 0 }, probe_statistics: ProbeStatistics { num_transfers: 0, num_extra_transfers: 0, num_io_calls: 0, num_wait_resp: 0, num_faults: 0 }, swd_settings: SwdSettings { num_idle_cycles_between_writes: 2, num_retries_after_wait: 1000, max_retry_idle_cycles_after_wait: 128, idle_cycles_before_write_verify: 8, idle_cycles_after_transfer: 8 } }
Probing target via JTAG

DEBUG probe_rs::probe::ftdi: Attaching...
 INFO nusb::platform::windows_winusb::device: Blocking control Out, 0 bytes
DEBUG probe_rs::probe::ftdi::ftdaye: Response to 00/0000: 0
 INFO nusb::platform::windows_winusb::device: Blocking control Out, 0 bytes
DEBUG probe_rs::probe::ftdi::ftdaye: Response to 0B/020B: 0
 INFO nusb::platform::windows_winusb::device: Blocking control Out, 0 bytes
DEBUG probe_rs::probe::ftdi::ftdaye: Response to 09/0001: 0
 INFO nusb::platform::windows_winusb::device: Blocking control Out, 0 bytes
DEBUG probe_rs::probe::ftdi::ftdaye: Response to 00/0002: 0
 INFO nusb::platform::windows_winusb::device: Blocking control Out, 0 bytes
DEBUG probe_rs::probe::ftdi::ftdaye: Response to 00/0001: 0
DEBUG nusb::platform::windows_winusb::transfer: Submit transfer 0x1cf106949d0 on endpoint 81 for 512 bytes IN
DEBUG nusb::platform::windows_winusb::transfer: Handling completion for transfer 0x1cf106949d0
DEBUG nusb::platform::windows_winusb::transfer: Transfer 0x1cf106949d0 on endpoint 81 complete: 2 bytes transferred
DEBUG probe_rs::probe::ftdi::ftdaye: Read [32, 60] bytes from USB
DEBUG probe_rs::probe::ftdi::ftdaye: read 0 bytes
DEBUG nusb::platform::windows_winusb::transfer: Submit transfer 0x1cf10694100 on endpoint 02 for 3 bytes OUT
DEBUG nusb::platform::windows_winusb::transfer: Handling completion for transfer 0x1cf10694100
DEBUG nusb::platform::windows_winusb::transfer: Transfer 0x1cf10694100 on endpoint 02 complete: 3 bytes transferred
DEBUG probe_rs::probe::ftdi::ftdaye: wrote 3 bytes
DEBUG nusb::platform::windows_winusb::transfer: Submit transfer 0x1cf10694160 on endpoint 02 for 3 bytes OUT
DEBUG nusb::platform::windows_winusb::transfer: Handling completion for transfer 0x1cf10694160
DEBUG nusb::platform::windows_winusb::transfer: Transfer 0x1cf10694160 on endpoint 02 complete: 3 bytes transferred
DEBUG probe_rs::probe::ftdi::ftdaye: wrote 3 bytes
DEBUG nusb::platform::windows_winusb::transfer: Submit transfer 0x1cf10693fe0 on endpoint 02 for 1 bytes OUT
DEBUG nusb::platform::windows_winusb::transfer: Handling completion for transfer 0x1cf10693fe0
DEBUG nusb::platform::windows_winusb::transfer: Transfer 0x1cf10693fe0 on endpoint 02 complete: 1 bytes transferred
DEBUG probe_rs::probe::ftdi::ftdaye: wrote 1 bytes
 INFO probe_rs::probe::ftdi: Setting speed to 1000 kHz (divisor: 29, actual speed: 1000 kHz)
DEBUG nusb::platform::windows_winusb::transfer: Submit transfer 0x1cf10694100 on endpoint 02 for 3 bytes OUT
DEBUG nusb::platform::windows_winusb::transfer: Handling completion for transfer 0x1cf10694100
DEBUG nusb::platform::windows_winusb::transfer: Transfer 0x1cf10694100 on endpoint 02 complete: 3 bytes transferred
DEBUG probe_rs::probe::ftdi::ftdaye: wrote 3 bytes
DEBUG nusb::platform::windows_winusb::transfer: Submit transfer 0x1cf10694670 on endpoint 02 for 1 bytes OUT
DEBUG nusb::platform::windows_winusb::transfer: Handling completion for transfer 0x1cf10694670
DEBUG nusb::platform::windows_winusb::transfer: Transfer 0x1cf10694670 on endpoint 02 complete: 1 bytes transferred
DEBUG probe_rs::probe::ftdi::ftdaye: wrote 1 bytes
DEBUG probe_rs::probe::common: Resetting JTAG chain by setting tms high for 5 bits
DEBUG probe_rs::probe::ftdi: Appending TmsBits { bit_count: 6, tms_bits: 31, tdi: true, capture: false }
DEBUG probe_rs::probe::ftdi: Appending None { tms: false }
DEBUG nusb::platform::windows_winusb::transfer: Submit transfer 0x1cf10693ef0 on endpoint 02 for 4 bytes OUT
DEBUG nusb::platform::windows_winusb::transfer: Handling completion for transfer 0x1cf10693ef0
DEBUG nusb::platform::windows_winusb::transfer: Transfer 0x1cf10693ef0 on endpoint 02 complete: 4 bytes transferred
DEBUG probe_rs::probe::ftdi::ftdaye: wrote 4 bytes
DEBUG probe_rs::probe::common: Response to reset: []
DEBUG probe_rs::probe::common: Write DR: [255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255], len=256
DEBUG probe_rs::probe::ftdi: Appending TmsBits { bit_count: 3, tms_bits: 1, tdi: false, capture: false }
DEBUG probe_rs::probe::ftdi: Appending TdiSequence { tdi_bytes: [255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255], bit_count: 7, tdi_bits: 127, capture: true }
[... 108 lines cut ...]
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: JTAG: No special sequence needed to connect to debug port
DEBUG debug_port_start:write_dp_register: probe_rs::architecture::arm::dp: Writing DP register ABORT, value=0x0000001f dp=Default register=Abort { .0: 31 }
DEBUG debug_port_start:write_dp_register: probe_rs::probe::arm_debug_interface: Performing 1 transfers (0 additional transfers) dp=Default register=Abort { .0: 31 }
DEBUG debug_port_start:write_dp_register: probe_rs::probe::common: Write IR: [8, 0, 0, 0], len=8 dp=Default register=Abort { .0: 31 }
DEBUG debug_port_start:write_dp_register: probe_rs::probe::ftdi: Appending TmsBits { bit_count: 1, tms_bits: 0, tdi: true, capture: false } dp=Default register=Abort { .0: 31 }
DEBUG debug_port_start:write_dp_register: probe_rs::probe::ftdi: Appending TmsBits { bit_count: 6, tms_bits: 3, tdi: false, capture: false } dp=Default register=Abort { .0: 31 }
DEBUG debug_port_start:write_dp_register: probe_rs::probe::ftdi: Appending TdiBits { bit_count: 5, tdi_bits: 2, capture: false } dp=Default register=Abort { .0: 31 }
DEBUG debug_port_start:write_dp_register: probe_rs::probe::common: Write DR: [8, 0, 0, 0, 0, 0, 0, 0], len=35 dp=Default register=Abort { .0: 31 }
```
</details>

Of note for me are:
- the ID is read correctly
- the TAP is detected correctly (IR length 8, 1 tap)
- at the end of the log there are messages "Trying to show ARM chip information" and from `probe_rs::architecture::arm::sequences` which seems to want to do a SWD reset.

**Expected behavior**
`probe-rs` should send RISC-V debug instructions, not ARM ones.

**Desktop (please complete the following information):**

- Windows 10

**Additional context**
The core itself implements the RISC-V debug spec v0.13.2 - but I guess this should make a difference since `probe-rs` never reads the version?

Tiwalun, 2024-06-17:
`probe-rs info`is a bit of a special case, because it tries to autodetect the connected chip. I'm not sure if it even considers the `chip` argument.

What happens if you try something like `probe-rs debug --chip riscv`?

bugadani, 2024-06-17:
I'm inclined to remove or rename `probe-rs info`. It's just not useful for anything non-ARM currently (even if some other chips may have coresight compatible components). `info` is very brute-force, and there's a very good chance it does something evil to the connected device, because it's essentially sending random bits to something unknown and hoping it'll understand.

> the ID is read correctly
> the TAP is detected correctly (IR length 8, 1 tap)

💓

> which seems to want to do a SWD reset.

Ugh... Anything related to protocols is a mess currently 😭

> The core itself implements the RISC-V debug spec v0.13.2 - but I guess this should make a difference since probe-rs never reads the version?

It should, if it gets to that point: https://github.com/probe-rs/probe-rs/blob/88d09eb2045f2b2d89feefc9c31cbd4896397efc/probe-rs/src/architecture/riscv/communication_interface.rs#L451-L457

Tiwalun, 2024-06-17:
> I'm inclined to remove or rename `probe-rs info`. It's just not useful for anything non-ARM currently (even if some other chips may have coresight compatible components). `info` is very brute-force, [...]

The whole reason `info` exists is to help with developing probe-rs, it was never intended to be a public tool. But people liked the CLI examples in probe-rs so much that they became an actual tool... 😄 

Maybe we should move it back to an example, but I would very much like to keep it, it is very useful when looking at ARM chips.

andreasWallnerIFX, 2024-06-17:
> `probe-rs info`is a bit of a special case, because it tries to autodetect the connected chip. I'm not sure if it even considers the `chip` argument.

That explains my initial issue - I thought I'd start out with a simple command 😢

> > The core itself implements the RISC-V debug spec v0.13.2 - but I guess this should make a difference since probe-rs never reads the version?
> 
> It should, if it gets to that point:

Above I meant that it shouldn't make a difference since `probe-rs` never got that far, that's all good now

> What happens if you try something like probe-rs debug --chip riscv?

That looks much better, but still does not work - I'll have to go through the log to see what's happening and why it's failing, but will do that tomorrow. Been a long day already...

andreasWallnerIFX, 2024-06-18:
Maybe to expand a bit on the issue here since I'm not 100% sure on how to read the debug spec.
What happens is:
- scan chain detection & IDCODE work as expected
- `dmcontrol` is written to 0x0001
- `dmstatus` is read - this read returns a "BUSY" sticky response (this read writes 0x44000000001 to the `DR`)
- `dmstatus` is immediately read again - it now successfully reads 0x00000001

The value read from `dmstatus` is inconsistent (incorrect version, also some other bits are incorrect)

OpenOCD will in this case add additional IDLE cycles until there are none reported anymore.
I guess they read the RISC-V debug spec v1.0, 6.1.5 - where the `op` description in the table says about the busy state: "If a debugger sees this status, it needs to give the target more TCK edges between Update-DR and Capture-DR" and decided to in these cases ignore the reponse from the DUT.

On the other hand there is the note in 6.1.5 stating that "In Update-DR, the DTM start the operation specified in op unless the current status reported in op is sticky - which to me does not explicitly invalidate the command sent before (and I couldn't find an indication for that anywhere else in the spec...