probe-rs/probe-rs #2579: Error on xtensa-esp32s3-espidf

lorenzo-pirchio, 2024-06-21:
Hi, 
I am currently working on a project for esp32 in Rust.
I followed the instructions of https://docs.esp-rs.org/book/tooling/debugging/probe-rs.html.
So I did cargo install probe-rs-tools on cmd and I downloaded probe-rs extention for vscode and I added probe-rs.exe as debugger in the extention settings.
I downloaded the esp-idf-template and I added the launch.json file.
But when I launch the debug I get this:
[image]
Can anyone help me?
Could it be a problem that I didn't set up the cmake extention?
Thanks

bugadani, 2024-06-21:
This particular error looks like what happens if you set `server:` yourself in your launch.json, but then don't actually start `probe-rs` yourself. The port number seems to support this, can you please try either following [Connecting to a standalone probe-rs dap-server server](https://probe.rs/docs/tools/debugger/#connecting-to-a-standalone-probe-rs-dap-server-server) or removing `server` from your launch config?

Other than that, even if you manage to resolve this particular error, our esp-idf support isn't really done (we don't correctly reset Xtensa ESP32s currently) so you might run into other issues.

lorenzo-pirchio, 2024-06-24:
This is my launch.json.

{
    "version": "0.24.0",
    "configurations": [
        {
            "type": "probe-rs-debug",
            "request": "launch",
            "name": "Launch",
            "cwd": "C:\\prova-debug4",
            "chip": "esp32s3", //!MODIFY
            "flashingConfig": {
                "flashingEnabled": true,
            },
            "coreConfigs": [
                {
                    "coreIndex": 0,
                    "programBinary": "C:\\S3WifiCD\\xtensa-esp32s3-espidf\\debug", //!MODIFY
                }
            ]
        },
        {
            "type": "probe-rs-debug",
            "request": "attach",
            "name": "Attach",
            "cwd": "C:\\prova-debug4",
            "chip": "esp32s3", //!MODIFY
            "coreConfigs": [
                {
                    "coreIndex": 0,
                    "programBinary": "C:\\S3WifiCD\\xtensa-esp32s3-espidf\\debug", //!MODIFY
                }
            ]
        }
    ]
}

I don't set any server and I removed the lines for the reset but I still have the same problems.
Everytime I get the error, the port number  is different.

bugadani, 2024-06-24:
Okay thank you. I'm not sure what's going on, maybe @noppej knows how we can get an ECONNREFUSED for a partially supported target 🤔

Also, while this is irrelevant to your issue, `"version": "0.24.0"` is not correct: the version is related to the `launch.json` format which should be `0.2.0`.

lorenzo-pirchio, 2024-06-25:
To be sure that I set up the debug environment correctly, I explain all the steps.
First of all, I'm using a Windows pc and Esp32S3.

1.  I did **_cargo install probe-rs-tools_** 
2. I downloaded the extension for VSCode
3.  I put the path to probe-rs.exe as debugger.
4. I generated launch.json in my project which is the following:
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "probe-rs-debug",
            "request": "launch",
            "name": "Launch",
            "cwd": "C:\\prova-debug4",
            "chip": "esp32s3", //!MODIFY
            "flashingConfig": {
                "flashingEnabled": true,
            },
            "coreConfigs": [
                {
                    "coreIndex": 0,
                    "programBinary": "C:\\S3WifiCD\\xtensa-esp32s3-espidf\\debug", //!MODIFY
                }
            ]
        },
        {
            "type": "probe-rs-debug",
            "request": "attach",
            "name": "Attach",
            "cwd": "C:\\prova-debug4",
            "chip": "esp32s3", //!MODIFY
            "coreConfigs": [
                {
                    "coreIndex": 0,
                    "programBinary": "C:\\S3WifiCD\\xtensa-esp32s3-espidf\\debug", //!MODIFY
                }
            ]
        }
    ]
}
5. I changed config.toml adding:
[target.'cfg(any(target_arch = "riscv32", target_arch = "xtensa"))']
runner = "probe-rs run --chip esp32s3"
linker = "ldproxy"

I have a dubt, Do I need to have cmake installed in my pc to use probe-rs?

Tiwalun, 2024-07-25:
The programBinary path looks wrong if it's a Rust project, it should point to the actual binary file which is usually in something like `target/<architecture>/debug/program`(not sure what the architecture for this chip is).

And can you see anything in the `Debug Console` output in vscode? Something like this:

[image]