probe-rs/probe-rs #2590: target-gen fails when trying to create a custom flash algorithm

idgaron, 2024-06-27:
**Describe the bug**
Hello. I was recently trying to target a currently unsupported Microchip microcontroller (SAMRH707) using [these instructions](https://github.com/probe-rs/flash-algorithm-template), but the program crashes after running `cargo run`. The template is generated completely fine after filling out all the memory sections. I then used `cargo run` which successfully compiled all the tools with 1 warning about an unused import `core::mem::MaybeUninit`. It then tries to run `target-gen test template.yaml target/definition.yaml target\thumbv7em-none-eabihf\debug\samrh707` at which point it crashes and panics with a message "index out of bounds: the len is 0 but the index is 0". What could be causing this issue? Any help is appreciated as I am new to using probe-rs.

**To Reproduce**
Steps to reproduce the behavior:

1. install the dependencies for the Flash Algorithm Template
2. generate the template using the correct memory/flash values for the microcontroller
3. run `cargo run` within the template

**Expected behavior**
The flash algorithm is supposed to be extracted and the target definition.yaml file is supposed to be generated and tested successfully.

**Stacktrace**
PS> cargo generate gh:probe-rs/flash-algorithm-template
 Favorite `gh:probe-rs/flash-algorithm-template` not found in config, using it as a git repository: https://github.com/probe-rs/flash-algorithm-template.git
 Project Name: samrh707
 Destination: C:\Users\igaron\samrh707 ...
 project-name: samrh707 ...
 Generating template ...
✔  What is the target architecture? · thumbv7em-none-eabihf
What is the target RAM start address where the flash algorithm should be placed?: 0x20000000
 What is the target RAM size in bytes?: 0x20000
What is the target Flash start address of the device the flash algorithm is for?: 0x10000000
 What is the target Flash size in bytes?: 0x20000
What is the target Flash page size in bytes?: 0x100
What is the target Flash sector size in bytes?: 0x8000
 What is the erased state of a byte in Flash?: 0xFF
 Moving generated files into: `C:\Users\igaron\samrh707`...
 Initializing a fresh Git repository
 Done! New project created C:\Users\igaron\samrh707
PS C:\Users\igaron> cd .\samrh707\
PS C:\Users\igaron\samrh707> cargo run
    Updating crates.io index
     Locking 16 packages to latest compatible versions
      Adding bare-metal v0.2.5 (latest: v1.0.0)
      Adding bitfield v0.13.2 (latest: v0.15.0)
      Adding embedded-hal v0.2.7 (latest: v1.0.0)
      Adding nb v0.1.3 (latest: v1.1.0)
      Adding rtt-target v0.3.1 (latest: v0.5.0)
      Adding rustc_version v0.2.3 (latest: v0.4.0)
      Adding semver v0.9.0 (latest: v1.0.23)
      Adding semver-parser v0.7.0 (latest: v0.10.2)
   Compiling semver-parser v0.7.0
   Compiling nb v1.1.0
   Compiling void v1.0.2
   Compiling vcell v0.1.3
   Compiling cortex-m v0.7.7
   Compiling bitfield v0.13.2
   Compiling flash-algorithm v0.4.0
   Compiling ufmt-write v0.1.0
   Compiling nb v0.1.3
   Compiling volatile-register v0.2.2
   Compiling semver v0.9.0
   Compiling embedded-hal v0.2.7
   Compiling rustc_version v0.2.3
   Compiling bare-metal v0.2.5
   Compiling rtt-target v0.3.1
   Compiling samrh707 v0.1.0 (C:\Users\igaron\samrh707)
warning: unused import: `core::mem::MaybeUninit`
 --> src/main.rs:4:5
  |
4 | use core::mem::MaybeUninit;
  |     ^^^^^^^^^^^^^^^^^^^^^^
  |
  = note: `#[warn(unused_imports)]` on by default

warning: `samrh707` (bin "samrh707") generated 1 warning (run `cargo fix --bin "samrh707"` to apply 1 suggestion)
    Finished `dev` profile [optimized + debuginfo] target(s) in 5.83s
     Running `target-gen test template.yaml target/definition.yaml target\thumbv7em-none-eabihf\debug\samrh707`
Generating the YAML file in `"target/definition.yaml"`
thread 'main' panicked at C:\Users\igaron\.cargo\registry\src\index.crates.io-6f17d22bba15001f\target-gen-0.24.0\src\commands\test.rs:126:51:
index out of bounds: the len is 0 but the index is 0
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
error: process didn't exit successfully: `target-gen test template.yaml target/definition.yaml target\thumbv7em-none-eabihf\debug\samrh707` (exit code: 101)

stack backtrace:
   0:     0x7ff74e6b0408 - <unknown>
   1:     0x7ff74e6d6309 - <unknown>
   2:     0x7ff74e6ab4e1 - <unknown>
   3:     0x7ff74e6b01e6 - <unknown>
   4:     0x7ff74e6b2d08 - <unknown>
   5:     0x7ff74e6b2977 - <unknown>
   6:     0x7ff74e6b3238 - <unknown>
   7:     0x7ff74e6b30f7 - <unknown>
   8:     0x7ff74e6b0d7f - <unknown>
   9:     0x7ff74e6b2da8 - <unknown>
  10:     0x7ff74e76e384 - hid_write
  11:     0x7ff74e76e52e - hid_write
  12:     0x7ff74e010368 - <unknown>
  13:     0x7ff74e045ded - <unknown>
  14:     0x7ff74e024e1a - <unknown>
  15:     0x7ff74e07237b - <unknown>
  16:     0x7ff74e041d66 - <unknown>
  17:     0x7ff74e05fcfa - <unknown>
  18:     0x7ff74e6a449e - <unknown>
  19:     0x7ff74e07249c - <unknown>
  20:     0x7ff74e74f2fc - hid_write
  21:     0x7fffdcf7257d - <unknown>
  22:     0x7fffddacaa48 - RtlUserThreadStart

**Desktop (please complete the following information):**

- Windows 11

**Additional context**
Add any other context about the problem here.

bugadani, 2024-06-27:
The problem is that target-gen finds the first MCU family in the registry, and tries to use it. The first family is a generic target without any flash algorithms in it. At this point it looks like `target-gen test` somehow diverged from its intended purpose and ended up being unusable.

idgaron, 2024-06-27:
Thank you for the reply! Do you have any advice on how to generate a target without using `target-gen test`?

bugadani, 2024-06-28:
I've been mainly using `cargo build` to get an ELF file and `target-gen elf` to process that into a YAML.

idgaron, 2024-06-28:
I had tried that before. When I build the elf file and run `target-gen elf path/to/elf` it gives me the error:
`Error: Failed to extract flash information from ELF file '.\target\thumb7em-none-eabihf\debug\flashing'.

Caused by:
Failed to find 'FlashDevice' symbol in ELF file.`
I am building using the [Cortex-M Quickstart Template](https://github.com/rust-embedded/cortex-m-quickstart) and have tried both the release and debug version.

bugadani, 2024-06-28:
I will try looking into this later this weekend. My guess is that the template is wrong, we don't have problems extracting from cmsis packs.

idgaron, 2024-06-28:
ah i think that might be the issue then. there is no CMSIS pack for the microcontroller i'm trying to target i think. I was thinking the ELF file method didn't require a CMSIS pack to be used but that might have been my misunderstanding with the `target-gen` utility

bugadani, 2024-06-28:
> I was thinking the ELF file method didn't require a CMSIS pack to be used

Well it shouldn't be, it's rather that the pack extraction builds on processing elf files correctly :)

idgaron, 2024-06-28:
ah i see. please let me know if you find anything out! I will also see if i can't pinpoint any error spots on my end as well

Yatekii, 2024-11-22:
Is this fixed? We had some fixes around this I believe.

pkoevesdi, 2024-11-30:
I think, this is fixed by https://github.com/probe-rs/probe-rs/commit/5ea7759f2fcec66b7166b0c36c86715a4322a00f
To try it out, You can (re)install cargo from master branch with this:
`cargo install target-gen --git https://github.com/probe-rs/probe-rs --locked --force`
 (adapted from [here](https://probe.rs/docs/getting-started/installation/#installing-from-source-(cargo-install)))

Yatekii, 2024-12-10:
I will mark this as resolved. Please reopen at will if it's still a problem.

closed 2024-12-10 by Yatekii