probe-rs/probe-rs #2593: Can't connect to nrf51

denysvitali, 2024-06-29:
**Describe the bug**

It looks like `probe-rs` (`0.24.0`) cannot connect to my `nrf51822` based device because the debug port cannot be connected.

Here is the result of using `openocd`:

```plain
$ openocd \
    -c 'adapter driver jlink; transport select swd; source [find target/nrf51.cfg];' -c 'init'
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : J-Link ARM-OB STM32 compiled Aug 22 2012 19:52:04
Info : Hardware version: 7.00
Info : VTarget = 3.300 V
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x0bb11477
Info : [nrf51.cpu] Cortex-M0 r0p0 processor detected
Info : [nrf51.cpu] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for nrf51.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```

And here is with `probe-rs`:

```
RUST_LOG="probe_rs::architecture::arm=debug" probe-rs info
Probing target via JTAG

Error identifying target using protocol JTAG: An error with the usage of the probe occurred

Probing target via SWD

DEBUG debug_port_setup: probe_rs::architecture::arm::sequences: Setting up debug port Default
DEBUG debug_port_setup: probe_rs::architecture::arm::sequences: Performing SWD line reset
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: SWD: Connecting to debug port with address Default
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: Performing SWD line reset
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: Reading DPIDR to enable SWD interface
DEBUG debug_port_setup: probe_rs::architecture::arm::sequences: Performing SWD line reset
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: SWD: Connecting to debug port with address Default
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: Performing SWD line reset
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: Reading DPIDR to enable SWD interface
DEBUG debug_port_setup: probe_rs::architecture::arm::sequences: Performing SWD line reset
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: SWD: Connecting to debug port with address Default
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: Performing SWD line reset
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: Reading DPIDR to enable SWD interface
DEBUG debug_port_setup: probe_rs::architecture::arm::sequences: Performing SWD line reset
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: SWD: Connecting to debug port with address Default
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: Performing SWD line reset
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: Reading DPIDR to enable SWD interface
DEBUG debug_port_setup: probe_rs::architecture::arm::sequences: Performing SWD line reset
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: SWD: Connecting to debug port with address Default
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: Performing SWD line reset
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: Reading DPIDR to enable SWD interface
Error showing ARM chip information for Debug Port Default: An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Trying alternate multi-drop debug ports
DEBUG debug_port_setup: probe_rs::architecture::arm::sequences: Setting up debug port Multidrop(1002927)
DEBUG debug_port_setup: probe_rs::architecture::arm::sequences: Performing SWD line reset
DEBUG debug_port_setup: probe_rs::architecture::arm::sequences: Select Dormant State (from JTAG)
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: SWD: Connecting to debug port with address Multidrop(1002927)
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: Performing SWD line reset
DEBUG debug_port_setup:debug_port_connect: probe_rs::architecture::arm::sequences: Writing targetsel 0x1002927
[... 78 lines cut ...]
(or any other combination)

**To Reproduce**
Steps to reproduce the behavior:

1. Connect your `SEGGER J-Link PLUS` via USB
1. Connect the NRF51 (`nRF51822`)
1. Check if it works with openocd (command above, it should work) 
1. Run `probe-rs info` (it shouldn't work)

**Expected behavior**

The debug mode is correctly activated and `probe-rs info` returns the relevant information about the chip. The user can also use `probe-rs` to flash a binary.

**Stacktrace**

N/A

**Desktop (please complete the following information):**

Arch Linux - although this is not really relevant.

**Additional context**

IIRC NRF52 works fine as I've used `probe-rs` already with another chip.

bugadani, 2024-06-29:
As we have a nRF51822_xxAA in our automated tester that works no problem, there are a few options:
 - maybe the probe is too old and is somehow incompatible with probe-rs
 - maybe `probe-rs info` does something that the nrf51 doen't like. Could you try a command that is actually useful?
 - maybe the default probe speed is off. You can try `--speed 1000` to mimic openocd here
 - maybe we had something in 0.24 that has since been fixed without us noticing. Could you please try installing probe-rs from source and retry?

denysvitali, 2024-06-29:
Thanks for the quick reply!

> maybe `probe-rs info` does something that the nrf51 doen't like. Could you try a command that is actually useful?

No subcommand really works - it seem to fail when setting up the debug mode

> maybe the default probe speed is off. You can try --speed 1000 to mimic openocd here

Changing the speed has the exact same effect unfortunately

> maybe we had something in 0.24 that has since been fixed without us noticing. Could you please try installing probe-rs from source and retry?

For debugging this further I tried to run this from `main` - but it still has the same effects.
My understanding is that the probe sequence is either wrong, or doesn't put the device in debug mode.

I have tried to follow the code and refer to [Open-CMSIS's specification](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/debug_description.html#debugPortStop) - but everything seems correct.

At this point I think that OpenOCD is doing something that `probe-rs` is not (to initialize the debug port).

bugadani, 2024-06-30:
My suspicion here is that our j-link driver has some issues with your quite ancient probe, most likely [this line](https://github.com/probe-rs/probe-rs/blob/5ea7759f2fcec66b7166b0c36c86715a4322a00f/probe-rs/src/probe/jlink/mod.rs#L243).

Can I ask you to paste the output of `RUST_LOG=probe_rs::probe::jlink=debug probe-rs erase`, please?

denysvitali, 2024-06-30:
The proposed PR (#2598) doesn't solve the problem - the probe actually returns a GET_MAX_MEM_BLOCK response (`11288`) and that's being used.

My probe is showing up as follows:

```
ID 1366:0101 SEGGER J-Link PLUS
```

Thus the probe settings become:

[image]

Changing the JTAG chunk size or limiting the `max_mem_block_size` doesn't help

denysvitali, 2024-07-09:
I took two packet captures with Wireshark, one with the OpenOCD command and one with `probe-rs info`. I've then processed the result with a small Python script and I've got this thing that should help us debug the issue further:

### OpenOCD

```
EMU_CMD_VERSION:
70 00   p.
4a 2d 4c 69 6e 6b 20 41 52 4d 2d 4f 42 20 53 54   J-Link ARM-OB ST
4d 33 32 20 63 6f 6d 70 69 6c 65 64 20 41 75 67   M32 compiled Aug
20 32 32 20 32 30 31 32 20 31 39 3a 35 32 3a 30    22 2012 19:52:0
34 00 43 6f 70 79 72 69 67 68 74 20 32 30 30 33   4.Copyright 2003
2d 32 30 30 38 20 53 45 47 47 45 52 3a 20 77 77   -2008 SEGGER: ww
77 2e 73 65 67 67 65 72 2e 63 6f 6d 00 00 00 00   w.segger.com....
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
EMU_CMD_GET_CAPS:
33 58 ea 88   3X..
EMU_CMD_GET_CAPS_EX:
33 58 ea 88 0c 28 00 00 00 00 00 00 00 00 00 00   3X...(..........
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
EMU_CMD_GET_HW_VERSION:
70 11 01 00   p...
EMU_CMD_GET_MAX_MEM_BLOCK:
18 2c 00 00   .,..
EMU_CMD_READ_CONFIG:
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
EMU_CMD_GET_STATE:
e4 0c 00 00 01 01 00 00   ........
Request: 
09 64 00 00 00 00 00 00 00 00 00 00 00 00   .d............
03 00 04 00 10 00 04 00 00 00 00 00 00 00 00 00   ................
00 00 03 00 44 f1 08 00 00 00 00 00 00 00 00 00   ....D...........
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
00 00 00 00 00 00 00 00 e4 0c 00 00   ............
Request: 
c7 ff   ..
03 00 00 00   ....
Request: 
c7 01   ..
01 00 00 00   ....
EMU_CMD_HW_RESET1:
EMU_CMD_HW_TRST1:
Request: 
05 e8 03   ...
Request: 
05 e8 03   ...
Request: 
cf 00 be 00 ff ff ff ff ff ff ff ff ff ff ff ff   ................
ff ff ff ff ff ff 00 00 00 00 c0 3f ff ff ff ff   ...........?....
ff ff ff 9e e7 ff ff ff ff ff ff ff 00 a5 00 00   ................
00 00 00 00   ....
ff ff ff ff ff ff ff 9e e7 ff ff ff ff ff ff ff   ................
00 a5 b9 a3 88 5d 08 00   .....]..
00   .
Request: 
[... 301 lines cut ...]
94 fe ff ff ff ff 00   .......
00   .
Request: 
cf 00 38 00 ff 03 00 00 00 00 ff 94 02 00 00 00   ..8.............
00 00   ..
94 fe ff ff ff ff 00   .......
00   .
Request: 
cf 00 38 00 ff 03 00 00 00 00 ff 94 02 00 00 00   ..8.............
00 00   ..
94 fe ff ff ff ff 00   .......
00   .
Request: 
cf 00 38 00 ff 03 00 00 00 00 ff 94 02 00 00 00   ..8.............
00 00   ..
94 fe ff ff ff ff 00   .......
00   .
Request: 
cf 00 38 00 ff 03 00 00 00 00 ff 94 02 00 00 00   ..8.............
00 00   ..
94 fe ff ff ff ff 00   .......
00   .

(...)
```

bugadani, 2024-07-11:
Please try this PR: https://github.com/probe-rs/probe-rs/pull/2674

denysvitali, 2024-07-12:
Unfortunately that didn't solve my issue :(
Is there anything else I can do from my side to help you debug this further?

bugadani, 2024-07-12:
I managed to source an nRF51 devkit (nrf52422) and it works without an issue. Can you tell us more about your hardware? What exact nRF chip do you use, down to the revision? How do you exactly connect your J-Link to it? Can you update the firmware on your J-Link? Can you try different debug probes with your nRF, or different chips with your debug probe? Is there a probe-rs version that works for you?

denysvitali, 2024-07-30:
I think the problem is the probe, as I've tried with different chips and the probe with probe-rs doesn't work.

As an example, I have here with me an `nrf52832-QFAA`: `openocd` works fine:

```
> nrf5 info
nrf5 info
nRF52832-QFAA(build code: E0) 512kB Flash, 64kB RAM
Target not halted
```

On the other hand, `probe-rs` just fails:
```
$ probe-rs info
Probing target via JTAG

Error identifying target using protocol JTAG: An error with the usage of the probe occurred

Probing target via SWD

Error showing ARM chip information for Debug Port Default: An ARM specific error occurred.

Caused by:
    0: An error occurred in the communication with an access port or debug port.
    1: Target device did not respond to request.

Trying alternate multi-drop debug ports
^C
```

bugadani, 2024-07-30:
Thanks, I've ordered one of these probes. Provided I get one with the same firmware, I can look into this.

denysvitali, 2024-07-30:
If there is anything from my side that can help you debug this further before you get the probe, just let me know (:

denysvitali, 2024-08-01:
`probe-rs` worked fine briefly with my `nrf52832-QFAA`:

```
nRF52832-QFAA(build code: E1) 512kB Flash, 64kB RAM
```

Unfortunately, as soon as I started writing this comment, `probe-rs` started returning the same errors as before

bugadani, 2024-08-01:
I think the `94 fe ff ff ff ff 00` response in your wireshark trace is nonsense. It's as though your device didn't recognise that we're talking to it. The first 2 idle bits and the request is included in the response, but the rest seems to be just a bunch of silence. I don't know why this would happen.

One notable difference I found is that we prepend two idle bits to our transfers. I'm removing those in https://github.com/probe-rs/probe-rs/pull/2730 - if you want to give it a spin, I'd appreciate feedback whether this changes anything. It really shouldn't, but this is still a difference.

denysvitali, 2024-08-01:
That didn't work unfortunately :(

bugadani, 2024-08-02:
How about https://github.com/probe-rs/probe-rs/pull/2732?

denysvitali, 2024-08-04:
It works!!!

```
Probing target via SWD

ARM Chip with debug port Default:
Debug Port: DPv1, DP Designer: ARM Ltd
├── 0 MemoryAP (AmbaAhb3)
│   └── ROM Table (Class 1), Designer: Nordic VLSI ASA
│       ├── Cortex-M4 SCS   (Generic IP component)
│       │   └── CPUID
│       │       ├── IMPLEMENTER: ARM Ltd
│       │       ├── VARIANT: 0
│       │       ├── PARTNO: Cortex-M4
│       │       └── REVISION: 1
│       ├── Cortex-M3 DWT   (Generic IP component)
│       ├── Cortex-M3 FBP   (Generic IP component)
│       ├── Cortex-M3 ITM   (Generic IP component)
│       ├── Cortex-M4 TPIU  (Coresight Component)
│       └── Cortex-M4 ETM   (Coresight Component)
└── 1 Unknown AP (Designer: Nordic VLSI ASA, Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x0)

Debugging RISC-V targets over SWD is not supported. For these targets, JTAG is the only supported protocol. RISC-V specific information cannot be printed.
Debugging Xtensa targets over SWD is not supported. For these targets, JTAG is the only supported protocol. Xtensa specific information cannot be printed.
```

denysvitali, 2024-08-04:
Actually, it seems like only `probe-rs info` works now, I seem to have problem with `run` now:

```
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.14s
     Running `probe-rs run --chip nRF52832_xxAA target/thumbv7em-none-eabi/debug/binary`
 WARN probe_rs::session: Could not clear all hardware breakpoints: Arm(Probe(ProbeSpecific(Usb(Kind(TimedOut)))))                                                                                         WARN probe_rs::session: Failed to deconfigure device during shutdown: Arm(Probe(ProbeSpecific(Usb(Kind(TimedOut)))))                                                                                     WARN probe_rs::architecture::arm::communication_interface: Failed to stop DP Default                                                                                                                    Error: The flashing procedure failed for 'target/thumbv7em-none-eabi/debug/binary'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: An ARM specific error occurred.
    2: Error using access port FullyQualifiedApAddress { dp: Default, ap: V1(0) }.
    3: Failed to write register DRW at address 0x0c
    4: The debug probe encountered an error.
    5: An error which is specific to the debug probe in use occurred.
    6: A USB transport error occurred.
    7: timed out

Stack backtrace:
   0: anyhow::error::<impl core::convert::From<E> for anyhow::Error>::from
   1: probe_rs::cmd::run::Cmd::run
   2: probe_rs::main
   3: std::sys_common::backtrace::__rust_begin_short_backtrace
   4: std::rt::lang_start::{{closure}}
   5: std::rt::lang_start_internal
   6: main
   7: <unknown>
   8: __libc_start_main
   9: _start
```

bugadani, 2024-08-04:
Well at least we took a step. I'll continue investigating once the probe reaches me - it might take some time, sorry about that.

denysvitali, 2024-08-04:
No worries. Thanks for all your efforts!

bugadani, 2024-08-17:
Awesome, I got my probe and I can reproduce the same "target did not respond" error with an RP2040. The joys of clone probes with ancient firmware.

bugadani, 2024-08-17:
I've updated https://github.com/probe-rs/probe-rs/pull/2732 with some things I found and I verified that the PR works with the RP2040.

Unfortunately I'm pretty sure this probe will bite us in the butt again in the future, as it reports 64K as max available memory, but the MCU itself has 64K so we shouldn't use that value directly. Thankfully it looks like every one of these probes as the serial number of `000000123456` so we'll be able to work around it if the problem ever pops up.

PR #2732 (2024-09-01): Fix compatibility with clone J-Link devices
Closes #2593

Old firmware needs some additional setup that we did not do. This PR fixes clone J-Link with firmware `J-Link ARM-OB STM32 compiled Aug 22 2012 19:52:04`. The config bits aren't necessary I think but they also doesn't really hurt - except for the fact where they can't be read as exactly 256 bytes for some reason, which requires the mess in `read()` to work around.
closed 2024-09-01 by bugadani