probe-rs/probe-rs #2606: Problems with debugging VA416xx

robamu, 2024-07-01:
Hi,

The support for the VA416xx devices was recently merged and I am definitely able to flash apps to the NVM now, and they also run after a reboot. Unfortunately, I now encountered another issue :(

Debugging with probe-rs seems to be problematic.

Some steps to re-produce for me:

1. Flash app with `cargo embed --example rtt-log`
2. Connect with `gdb-multiarch`
3. Set a breakpoint in a permanent-loop of the app using `b rtt-log.rs:29`
4. The program does not stop at the breakpoint

Output of `gdb-multiarch`, I also printed the breakpoint:

```txt
va416xx-rs on  flashing-update [$!] via 🦀 v1.78.0
❯ gdb-multiarch  target/thumbv7em-none-eabihf/debug/examples/rtt-log
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from target/thumbv7em-none-eabihf/debug/examples/rtt-log...
(gdb) target remote localhost:2331
Remote debugging using localhost:2331
0x00000b4c in lib::inline::__delay (cyc=<optimized out>) at asm/inline.rs:62
62      asm/inline.rs: No such file or directory.
(gdb) b rtt-log.rs:29
Breakpoint 1 at 0x806: file examples/simple/examples/rtt-log.rs, line 29.
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) c
Continuing.
^C
Program received signal SIGINT, Interrupt.
lib::inline::__delay (cyc=<optimized out>) at asm/inline.rs:62
62      in asm/inline.rs
(gdb) info b
Num     Type           Disp Enb Address    What
1       breakpoint     keep y   0x00000806 in rtt_log::__cortex_m_rt_main
                                           at examples/simple/examples/rtt-log.rs:29
(gdb)

```
If I flash the program directly to RAM, i get the following expected behaviour:

```txt
va416xx-rs on  flashing-update [$!] via 🦀 v1.78.0 took 2m9s
❯ cargo run --example rtt-log
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.15s
     Running `gdb-multiarch -q -x jlink/jlink.gdb target/thumbv7em-none-eabihf/debug/examples/rtt-log`
Reading symbols from target/thumbv7em-none-eabihf/debug/examples/rtt-log...
rtt_log::__cortex_m_rt_main () at examples/simple/examples/rtt-log.rs:29
29          loop {
Resetting target
Breakpoint 1 at 0x5be: file examples/simple/examples/rtt-log.rs, line 13.
Loading section .vector_table, size 0x350 lma 0x0
Loading section .text, size 0x3674 lma 0x350
Loading section .rodata, size 0xf1c lma 0x39d0
Start address 0x00000350, load size 18656
Transfer rate: 73 KB/sec, 6218 bytes/write.
(gdb) b rtt-log.rs:29
Breakpoint 2 at 0x806: file examples/simple/examples/rtt-log.rs, line 29.
(gdb) c
Continuing.

Breakpoint 1, rtt_log::__cortex_m_rt_main_trampoline () at examples/simple/examples/rtt-log.rs:13
13      #[entry]
(gdb) c
Continuing.

Breakpoint 2, rtt_log::__cortex_m_rt_main () at examples/simple/examples/rtt-log.rs:29
29          loop {
(gdb) info b
Num     Type           Disp Enb Address    What
1       breakpoint     keep y   0x000005be in rtt_log::__cortex_m_rt_main_trampoline
                                           at examples/simple/examples/rtt-log.rs:13
        breakpoint already hit 1 time
2       breakpoint     keep y   0x00000806 in rtt_log::__cortex_m_rt_main
                                           at examples/simple/examples/rtt-log.rs:29
        breakpoint already hit 1 time
(gdb)

```

One thing I am wondering about is the line `Note: automatically using hardware breakpoints for read-only addresses.`.
Why does this only occur with the app which was flashed to the NVM? For test purposes, I explicitly disabled the ROM protection in the rtt logger application, so I am not sure why exactly this is happening. Even when flashing to RAM directly, the ROM protection will be enabled after the flash process again, so the ROM protection should not be the issue.

Could this also be related to the way the device boots? On reset or startup, the MCU will copy instruction code from a connected SPI FRAM to the instruction SRAM mapped from 0x0 to 0x40000 and then execute that code.

One way I used to get a breakpoint was to use the `attach` mode instead of the `launch` mode in VS Code. However, the debugging here is kind of brittle. Stepping around/stepping into the wrong functions will cause the same behaviour where the debugger does not stop at breakpoints anymore. 

One thing I also tried was just flashing the app in VS Code with the `launch` config. Here, I always end up in the ASM delay  function (at least pausing seems to work then?). Trying to step out of that function will lead to weird behaviours and errors:

```txt
probe-rs-debug: RTT Window reused, and ready to receive RTT data on channel 0
ERROR probe_rs::debug::debug_step:356: The core did not halt after stepping to 0x00000B4E. Forced a halt at 0x00000B4A. Long running operations between debug steps are not currently supported.
 WARN probe_rs::probe::arm_debug_interface:997: Error in access 1/2 of block access: Target device responded with a FAULT response to the request.
ERROR: probe-rs-debug: "An ARM specific error occurred.\n\tError using access port.\n\t\tFailed to write register CSW at address 0x00000000\n\t\t\tAn error occurred in the communication with an access port or debug port.\n\t\t\t\tTarget device responded with a FAULT response to the request.\n"
ERROR: probe-rs-debug: "Debug Adapter terminated unexpectedly with an error: ProbeRs(Arm(AccessPort { address: ApAddress { dp: Default, ap: 0 }, source: RegisterWrite { address: 0, name: \"CSW\", source: Dap(FaultResponse) } }))\n"
 WARN probe_rs::probe::arm_debug_interface:1031: DAP FAULT
 WARN probe_rs::probe::arm_debug_interface:1041: Writing DAP register failed. Ctrl/Stat register value is: Ctrl {
    .0: 4026535715,
    csyspwrupack: true,
    csyspwrupreq: true,
    cdbgpwrupack: true,
    cdbgpwrupreq: true,
    cdbgrstack: false,
    c_dbg_rst_req: false,
    errmode: false,
    trn_cnt: 0,
    mask_lane: 15,
    w_data_err: false,
    read_ok: false,
    sticky_err: true,
    stick_cmp: false,
    trn_mode: 0,
    sticky_orun: true,
    orun_detect: true,
}
probe-rs-debug: Session ended: An ARM specific error occurred.
probe-rs-debug: DAP Protocol server exiting
```

Is this a problem which has occured for other MCUs as well? How can I do troubleshooting here?

robamu, 2024-07-01:
Small followup: Is there an easy way to directly flash an app to the RAM instead of programing the NVM, similarly to how flashing with the J-Link works? I'd be interested if debugging/stepping works better here..

bugadani, 2024-07-01:
> Is there an easy way to directly flash an app to the RAM instead of programing the NVM

not yet. we can download to ram, but we can't boot an app from ram

robamu, 2024-07-08:
To be honest, after some more work with the MCU, booting an app from RAM would probably be the most important way to develop applications for that chip. At some stage, the NVM will hold a bootloader and two application images for software update and redundancy/safety purposes. The chip erasure will then be a problem.

Is the boot from RAM planned as a feature in the next time? What would be some tasks that need to be done to enable that? Is there any way I can help?

bugadani, 2024-07-08:
> Is the boot from RAM planned as a feature in the next time?

It's not on my radar. I personally haven't look into what's needed yet. I know that [teleprobe](https://github.com/embassy-rs/teleprobe/) can run its tests from RAM, so if you want to help us get this implemented, studying that might be helpful - but keep in mind that teleprobe's scope is very limited.

My approach would be something like:
- add a debug sequence that can do basic architecture-specific setup to run a random piece of code from RAM. Since we have a flasher that essentially does the same, maybe it's possible to reuse/generalize parts of that. I'd add this to the default 3 debug sequences, so that targets with special requirements can override it. Since you only need ARM, we might even leave the other 2 unimplemented (and return an appropriate-ish error).
- after flashing, determine if the entry point is in RAM
- if it is, call the debug sequence that sets said entry point up and optionally starts the CPU

robamu, 2024-07-08:
I will open up a new issue to track this.

robamu, 2024-07-10:
The RAM flashed image unfortunately seems to have the same weird behaviour as the NVM flashed one.
I still think the RAM flash/boot is a useful feature though. I guess the issue with the weird debugging behaviour is something different related to the DAP server or some other weird device behaviour.

This is again something I do not have in-depth familiarity with.. What would be some possible ways to better debug/investigate this issue?

Yatekii, 2024-11-22:
`Note: automatically using hardware breakpoints for read-only addresses.` that's because we declare the NVM regions as read-only to force GDB to use HW instead of SW BPs. This was probably a bad choice but can't say for sure. GDB has no way of telling `break` to make a HW breakpoint by default. Only `hbreak` does this. As soft breaks are much much more effort, at the time I made it so `break` is always a HW break through the memory map instead of making the command set a HW breakpoint hardcoded.

robamu, 2024-11-25:
Ah okay. I wonder if this is related to the issues..