probe-rs/probe-rs #2617: probe-rs / target-gen using / failing memory verification aside official CMSIS functions

pkoevesdi, 2024-07-02:
**Describe the bug**
`target-gen test` and `probe-rs run` fail after first erase operation when the external flash is not memory-mapped.

**To Reproduce**
1. program a (working) flash algorithm without memory mapping
2. run `target-gen test`

**Expected behavior**
Test runs through without error message, flash gets written.

**Desktop (please complete the following information):**
- Linux Ubuntu

**Additional context**
The [CMSIS](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/algorithmFunc.html) specification has a verify function. Looking at the [source of target-gen](https://github.com/probe-rs/probe-rs/blob/master/target-gen/src/commands/test.rs#L168), I see that the verifiction function that might be provided by the flash algorithm is overriden by an own implementation, which doesn't work, when the flash is not memory-mapped. 
Whether it is or not, I assume as a descision of the user's project - it must not be. The CMSIS function I consider as the official interface, that the user must provide, not more. Don't force him to memory-map something, because 
- that's sometimes not possible (for instance not supported by the MCU), 
- or not necessary 
   - for instance, if the internal part of the app implements some own functions for reading/writing the flash, for numerous possible reasons. 
   - Also, if we want use an exisiting bootloader binary, even with memory-mapped mode, but which is not CMSIS compatible and thus cannot be used by `target-gen` / `probe-rs`. Example [here](https://github.com/cturvey/stm32extldr/blob/main/h7_is25lp128/CLIVEONE-IS25LP128_STM32H72X-PA3-PC11-PC9-PC10-PE2-PA6.stldr)
 
Some has to think about how to handle the `--restore-unwritten` use cases (because CMSIS has no read function), but first of all, it should generally work with any given CMSIS flash algorithm and _it's_ functions, and not assume any functionality of the MCU or memory beside these. In the end, it's ok to make it a condition to have memory-mapped mode when using  `--restore-unwritten`, but only then.

Hope, You don't mind bringing this up here, it's in total respect and appreciation for Your work, and of course, I'm totally open to discuss and help, as far as I can with my limited knowledge. :-)

FYI: 
- Here's my project with which I tested: https://github.com/pkoevesdi/stm32h730xx_is25lp128
- I discussed this a bit with / got support from @bugadani on [Matrix Rust Embedded](https://matrix.to/#/!vhKMWjizPZBgKeknOo:matrix.org/$aQowb1C8RLUG3RN0gLrLZXKm57YvNJaNmDtXEFcJwZs?via=matrix.org&via=chat.berline.rs&via=tchncs.de) (above and below).

bugadani, 2024-07-02:
To be quite honest, running code from external flash will also fail if said flash is not memory mapped. The one exception I can think of is a bootloader that loads code into RAM - but otherwise the MCU needs to be able to read from an address transparently.

We can't tell whether a memory region can or can't be read (well, we can, but an unreadable memory region makes no sense). We assume that if a target lists a memory region, we can read it. It is therefore the job of the target (either its flash algorithm or a debug sequence) to make it possible for us to read the memory.

That being said, in the context of flashing, we **could** provide an optional read function. We are permitted to add our own functions to the collection offered by CMSIS, it just will not be standard.

This would allow us to use the Verify operation to read memory - for example by copying flash into RAM, as sort-of an inverse operation to flashing. It is easy to check whether this function is provided (it is either set or not set in the flash algo's fields). We would need a function signature like `fn read(from_flash_addr: u32, to_ram_dst: u32, bytes: usize) -> ResultCode`.

pkoevesdi, 2024-07-02:
As far as I saw, the reading happens in `target-gen test` only for the purpose of verification of previous write or erase operations. And that's what `verify` is intended for in my opinion. What are the use cases of reading, except verification (and the mentioned `restore-unwritten`)?

In general, I like the idea of extending the flash algorithm by a read function, and consequently useing only that (if provided), to become agnostic of any memory-mapped mode or not.

bugadani, 2024-07-30:
Turns out we need this for the ESP32, too 🙃

pkoevesdi, 2024-11-01:
Any news on this? I see the merge, which seems about reading. I'd really like to make probe-rs work with my external QSPI flash  without being forced to memory-map it (which turns out to become hard for a beginner like me).

That's why I'd appreciate, if the linked lines in target-gen could be replaced by a user-defined [verify function](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/algorithmFunc.html#Verify), as it is standardized in Open-CMSIS - probably out of a reason, so no doubt that it makes sense not to override it per se, and thus limit probe-rs to certain use cases / exclude other use cases.

Yatekii, 2024-11-23:
This should be possible now, no?

bugadani, 2024-11-23:
Yes, but `target-gen test` doesn't use the new reads so it would fail even if the flash algo is otherwise working.

Yatekii, 2024-11-23:
Ah, thanks for the clarification, I'll have a look!