probe-rs/probe-rs #2658: Allow flashing/running/debugging an elf file into RAM directly

robamu, 2024-07-08:
This discussion was started [here](https://github.com/probe-rs/probe-rs/issues/2606).
Basically, I think flashing and debugging in RAM directly might be a really useful feature. For a lot of the embedded targets I have worked with so far, this was the standard way I debugged applications on embedded targets. Once a bootloader and multiple images are located on the NVM, the full/partial erasure during a NVM flash can also be something that needs to be avoided, and a RAM flash then becomes the standard development process. I might have a try of implementing this.

This might be unique to the chip I am using:  The addresses of the NVM and the instruction RAM are identical. Determining whether the entry point is in RAM or in the NVM would not work in that way, so I was thinking of a separate command like `probe-rs ramflash`. Thanks for the pointer to `teleprobe`, this might be very useful. Considering the special sequence required to start an App from RAM, the code seems to be similar to the code I have seen when jumping from bootloader to an other application and involves setting up [PC/SP and the VTOR register](https://github.com/embassy-rs/teleprobe/blob/f29415d731b673a74fe1fba516b5f31df1e7cda8/teleprobe/src/run.rs#L195).

I will probably start with a similar approahc like teleprobe by using `probe-rs` as a library first to try to flash/debug a simple blinky image in RAM first.

bugadani, 2024-07-08:
> This might be unique to the [VA416XX](https://www.voragotech.com/products/va41630) chip I am using: The addresses of the NVM and the instruction RAM are identical, and the MCU has a startup process which copies the first 256 kB of the QSPI NVM to the instruction RAM at boot before performing the regular Cortex-M boot sequence.

The target you added contains two ram regions literally called IRAM, plus an NVM region, so I guess the regions need to be more aptly named.

I would strongly prefer not introducing a new command as I don't believe it is necessary. For your particular target, you can introduce a variant where you redefine the current NVM region as RAM. Then you'll be able to choose whether you want the same app to be flashed or ram-loaded. Even if NVM and IRAM are aliased, code running from RAM should be able to be linked, and then loaded to any executable RAM region.

robamu, 2024-07-08:
Yes, the names are definitely confusing and a probably based on the Keil IDE names. IRAM2 is not even accessible for the instruction RAM which makes this even more confusing. I'll definitely rename those memory regions.

I actually managed to flash an app and make it run in RAM, but of course the MCU requires some weird adaptions to existing code again to make the flash process work.. During the RAM flash process [here](https://github.com/probe-rs/probe-rs/blob/0d222fe6adaa1bdfd1179eb0deca40380e87705c/probe-rs/src/flashing/loader.rs#L506), `write_8` is used to write the memory while `read` is used to read back the memory. Is there a specific reason why read is used instead of `read_8`? In any case, I get a verification failure during the flash process. I have to use `read` and `write`, and then the flash process works and the application seems to run after the PC/SP/VTOR setup.
Maybe this is possibly related to the IDLE cycles between reads/writes again?

bugadani, 2024-07-08:
> write_8 is used to write the memory while read is used to read back the memory. Is there a specific reason why read is used instead of read_8

None that I know of.

> Maybe this is possibly related to the IDLE cycles between reads/writes again?

Your MCU might have interesting ideas about accessing RAM. This isn't really enough info to know more, and I don't exactly have a lot of relevant experience with ARM at all.

robamu, 2024-07-09:
Hello,

I added and wrote a new debug sequence like suggested:

```rust
    /// This ARM sequence is called if an image was flashed to RAM directly.
    /// It will perform the necessary preparation to run that image.
    fn ram_debug_start(
        &self,
        vector_table_offset: u64,
        interface: &mut dyn ArmProbe,
        core: &mut dyn CoreInterface,
        core_type: CoreType,
    ) -> Result<(), crate::Error> {
        const SP_MAIN_OFFSET: usize = 0;
        const RESET_VECTOR_OFFSET: usize = 1;
        match core_type {
            CoreType::Armv7a => unimplemented!(),
            CoreType::Armv8a => unimplemented!(),
            CoreType::Armv6m | CoreType::Armv7m | CoreType::Armv7em | CoreType::Armv8m => {
                // See ARMv7-M Architecture Reference Manual Chapter B1.5 for more details.
                let vtor = Vtor(vector_table_offset as u32);
                let mut first_table_entries: [u32; 2] = [0; 2];
                interface.read_32(vector_table_offset, &mut first_table_entries)?;
                // The first entry in the vector table is the SP_main reset value of the main stack pointer,
                // so we set the stack pointer register accordingly.
                core.write_core_reg(SP.id, first_table_entries[SP_MAIN_OFFSET].into())?;
                // The second entry in the vector table is the reset vector. It needs to be loaded
                // as the initial PC value on a reset, see chapter A2.3.1 of the reference manual.
                core.write_core_reg(PC.id, first_table_entries[RESET_VECTOR_OFFSET].into())?;
                interface.write_word_32(Vtor::get_mmio_address(), vtor.0)?;
            }
            _ => {
                panic!("Logic inconsistency bug - non ARM core type passed {core_type:?}");
            }
        }
        Ok(())
    }
```

The vector_table_offset function argument is the entry point here. Oftentimes, this is just 0x0, but it can be another
value, and that value is the start of the `.vector_table` region for cortex-m-rt based applications.
I determined this table offset by parsing the ELF file (can of course also be done with the Object library):

```rust
    let path = Path::new("../va416xx-rs/target/thumbv7em-none-eabihf/debug/examples/rtt-log");
    let elf_bytes = fs::read(path).expect("failed to read elf file");
    let mut vector_table = None;
    match goblin::Object::parse(&elf_bytes) {
        Ok(obj) => match obj {
            goblin::Object::Elf(elf) => {
                for program_header in &elf.program_headers {
                    println!("program header: {:?}", program_header);
                }
                for section_header in &elf.section_headers {
                    let str = elf
                        .shdr_strtab
                        .get_at(section_header.sh_name)
                        .expect("getting section name failed");
                    println!("section header string: {}", str);
                    if str == ".vector_table" {
                        let vector_table_raw = &elf_bytes[section_header.sh_offset as usize
                            ..section_header.sh_offset as usize + section_header.sh_size as usize];

                        let first_four_vectors = vector_table_raw
                            .chunks_exact(4)
                            .map(|chunk| u32::from_le_bytes(chunk.try_into().unwrap()))
                            .collect::<Vec<_>>();
                        vector_table = Some(VectorTable {
                            offset: section_header.sh_addr as u32,
                            // Initial stack pointer
                            initial_sp: first_four_vectors[0],
                            reset: first_four_vectors[1],
                            hard_fault: first_four_vectors[3],
                        });
                    }
                }
            }
            _ => {
                panic!("expected elf, got other object");
            }
        },
        Err(e) => {
            panic!("error {}", e);
        }
    }
```
 
I am not sure how this is handled for other architectures and this might be something
ARM specific in general. Is it sufficient to just assume that the lowest
address is the vector table offset, and then perform the memory region check on that address before calling
the `ram_flash_start` function for ARM targets only?

bugadani, 2024-07-09:
> Is it sufficient to just assume that the lowest
address is the vector table offset, and then perform the memory region check on that address before calling
the ram_flash_start function for ARM targets only?

I don't know, but we can figure it out later. You can return `probe_rs::error::Error::NotImplemented("boot_from_ram")` for the other architectures.

I would suggest not passing any arm-specific interfaces to the sequence. You can access memory through `CoreInterface`, but I think the best would be to pass `Session`, and let the sequence figure out what it needs. For multi-core chips, there might be different setup needed.

robamu, 2024-07-09:
Thanks, I'll adapt it. Can I just add an `entry_point` field in the flash loader structure which then gets passed to the new debug sequence?

.After some more testing about the `read_8` / `write_8` issue: The MCU behaves like it does not support 8 bit reads and writes. `read_8` and `write_8` do not appear to work at all. Only 4-byte aligned `read`s and `write`s do. 8 bit support is determined by reading the CSW register on the AP, and the value here suggests that 8-bit read/writes should be supported as well. I checked the datasheets and errata again and did not find any Cortex-M or AP specific information. This MCU really seems to have interesting ideas.. I'll do some more testing and checking. The idle delays do not make a difference, the transfer does not seem to run into a timeout anyway.

bugadani, 2024-07-09:
> Would adding a entry_point field in the flash loader structure which then gets passed to the new debug sequence be a good first solution?

I'm not sure. I think it's okay to do this, but I'm also prone to rushing things and cleaning up later :) At this point, however, the `FlashLoader` name is a bit of a misnomer and that struct now represents the actual loaded image, so I think the entry point fits there.

It's perfectly valid for your MCU to only implement word-size access to at least its instruction memory. I believe we can change the flasher to avoid 8-bit reads.

robamu, 2024-07-09:
I am struggling a bit with retrieving an `ArmProbe` (or something similar) to allow writes to instruction memory.

I saw this method:

```rust
    /// Initialize core debug system. This is based on the
    /// `DebugCoreStart` function from the [ARM SVD Debug Description].
    ///
    /// [ARM SVD Debug Description]: https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/debug_description.html#debugCoreStart
    #[doc(alias = "DebugCoreStart")]
    fn debug_core_start(
        &self,
        interface: &mut dyn ArmProbeInterface,
        core_ap: MemoryAp,
        core_type: CoreType,
        debug_base: Option<u64>,
        cti_base: Option<u64>,
    ) -> Result<(), ArmError> {
        let mut core = interface.memory_interface(&core_ap)?;

        // Dispatch based on core type (Cortex-A vs M)
        match core_type {
            CoreType::Armv7a => armv7a_core_start(&mut *core, debug_base),
            CoreType::Armv8a => armv8a_core_start(&mut *core, debug_base, cti_base),
            CoreType::Armv6m | CoreType::Armv7m | CoreType::Armv7em | CoreType::Armv8m => {
                cortex_m_core_start(&mut *core)
            }
            _ => panic!("Logic inconsistency bug - non ARM core type passed {core_type:?}"),
        }
    }
  ```
  
  here the `core_ap` is passed explicitely. What would be the best way to determine this from the session?

bugadani, 2024-07-09:
Why do you want an `ArmProbe`, why is a `Core` insufficient for this? You may need to import the `MemoryInterface` trait, but it should work I believe.

robamu, 2024-07-09:
That was an easy fix, thank you :)

robamu, 2024-07-10:
It seems that the core is not halted for the flash process, and that probably would be a good idea when directly overwriting code that is currently running. Should I introduce a new debug sequence like `ram_flash_setup` which halts the core by default? This would also allow special devices like my MCU to disable the ROM protection which prevents writes to the instruction memory by default.

bugadani, 2024-07-10:
> Should I introduce a new debug sequence like ram_flash_setup

I don't think we should, not for this. For one, the name is super confusing :) Instead, you can disable relevant protections in your device's `debug_core_start`, maybe.

Regarding resets, the flasher itself has a `reset_and_halt` but I guess writing RAM doesn't need a flasher... Maybe you can add a `reset_and_halt` to `FlashLoader::commit`

robamu, 2024-07-10:
> It's perfectly valid for your MCU to only implement word-size access to at least its instruction memory. I believe we can change the flasher to avoid 8-bit reads.

That seems to be correct, 8-bit reads/writes from SRAM1 seem to work.