probe-rs/probe-rs #2660: error flash armv7m with " core is in locked up status"

USTHzhanglu, 2024-07-09:
**Describe the bug**
A clear and concise description of what the bug is.

ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception

**To Reproduce**
Steps to reproduce the behavior:

use: probe-rs: 0.24.0
```bash
probe-rs download --chip HC32F460PETB --chip-description-path .\HC32F460-Series.yaml --binary-format hex --speed 8000 hc32f460.hex
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exceptionError: The flashing procedure failed for 'hc32f460.hex'.
Caused by:
    0: The page write of the page at address 0x0001a000 failed.
    1: The core entered an unexpected status: LockedUp.
```
and use:probe-rs:0.23.0

```bash
probe-rs download --chip HC32F460PETB --chip-description-path .\HC32F460-Series.yaml --format hex --speed 8000 hc32f460.hex
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
      Erasing ✔ [00:00:01] [#############################################] 352.00 KiB/352.00 KiB @ 248.07 KiB/s (eta 0s )
  Programming ✔ [00:00:21] [##############################################] 349.00 KiB/349.00 KiB @ 15.94 KiB/s (eta 0s )
    Finished in 23.343s
```

probe-rs:0.22.0

```bash
probe-rs download --chip HC32F460PETB --chip-description-path .\HC32F460-Series.yaml --format hex --speed 8000 
hc32f460.hex
      Erasing ✔ [00:00:02] [#############################################] 352.00 KiB/352.00 KiB @ 167.51 KiB/s (eta 0s )
  Programming ✔ [00:00:22] [##############################################] 349.00 KiB/349.00 KiB @ 15.86 KiB/s (eta 0s )
    Finished in 24.17s
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

**Desktop (please complete the following information):**

- Windows 10

**Additional context**
Add any other context about the problem here.

bugadani, 2024-07-09:
There have been many changes between 0.23 and 0.24. Can you help us by trying to bisect which change introduced this issue?

USTHzhanglu, 2024-07-09:
> There have been many changes between 0.23 and 0.24. Can you help us by trying to bisect which change introduced this issue?

how to do it?
seems like #2337

USTHzhanglu, 2024-07-10:
> There have been many changes between 0.23 and 0.24. Can you help us by trying to bisect which change introduced this issue?

i download master,it`s always some error,but like 0.23, it`s download success.

closed 2024-07-17 by USTHzhanglu

standarddeviant, 2024-07-20:
FYI, On a completely different chip, I observe the same general pattern for erasing a chip:
* 0.24.0 produces `locked up...` error
* 0.23.0 produces `locked up...` error
* 0.22.0 seems to work 

I downloaded windows releases of `0.22.0`, 0.23.0`, and `0.24.0` and am testing with an RPi-Pico-OpenOCD probe and am targeting *MDBT42T-DB-Nordic-nRF52805-Bluetooth-Antenna* eval board.

Here are the three different directories w/ versions in the change logs
```
~\code\perase
❯ rg `## \[0\.2[234].\d+\]` .\22
.\22\CHANGELOG.md
8:## [0.22.0]

~\code\perase
❯ rg `## \[0\.2[234].\d+\]` .\23
.\23\CHANGELOG.md
6:## [0.23.0]
75:## [0.22.0]

~\code\perase
❯ rg `## \[0\.2[234].\d+\]` .\24
.\24\CHANGELOG.md
8:## [0.24.0]
119:## [0.23.0]
185:## [0.22.0]
```

And then issuing a simple erase commands using the respective downloaded `probe-rs.exe` for minor versions 22, 23, 24.
```
~\code\perase
❯ .\22\probe-rs.exe erase --chip nrf52805_xxAA

~\code\perase
❯ .\23\probe-rs.exe erase --chip nrf52805_xxAA
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception

~\code\perase
❯ .\24\probe-rs.exe erase --chip nrf52805_xxAA
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
```