probe-rs/probe-rs #2665: Regression for VA416xx Flash process

robamu, 2024-07-10:
I am doing some regression testing to ensure the NVM flash for the VA416xx MCU still works.

The PR https://github.com/probe-rs/probe-rs/pull/2626/files seems to have introduced a regression where the flashing process now
fails with the following error:

```txt
va416xx-rs on  bootloader-flashloader [$!] via 🦀 v1.78.0 took 1m6s
❯ cargo run --example blinky
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.06s
     Running `probe-rs run --chip VA416xx target/thumbv7em-none-eabihf/debug/examples/blinky`
 INFO probe_rs::config::target: Using sequence Arm(Va416xx)
 INFO probe_rs::probe::jlink: J-Link: Firmware version: J-Link OB-STM32F103 V1 compiled Oct 30 2023 12:12:03
 INFO probe_rs::probe::jlink: J-Link: Hardware version: J-Link 1.0.0
 INFO probe_rs::probe::jlink: J-Link: Target voltage: 3.30 V
 INFO probe_rs::architecture::arm::sequences: Debug port Default is powered down, powering up
 INFO probe_rs::session: Core status: Running
 INFO probe_rs::session: Halting core...
 INFO probe_rs::session: Resuming core...
 INFO probe_rs::flashing::download: Found loadable segment, physical address: 0x00000000, virtual address: 0x00000000, flags: 0x4
 INFO probe_rs::flashing::download: Matching section: ".vector_table"
 INFO probe_rs::flashing::download: Found loadable segment, physical address: 0x00000350, virtual address: 0x00000350, flags: 0x5
 INFO probe_rs::flashing::download: Matching section: ".text"
 INFO probe_rs::flashing::download: Found loadable segment, physical address: 0x00004220, virtual address: 0x00004220, flags: 0x4
 INFO probe_rs::flashing::download: Matching section: ".rodata"
 INFO probe_rs::flashing::loader: Found 3 loadable sections:
 INFO probe_rs::flashing::loader:     .vector_table at 0x00000000 (848 bytes)
 INFO probe_rs::flashing::loader:     .text at 0x00000350 (16072 bytes)
 INFO probe_rs::flashing::loader:     .rodata at 0x00004220 (4236 bytes)
 INFO probe_rs::flashing::flash_algorithm: Chosen RAM to run the algo: RamRegion { name: Some("IRAM1"), range: 1fff8000..20000000, cores: ["main"], access: None } INFO probe_rs::flashing::flash_algorithm: Data will be loaded to: RamRegion { name: Some("IRAM1"), range: 1fff8000..20000000, cores: ["main"], access: None } INFO probe_rs::flashing::flash_algorithm: The flash algorithm will be configured with 512 bytes of stack           INFO probe_rs::flashing::flash_algorithm: Stack top: 0x1fff8d1c                                                    INFO reset_and_halt: probe_rs::architecture::arm::sequences: Powering down debug port Default timeout=500ms        WARN reset_and_halt: probe_rs::architecture::arm::communication_interface: Failed to stop DP Default timeout=500ms WARN session_drop: probe_rs::session: Could not clear all hardware breakpoints: An ARM specific error occurred.

Caused by:
    0: Error using access port.
    1: Failed to write register CSW at address 0x00000000
    2: An error occurred in the communication with an access port or debug port.
    3: Target device did not respond to request.                                                                    WARN session_drop: probe_rs::session: Failed to deconfigure device during shutdown: An ARM specific error occurred.

Caused by:
    0: Error using access port.
    1: Failed to write register CSW at address 0x00000000
    2: An error occurred in the communication with an access port or debug port.
    3: Target device did not respond to request.                                                                    INFO probe_rs::architecture::arm::sequences: Powering down debug port Default                                      WARN probe_rs::architecture::arm::communication_interface: Failed to stop DP Default                              Error: The flashing procedure failed for 'target/thumbv7em-none-eabihf/debug/examples/blinky'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: An ARM specific error occurred.
    2: An error occurred in the communication with an access port or debug port.
    3: Target device did not respond to request.

```

bugadani, 2024-07-10:
https://github.com/probe-rs/probe-rs/pull/2626/files#diff-dc318eff50c95da560a737ec14797522ceb61240d0cff20d870d9833c6cf4653R518 seems to be the only relevant change, can you test reverting that? If that solves your issue, we can move ARM detection to the last step in the process.

bugadani, 2024-07-10:
Okay, we have two options:

- Retry faults, but not with the current (1000) limit
- Retry faults as previously, but do ARM chip detection last.

I prefer the first one but none of these tells us what actualy is wrong, so figuring that out would be ideal first. I also wonder if this is related to https://github.com/probe-rs/probe-rs/issues/2567#issuecomment-2218738616 so let's try resolve that first.

bugadani, 2024-07-10:
Do you know if this error happens during/after resetting the chip? If yes, can we make that more robust, by waiting somehow for some valid response?

robamu, 2024-07-10:
There is a reset before this happens, yes. There is the custom debug sequence which reinitializes the interface and calls `debug_core_start`, wouldn't that require a working interface?

bugadani, 2024-07-10:
What I'm saying is that ` 3: Target device did not respond to request.` ([where I think we raise it](https://github.com/probe-rs/probe-rs/blob/51de1b9538be153aa2ec84e84925cc8eb5df7e1f/probe-rs/src/probe/arm_debug_interface.rs#L209)) may hapen because of the reset. You don't wait for anything, you don't hand any errors, so I believe this error might be bubbled up from your function. The underlying cause is that the MCU doesn't respond to a command you send, because the MCU might still be trying to come alive. You can try verifying this, as well as handling this error in/around https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/vendor/vorago/sequences/va416xx.rs#L56-L57

robamu, 2024-07-10:
Hmm, I added [some re-try](https://github.com/us-irs/probe-rs/commit/f5f164d377f2bfbf2be1b39f0d7c342a333cc013) handling and now get the following error:

```
 INFO probe_rs::flashing::loader: Found 3 loadable sections:
 INFO probe_rs::flashing::loader:     .vector_table at 0x00000000 (848 bytes)
 INFO probe_rs::flashing::loader:     .text at 0x00000350 (16072 bytes)
 INFO probe_rs::flashing::loader:     .rodata at 0x00004220 (4236 bytes)
 INFO probe_rs::flashing::flash_algorithm: Chosen RAM to run the algo: RamRegion { name: Some("IRAM1"), range: 1fff8000..20000000, cores: ["main"], access: None } INFO probe_rs::flashing::flash_algorithm: Data will be loaded to: RamRegion { name: Some("IRAM1"), range: 1fff8000..20000000, cores: ["main"], access: None } INFO probe_rs::flashing::flash_algorithm: The flash algorithm will be configured with 512 bytes of stack                                    INFO probe_rs::flashing::flash_algorithm: Stack top: 0x1fff8d1c                                                                             INFO reset_and_halt: probe_rs::vendor::vorago::sequences::va416xx: attempting reinitialize for 0 time timeout=500ms                         INFO reset_and_halt: probe_rs::architecture::arm::sequences: Powering down debug port Default timeout=500ms                                 WARN reset_and_halt: probe_rs::architecture::arm::communication_interface: Failed to stop DP Default timeout=500ms                          INFO reset_and_halt: probe_rs::architecture::arm::sequences: Debug port Default is powered down, powering up timeout=500ms                  INFO reset_and_halt: probe_rs::vendor::vorago::sequences::va416xx: reinit done timeout=500ms                                                INFO reset_and_halt: probe_rs::vendor::vorago::sequences::va416xx: debug core start done timeout=500ms                                      INFO reset_and_halt: probe_rs::vendor::vorago::sequences::va416xx: reset system done timeout=500ms                                          WARN reset_and_halt: probe_rs::architecture::arm::core::armv7m: Core is running, but we expected it to be halted timeout=500ms              INFO session_drop: probe_rs::session: Core status: Running                                                                                  INFO session_drop: probe_rs::session: Halting core...                                                                                       WARN session_drop: probe_rs::session: Could not clear all hardware breakpoints: An ARM specific error occurred.

Caused by:
    Timeout occurred during operation.                                                                                                       INFO probe_rs::architecture::arm::sequences: Powering down debug port Default                                                               INFO probe_rs::architecture::arm::sequences: Powering down debug port Default                                                              Error: The flashing procedure failed for 'target/thumbv7em-none-eabihf/debug/examples/blinky'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: An ARM specific error occurred.
    2: The core needs to be halted for this operation but was not

```

robamu, 2024-07-10:
Hmm, it could be that I just need to wait until the core is really halted.. is there any way to access the `CoreInterface` for helper functions like `wait_for_core_halted`? Otherwise, I just copy&paste a lot of existing code..

robamu, 2024-07-10:
With [this adapted sequence](https://github.com/us-irs/probe-rs/blob/c5d980b4b31cfd3ad8186c1e7d9c72ce88fb152a/probe-rs/src/vendor/vorago/sequences/va416xx.rs#L31) which involves re-try handling for halting the core as well, it works. I am just confused why I have to do both the write and read for the halting so often (and why I have to also re-do the write).. I would have expected this to be really fast.

PR #2670 (2024-07-11): Re-try handling for VA416xx reset sequence
Fixes #2665
closed 2024-07-11 by bugadani