probe-rs/probe-rs #2686: just "verify" frimware

USTHzhanglu, 2024-07-17:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

i download my hex file over 24Mib to my board, it' cost a lot time. but sometime when i used "--verify", Verify Failed,i must retry download and verify. but when i use jflash, run "verify", not any err. i can't find why probe-rs verify failed.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

may add verify to probe-rs-cli? then i not need retry download that cost so much time. thanks

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Yatekii, 2024-07-18:
Sounds like a great addition :)

bugadani, 2024-07-18:
I'm going to link #2470 here - a verify mode could be implemented in such a way that we pre-verify during `probe-rs run`, too with a switch.

PR #2722 (2024-08-11): Verify command and pre-verify option
Closes #2470
Closes #2686
Closes #1866

The algo is pretty stupid, we could pre-verify per flash block, but per-flashalgo is at least safe even if the flash erase granularity info we have is incorrect.

Open question: should pre-verify be on by default? Obviously it has the potential of doubling the flashing time, but I expect it to be very rare for only the last bytes of a firmware to differ.

Improvement possibilities:

Progress indication. We currently don't give any, and it looks like probe-rs hangs for a few seconds before/after flashing. This isn't id [... cut]
closed 2024-08-11 by Yatekii