probe-rs/probe-rs #2687: probe-rs crashes from VSCode when viewing variables

watsug, 2024-07-17:
**Describe the bug**
When there are no static variables (it might be related to any kind of variables) the probe-rs crashes trying to dereference the element at index 0 from an empty list.

**To Reproduce**
Steps to reproduce the behavior:

1. start a debugging session
2. go to VARIABLES windows
3. navigate to 'Statis' when no static variables are present in the application

**Expected behavior**
Should not crash.

**Stacktrace**
probe-rs-debug: Log output for "probe_rs=warn" will be written to the Debug Console.
probe-rs-debug: Starting probe-rs as a DAP Protocol server
probe-rs-debug: Listening for requests on port 37429
probe-rs-debug: Starting debug session from: 127.0.0.1:57754
FLASHING: Starting write of "/home/user/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/zephyr.elf" to device memory
FLASHING: Completed write of "/home/user/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/zephyr.elf" to device memory
thread 'main' panicked at probe-rs/src/debug/unit_info.rs:1341:24:
index out of bounds: the len is 0 but the index is 0
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

**Desktop (please complete the following information):**

- Linux (Ubuntu/wsl)

**Additional context**
none

bugadani, 2024-07-17:
Can you share your elf, please?

bugadani, 2024-07-22:
I've tried my best to reproduce this issue - but creating a firmware with no static variables is actually pretty hard. Nevertheless, probe-rs does not crash for me, so I'd appreciate a reproducer.

[image]

Tiwalun, 2024-07-26:
I was also not able to reproduce this, I tried the same Zephyr sample but seems to work fine.

Looking at the location, it seems to come from this line: 

https://github.com/probe-rs/probe-rs/blob/6fc653ad01f1714cb183dc5ff5a5374cf306f61b/probe-rs/src/debug/unit_info.rs#L1341

So it seems to be somehow related to the handling of arrays. Preventing the panic shouldn't be too hard, but something like a core dump would be great to see why we end up with this problem.