probe-rs/probe-rs #2693: Inaccurate or outdated dates?

amcolex, 2024-07-24:
I'm trying to use the probe-rs library in order to make an automated flashing utility. however, the QuickStart examples don't seem to match the library at all. Or am i just missing something?

Here is my simple test code:

```
use probe_rs::{Permissions, Probe};

fn main() -> Result<(), probe_rs::Error> {
    // Get a list of all available debug probes.
    let probes = Probe::list_all();

    // Use the first probe found.
    let probe = probes[0].open()?;

    // Attach to a chip.
    let mut session = probe.attach("nRF52840_xxAA", Permissions::default())?;

    // Select a core.
    let mut core = session.core(0)?;

    // Halt the attached core.
    core.halt(std::time::Duration::from_millis(300))?;

    Ok(())
}
```

and Cargo.toml:

```
[package]
name = "naviq-updater"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
probe-rs = "0.24.0"
```

however i get:

error[E0432]: unresolved import `probe_rs::Probe`

If i correct the import with:

```
use probe_rs::probe::Probe;
```

It fails to find the list_all() function:

```
5   |     let probes = Probe::list_all();
    |                         ^^^^^^^^ function or associated item not found in `Probe`
```

bugadani, 2024-07-24:
Thanks, fixed in #2694. Please note that these are illustratory examples, and you should look for more details on https://docs.rs/probe-rs/latest/probe_rs/

amcolex, 2024-07-24:
Great, thanks for pointing to in the right direction for the docs. Got it all working. Awesome library.

It would help if that link was shared somewhere on the website https://probe.rs/docs/library/quickstart/

closed 2024-07-24 by amcolex

bugadani, 2024-07-24:
Indeed, thanks for the suggestion

PR #2694 (2024-07-24): Update outdated example
Closes #2693