probe-rs/probe-rs #2696: RP2040 Failed to switch to DP Multidrop

FlareFlo, 2024-07-25:
**Describe the bug**
Debugging RP2040 via picoprobe yields warnings and no defmt output.

**To Reproduce**
Steps to reproduce the behavior:

1. clone https://github.com/FlareFlo/magneto_pad
2. cargo run --release

or

1. probe-rs reset --chip rp2040 --protocol swd

**Expected behavior**
The operation (attach, reset or erase) returns properly

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

**Desktop (please complete the following information):**
- Linux (Ubuntu, Gentoo, Debian, Fedora, etc.)

**Additional context**
This message alternates around 10 times when running reset, for example.
```
WARN probe_rs::architecture::arm::communication_interface: Failed to switch to DP Multidrop(11002927): An error occurred in the communication with an access port or debug port.
```

```
WARN probe_rs::architecture::arm::communication_interface: Failed to switch to DP Multidrop(1002927): An error occurred in the communication with an access port or debug port.
```

FlareFlo, 2024-07-25:
I should add that, the repo linked does flash and run on the chip, without defmt output however.

bugadani, 2025-05-15:
Please re-test with current git master.

9names, 2025-05-15:
I just tested https://github.com/FlareFlo/magneto_pad rev 40235373fcf07a698b5d8caabe2ba2d4335adc10 with my pico (I picked a random commit from after this issue creation date and before it switched to stm32) and a nanoDAP-hs probe (cmsis-dap).
Works with probe-rs 20, 21.1, 22, 23, 24, 25, and 27.

@rursprung can you please share a reproducer for this issue?

9names, 2025-05-15:
Actually: after reading probe-rs matrix, your issue is different.
This one is specifically "can't ever communicate with the target", yours is a spurious error message.
@rursprung please make a new issue that actually states your problem, as well as which OS you're on etc etc

9names, 2025-05-15:
Also tested with a pico debug probe now. Still no issues running magneto_pad or running `probe-rs reset`, on any version of probe-rs I have.
Could you try updating your picoprobe firmware @FlareFlo?

rursprung, 2025-05-16:
> Actually: after reading probe-rs matrix, your issue is different. This one is specifically "can't ever communicate with the target", yours is a spurious error message. [...]

ah, sorry, i missed that. i've now raised https://github.com/probe-rs/probe-rs/issues/3316

closed 2025-05-19 by Yatekii