probe-rs/probe-rs #2711: Using with ESP32-C3-DevKitC-02 (with soldered cable)

lure23, 2024-07-27:
**Describe the bug**

The ESP32-C3 devkit has JTAG/USB support, if only a USB cable is [soldered in the right pins](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html).

While I can detect the JTAG device this way, my experience is I cannot get it to work in reality (flashing, erasing) with `probe-rs`.  It may be that it's rather rare that people actually solder such extensions, but decided to report this.  I have another (C6-based) devkit with built-in JTAG USB port, and there `probe-rs` works flawlessly (so I know what the good experience is like).

```
$ probe-rs list
The following debug probes were found:
[0]: ESP JTAG -- 303a:1001:54:32:04:41:7D:60 (EspJtag)
```

```
$ probe-rs erase
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: Unable to load specification for chip
    1: The connected chip could not automatically be determined.
```

```
$ probe-rs --version
probe-rs 0.24.0 (git commit: crates.io)
```

Is there something I can do to manually force proper detection? Have tried `--protocol jtag` and `--chip esp32c3` but they didn't help.

**To Reproduce**

Steps to reproduce the behavior:

1. have access to [ESP32-C3-DevKitC-02](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c3/esp32-c3-devkitc-02/user_guide.html) -- revision 0.4
2. attach a USB cable to pins as [mentioned here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html) (same link as in the beginning)
3. try to erase the board, using `probe-rs`

**Expected behavior**
`probe-rs erase` etc. work

**Desktop (please complete the following information):**

I am accessing the board from a VM (running Ubuntu Linux), using USB/IP. This arrangement works for the other (C6) board.

**Additional context**
Add any other context about the problem here.

lure23, 2024-07-27:
I notice there is an effort for [Automatic chip detection](https://github.com/probe-rs/probe-rs/issues/2449). I guess that could be related.

bugadani, 2024-07-27:
When you say you solder a USB cable to the C3, do you remove the resistors that connect the onboard USB-UART converter, right?

I have one of these devkits in my tester setup, slightly modified according to [this](https://esp32.com/viewtopic.php?t=21007#p76800) and it works just fine.

With the current probe-rs release, you'll need to pass `--chip esp32c3`, but this will not be necessary for some chips (esp32c3 included) in the future.

lure23, 2024-07-28:
Thanks @bugadani for confirming that the setup should work.

The root cause seems to be that I had moved the USB/IP hosting from Windows 10 to Raspberry Pi. Moving back, the board and `probe-rs erase --chip=esp32c3` work. 🌞

Closing, but for completeness the output when things didn't work:

```
$ probe-rs erase --chip=esp32c3
 WARN probe_rs::session: Could not clear all hardware breakpoints: An error with the usage of the probe occurred

Caused by:
    0: USB Communication Error
    1: endpoint STALL condition
 WARN probe_rs::session: Failed to deconfigure device during shutdown: An error with the usage of the probe occurred

Caused by:
    0: USB Communication Error
    1: endpoint STALL condition
Error: Failed to erase the whole chip.

Caused by:
    0: Something during the interaction with the core went wrong
    1: An error with the usage of the probe occurred
    2: USB Communication Error
    3: endpoint STALL condition
```

closed 2024-07-28 by lure23